arendst/Tasmota #3661: Sonoff basic web interface

vaidyasr, 2018-09-01:
My 2 sonoff's are running fine on Tasmota latest firmware. But can't access via web interface. I can able to ping to the IP address. What could be the issue?

ascillato, 2018-09-01:
Please, as use the template

ascillato2, 2018-09-01:
Hi, I could not reproduce your issue. Please, erase your flash and recompile and flash.

Frogmore42, 2018-09-01:
Status 0 would at least give more of a hint as to what you are actually using.  I found that unless I completely erased the flash and then reflash do, my devices that were previously using an older version would not work when I updated them using new firmware with core 2.4.2.  The serial port output indicated that they were restarting.  So, that is another clue about what is happening.  What do yours say?

vaidyasr, 2018-09-01:
I have flashed initially using tasmota OTA. Then I upgraded the firmware via web interface and it was successful. I can even control via mqtt using Hass now. But can't login via web interface. To erase and reflash do I need to open and do soldering?

ascillato2, 2018-09-01:
Yes

onkelbeh, 2018-09-02:
No, soldering not necessarily, on a Sonoff Basic Touch or S20 the ports are plated-through, so you can insert the pins of your serial cable in the board and press them a bit sideways with a finger while flashing. Then, dont forget to press the button while you insert the usb plug into your computer. If you have to flash them frequently it could be easier if you solder some pins.

ascillato2, 2018-09-03:
Hi

Have you managed to solve your issue?

vaidyasr, 2018-09-03:
Will be doing today evening.

ascillato2, 2018-09-05:
Hi,

No problem. When you have time to flash your sonoff again, please report it back. Thanks

closed 2018-09-05 by ascillato2