arendst/Tasmota #3663: Boot loop after new flash

Karl-opec, 2018-09-01:
**Describe the bug**
After flashing I get the following output during Boot:

```ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
   
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_176705, GroupTopic sonoffs) Version 6.2.0-2_4_2

Exception (0):
epc1=0x4000dce5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffffd50 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffef0:  00000000 00000000 3fff55e4 4023bd50  
3fffff00:  40104d40 0004f3b7 3fff55fc 00000000  
3fffff10:  3ffef610 3fff55fc 00000014 401004e8  
3fffff20:  00000001 00002f03 3fff0484 40230654  
3fffff30:  4020aa70 00000024 3fff0484 4022b182  
3fffff40:  00000000 4bc6a7f0 00000046 3fff3fc0  
3fffff50:  00000000 4bc6a7f0 00000046 3fff3fc0  
3fffff60:  00000000 00000000 4bc6a7f0 00000000  
3fffff70:  00000000 00000024 4010715a 66a7ef9d  
3fffff80:  00000000 3fff2338 00000000 4020535f  
3fffff90:  00000046 3fff2338 00000000 3fff3fc0  
3fffffa0:  3fffdad0 00000000 3fff2338 4021eb38  
3fffffb0:  3fffdad0 00000000 3fff3fb9 4022f950  
3fffffc0:  feefeffe feefeffe 3ffe8728 40100739  
<<<stack<<<
```

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [* ] _Searched the problem in issues and in the wiki_
- [ *] _Hardware used_ : esp12f 4MB flash

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
It should boot normaly

**Screenshots**

See above

Karl-opec, 2018-09-01:
I have done that several times, either by arduinio ide or with esptool.py, with or without erasing everything before flashing, always the same. 
I used a fresh git clone from today development branch.

During compilation it says: unvalid library esp-epaper-29-ws-20171230-gemu-1.0.
which is the git cloned one. But compilation does work i disabled USE_DISPLAY. Perhaps this is related?

ascillato, 2018-09-01:
So, seems to be a problem with your libraries and arduino. Please, try to flash the compiled versions from release.

Karl-opec, 2018-09-01:
I cloned branch release-6.2 and used the libraries from there and it is working now.

closed 2018-09-01 by Karl-opec