arendst/Tasmota #3664: Additional Wifi parameter "channel"

Pewibe, 2018-09-01:
My problem: 
I run Tasmota devices in a house with 3 access points, all with same SSID. MAC is not convenient to recognize the physical AP, so I use the Wifi channels 1, 7, 13 for this 3 access points to know the used AP.

The solution I like: 
In RSL: STATE = {...,"Wifi":{"AP":1,"SSId":"myNet","RSSI":54,"APMac":"BC:05:43:XX:YY:ZZ"}} an additional parameter like "Channel":13 makes life easier.

Frogmore42, 2018-09-01:
It doesn't appear that the library supports providing channel information.

https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html

It does support it in the scan and providing it for the connect, but it is not in the documentation on how to get which channel it is actually using.

Why do you care which access point it uses?
In NodeRED you can add a translation layer for mac to a more friendly name.  Maybe you could do that with rules too 😀.

mike2nl, 2018-09-01:
@Frogmore42
i found this based on your link:
search for channel in chrome:
-> WiFi.begin(ssid, password, channel, bssid, connect)
--> channel of AP, if we like to operate using specific channel, otherwise this parameter may be omitted

So there is a possibility for that i think. Did not read the whole document
and i have no experience with it, yet...

Frogmore42, 2018-09-01:
As I said:

It does support it in the scan and providing it for the connect, but it is not in the documentation on how to get which channel it is actually using.

mike2nl, 2018-09-01:
@Frogmore42:
Found something about the used channel:
1. https://www.esp8266.com/viewtopic.php?p=47112 = Simply use the WiFi.scanNetworks() functions
2. https://www.esp8266.com/viewtopic.php?f=29&t=12512

Link in 2nd link:
-> https://links2004.github.io/Arduino/df/d1c/class_e_s_p8266_wi_fi_scan_class.html

That must be solve the question, or not? 
So as i said i have no experience with the wifi part, yet.

Pewibe, 2018-09-01:
To answer Frogmore42's question: "Why do you care which access point it uses?"
If everything works fine, I do not care. In reality I have Sonoff Basic devices which have connection problems (I hope V6.2 solves it). In such a case I like to know where device is booked in and view status and logs of the used AP.

Frogmore42, 2018-09-01:
@mike2nl good detective work.  It looks like it is possible to get the channel number even when in station mode and not just from the scan as the other documents seem to imply.  I am not really sure of the value of it, but it seems no less valuable than the BSSID.

mike2nl, 2018-09-01:
@Pewibe 
You are 100% right in that case.
My intention was that there must be a solution and the docu must give help with it.
So as i said i have 10% experinece with esp8266 & wifi and that is nothing equal to you guys.
I'm someone who begin to look deeper than normal when someone tells me it is not in there.
So i was searching and it's not meaned negative. That's how i working my whole live.

BTW:
i have nearly the same question because i use 4 AP's from cisco and linksys.
So idea is not nothing in some cases. 

At least:
I think it must be a personal solution to do so because Theo won't add it to the firmware.
But it's easy to do in the source. Will work on that because it is a 
(in dutch) uitdaging
(german) herausforderung
(english) challenge for me. ;-)

@Frogmore42 
Thanks for your kind words. 
I have an idea to get only the channles from the ssid (ap's) i use for some source code.

ascillato, 2018-09-01:
@mike2nl

Please, if you can, make a PR to show in json status the channel and bssid if possible.
Due to your interest to work on this, I'm adding the label enhancement.
Thanks.

mike2nl, 2018-09-01:
@ascillato
Interested yes, but i'm not the C++ guy. I can read the code 100% and i learn right now with tasmota ;-).
I had made some changes to my code in tasmota and all are working very well. So i have to look first
how deep it goes. Will return with an answer ;-).

andrethomas, 2018-09-01:
It looks like the API supports it

[image]

But this is not implemented in the ESP8266 WiFi Arduino library so that would be a good place to start I guess...  I **guess** it would have to start by getting the maintainers of https://github.com/esp8266/Arduino to add an additional .begin() to be able to specify the channel and BSSID

mike2nl, 2018-09-02:
Good morning all:
so i followed the link given by you @andrethomas .
I found this when i follow the given link about the ESP8266 core for Arduino
link: https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/scan-class.html

Example output:
5 network(s) found
1: Tech_D005107, Ch:6 (-72dBm)
2: HP-Print-A2-Photosmart 7520, Ch:6 (-79dBm)
3: ESP_0B09E3, Ch:9 (-89dBm) open
4: Hack-4-fun-net, Ch:9 (-91dBm)
5: UPC Wi-Free, Ch:11 (-79dBm)
----------------------------------------------------------------------------
So then  i got back to the sonoff.ino where it is included:
line 49: /: #include <ESP8266WiFi.h>         // MQTT, Ota, WifiManager

Then take look into ESP8266WiFi.h and i found this:
line 36: #include "ESP8266WiFiScan.h"

Next take a look ESP8266WiFiScan.h and i found bssid and channel implemented:
Line 43..52:
        // scan result
        bool getNetworkInfo(uint8_t networkItem, String &ssid, uint8_t &encryptionType, int32_t &RSSI, uint8_t* &**BSSID**, int32_t &**channel**, bool &isHidden);

        String SSID(uint8_t networkItem);
        uint8_t encryptionType(uint8_t networkItem);
        int32_t RSSI(uint8_t networkItem);
        uint8_t * BSSID(uint8_t networkItem);
        String BSSIDstr(uint8_t networkItem);
        int32_t channel(uint8_t networkItem);
        bool isHidden(uint8_t networkItem);

That must workso far i can read the code, or i'm totally wrong?

mike2nl, 2018-09-02:
test reply:
-> i got it...

What i have done to reproduce:
Tools: 
- arduino ide 1.8.5 (.6 is out), verify and compile
- syslog server 1.2.3 GNU/GPL License, to get all debug & info
- notepad++ 7.5.8, for searching files

Project: tasmota v6.2.0
File: xdrv_02_webserver.ino
Line: 885

First a screenshot form the syslog server with the original code to show the strengh of the wifi signal:
Figure: 1 - Wifi signal strengh
[image]

Next the source and a screenshot after i changed Line 885 or better added a line:
Line 885 & 886:
//        snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_WIFI D_SSID " %s, " D_RSSI " %d"), WiFi.SSID(indices[i]).c_str(), WiFi.RSSI(indices[i]));
        snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_WIFI D_SSID " %s, " D_RSSI " %d"), WiFi.SSID(indices[i]).c_str(), WiFi.channel(indices[i]));

Then i changed WiFi.RSSI(indices[i])); to WiFi.channel(indices[i])); in Line 889:
Figure: 2 - Wifi channel information
[image]

Anhere the real info in the syslog message:
first signal strengh: sonoff-4344 ESP-WIF: SSId Linksys22267_2GHz, RSSI -44
second channel: sonoff-4344 ESP-WIF: SSId Linksys22267_2GHz, RSSI 6

So i use Channel 6 on my 3rd AP. Next test will be the BSSID and i will report back.

mike2nl, 2018-09-02:
relpy for BSSID:

I used same tools and same tasmota verion.
What i have changed/added in the webserver (short way of working) to get the BSSID information.
I deleted nothing only add a line and comment out the not needed stuff.

Line: 885..887:
//        snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_WIFI D_SSID " %s, " D_RSSI " %d"), WiFi.SSID(indices[i]).c_str(), WiFi.RSSI(indices[i]));
//        snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_WIFI D_SSID " %s, " D_RSSI " %d"), WiFi.SSID(indices[i]).c_str(), WiFi.channel(indices[i]));
        snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_WIFI D_SSID " %s, " D_RSSI " %d"), WiFi.SSID(indices[i]).c_str(), **WiFi.BSSID**(indices[i]));

Screenshot from syslog server with my AP i use for the tests (black marked):
Figure: 1 Wifi BSSID info
[image]

So that means that we have all tools we need.
Ja, yes, jep, now the source code. After i have seen the used macros in C++ i stand in front
of that big wall where i need help to get over it, military like. So as i said i'm an ANSI-C coder 
(one of many languages i work with) but C++ is totally new for me.

To get it done i would take the help or better the coorperation with an experienced 
C++ coder here by tasmota because i can't get it running in short times. When i start
i have to learn and then we are one year further. That would be bad seen the users and
that would not be fair from me.

So which one of you guys will work with me together to get it done, help please...

arendst, 2018-09-02:
So this is what you want:
```
state
12:08:27 CMD: state
12:08:27 SRC: Serial
12:08:27 RSL: Group 0, Index 1, Command STATE, Data 
12:08:27 MQT: stat/ring2/STATE = {"Time":"2018-09-02T12:08:27","Uptime":"0T00:00:38","Vcc":3.026,"POWER":"OFF","Dimmer":2,"Color":"5,0,0","HSBColor":"0,100,2","Channel":[2,0,0],"Scheme":0,"Width":1,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","Channel":6,"RSSI":84,"APMac":"00:22:6B:FE:8E:20"}}
```

arendst, 2018-09-02:
Like this:
```
12:37:52 WIF: Scan done
12:37:52 WIF: SSId indebuurt2, BSSId 00:22:6B:FE:8E:20, Channel 6, RSSI -54
12:37:52 WIF: SSId indebuurt1, BSSId 2C:95:7F:56:93:57, Channel 11, RSSI -80
12:37:52 WIF: SSId indebuurt3, BSSId 60:E3:27:58:77:E6, Channel 1, RSSI -91
```
and

[image]

mike2nl, 2018-09-02:
@arendst
- That's 100% super, Thank You.

but not only for me while Pewibe has startet the posting and i was shure
that it could be done. But that´s much more i expected and it looks like
Pewibe has asked when i understand it in the right way.

And i have seen you have done a commitment to the project
including the language files. There i had started too. Ha, now i can
spare that work You where to fast...
- https://github.com/arendst/Sonoff-Tasmota/commit/a60876378f0adc066352eb978ef07d2d54fc83f7

Now i can read and learn about the code and that is one good thing i can learn today.

mike2nl, 2018-09-02:
@ascillato 
So as you asked:
-> Please, if you can, make a PR to show in json status the channel and bssid if possible.
-> Due to your interest to work on this, I'm adding the label enhancement.

I have done the investigation and some tests to show that it is possible.
@arendst has done the code changings after i asked for help (Thanks again)
and he comitted the new source code.

I have done a new test with tasmota 6.2.0.1 10 minutes ago with a sonoff basic
and here is the status output for the wifi channel.

status 5:
STATUS5 = {"StatusNET":{"Hostname":"sonoff-4344","IPAddress":"192.168.2.101","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1",**"Mac":"84:F3:EB:A7:50:F8",**"Webserver":2,"WifiConfig":5}}

status 11;
STATUS11 = {"StatusSTS":{"Time":"2018-09-02T15:37:41","Uptime":"0T00:19:08","Vcc":3.458,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Linksys22267_2GHz",**"Channel":6,**"RSSI":100,"**APMac":"C0:56:27:61:5E:06"**}}}

We have the MAC address and normally that is enough. When we need the BSSID
i can add the code for a test again.

What do you think? Can we close the thread here?

arendst, 2018-09-02:
The APMac = BSSId

arendst, 2018-09-02:
So this would become:
```
status 11
15:51:53 MQT: stat/ring2/STATUS11 = {"StatusSTS":{"Time":"2018-09-02T15:51:53","Uptime":"0T00:00:11","Vcc":3.007,"POWER":"OFF","Dimmer":2,"Color":"5,0,0","HSBColor":"0,100,2","Channel":[2,0,0],"Scheme":0,"Width":1,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"00:22:6B:FE:8E:20","Channel":6,"RSSI":76}}}
```

arendst, 2018-09-02:
Just did

mike2nl, 2018-09-02:
@arendst
yeah you the hero today ;-) and to fast for me on sundays.
Jesus, i was busy with a change in the language file to show it here. 
Ha, ha, i love that style of working. Fast and precisely.

ascillato2, 2018-09-03:
Closing as feature has been added. Thanks

closed 2018-09-03 by ascillato2

meingraham, 2018-09-03:
Trying to distill this track... in addition to returning Wi-Fi channel information, the JSON payload `"APMac"` is now `"BSSId"` instead? I have a maintenance openHAB Item where I make sure my Sonoff is connected to the right AP by displaying the `"APMac"`. I just want to make sure that this now needs to parse the `"BSSId"` when I upgrade to TASMOTA 6.2. Yes?

ascillato, 2018-09-04:
yes

hamwong, 2018-10-17:
I got a question, I tried look around the issues and I can't found what I look for,
my home have similar case to Pewibe, I got same AP on difference channel, on very few devices, it must only use the neariest AP but somehow it still connect to far one as SSID is same, I don't mind use the individual rule on each sonoff, how can I set the rule if RSSI lower than 60 then force switch to channel X?

maiaz, 2018-12-15:
Just faced this issue. From all available APs it always connected to one with weakest signals.. even with setoption56/57.. I'm not sure, but looks like someone missed sign at RSSI level and soft think that -20 better then -10. Also it possible that device remembers BSSId it first  connected to, and try connect same even if higher level available. Will continue investigate.

arendst, 2018-12-15:
Well there is a threshold of 10 so if one signal is -10 and another is -19 it still stays at -10 to prevent toggling every 44 minutes when it checks for AP signal strength again.

arendst, 2018-12-15:
It also only selects an AP from the one or two configured SSIDs. It never selects an unconfigured AP(SSID)

arendst, 2018-12-15:
And then there is your version of Tasmota. Below version 6.3.0.10 SetOption56/57 do nothing.

andrethomas, 2018-12-18:
@Pewibe The channel number is already displayed!

[image]

andrethomas, 2018-12-18:
@Pewibe PR made: https://github.com/arendst/Sonoff-Tasmota/pull/4667

rgitzel, 2018-12-28:
> The APMac = BSSId
> 
> Perhaps is best I'll change it's name.

D'oh! That explains why my latest Sonoff doesn't show up in my wifi strength graph. The joys of moving up to a new major version... ;-)

[image]

rgitzel, 2018-12-29:
@andrethomas I receive it every minute via `tele/+/STATE` messages (which is where the name was changed), so I don't need to explicitly request it.

meingraham, 2018-12-29:
@andrethomas @rgitzel 6.2.0 is when the parameter changed from being named `APMac` to being named `BSSId`.