arendst/Tasmota #3667: Problem flashing from Arduino IDE

SteveRMann, 2018-09-01:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
status 0
00:01:20 CMD: status 0status 0status 0
00:01:20 RSL: stat/rum/STATUS = {"Status":{"Module":1,"FriendlyName":["Xyzzy Labs"],"Topic":"rum","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":6,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:01:20 RSL: stat/rum/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:01:20","StartupUTC":"","Sleep":0,"BootCount":28,"SaveCount":52,"SaveAddress":"FA000"}}
00:01:20 RSL: stat/rum/STATUS2 = {"StatusFWR":{"Version":"6.1.1.14","BuildDateTime":"2018-09-01T18:14:36","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:01:20 RSL: stat/rum/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Kaywinnet",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
00:01:20 RSL: stat/rum/STATUS4 = {"StatusMEM":{"ProgramSize":477,"Free":524,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FC8E790","00000000","02B400CE","00000000"]}}
00:01:20 RSL: stat/rum/STATUS5 = {"StatusNET":{"Hostname":"qwerty","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:69:2D:B5","Webserver":2,"WifiConfig":5}}
00:01:20 RSL: stat/rum/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.124","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_692DB5","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:20 RSL: stat/rum/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:20 1970","Local":"Thu Jan 01 00:01:20 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":99,"Sunrise":"12:15","Sunset":"21:22"}}
00:01:20 RSL: stat/rum/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:20","Switch1":"ON"}}
00:01:20 RSL: stat/rum/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:20","Uptime":"0T00:01:20","Vcc":3.392,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Kaywinnet","RSSI":100,"APMac":"44:D1:FA:1E:94:3D"}}}
00:01:26 WIF: Connect failed as AP cannot be reached
00:01:26 WIF: Connecting to AP1 Kaywinnet in mode 11N as qwerty...

```
I have a few sonoff Basic switches that I have been flashing with Tasmota 6.1.1a using the Arduino IDE.  The first three went well and all I changed between them is the #define PROJECT line.  When I got to the 4th and 5th Sonoff's, I started getting this error: "Connect failed with AP incorrect password".  I downloaded the latest Tasmota sources and it compiles and flashes OK, but I keep getting the same error. 

I downloaded the latest sonoff.bin file and flashed it with ESPeasy, and everything works fine.  It connects to the AP and I can change the parameters in the console.

But I can't flash and connect from the Arduino IDE.

Any ideas would be appreciated because I don't like it when things stop working.  Plus I want to experiment with the code and I can't do it if the flash via Arduino won't connect to the AP.

Frogmore42, 2018-09-02:
You are using core 2.4.2.  I have found that this can work okay, but you have to do the following.

Erase the flash before flashing the image.
Re-enter the password using the serial port connection.
Don't leave the web UI connected for an extended period of time (many hours).

Before I erased the flash, my device would reboot without connecting.

I have two APs configured.  My device connected fine to the first one, but would not connect to the second one.  After I set WiFi mode to 4, it would attempt, but said the password was wrong.  

So, I updated the password and then it connected.  My password has a space in it, but this has not been an issue in the past.

Since I wanted to use 2.4.2 with lwip 2, I am hoping it fixes the NTP issue I have seen, I stuck with it instead of just going back to 2.3.0.

I am only running 2.4.2 on two devices and only for a week or so, so not a lot of confidence in it yet.

Jason2866, 2018-09-02:
@Frogmore42 
What NTP issues in 2.4.2? I have some test devices rumning fine except sleep...

Frogmore42, 2018-09-02:
The NTP issue is from 2.3.0.  When there is no internet for an extended period of time on boot (like a major power failure, and then backup power comes on, but internet is out for hours) even after internet comes back NTP time sync fails.  The failure is persistent even with a reboot and even a quick power cycle.  It takes a very long power off period to resolve the issue.  The issue is in the underlying NTP code, not Tasmota.  I am hoping it is fixed with lwip2, since NTP is significantly different.

ascillato2, 2018-09-03:
@SteveRMann 

Hi,

Have you managed to solve your issue?

SteveRMann, 2018-09-03:
No, but thanks for asking.  I will close the topic and start over since I can't change the title.

In brief- I have no problem flashing; I have a problem with the Sonoff connecting to the network that just developed after updating the Arduino libraries.

I started with five Sonoff Basic switches and was using the Arduino IDE to flash them with Tasmota.  (Each with a different project name).    The first three flashed and connected to the network just fine.  When the flash finished the LED blinked a few seconds then stopped, at which time I opened the web console to finish the configuration.  I took a break and when I returned, the IDE was prompting me to update the libraries- which I did.  And that's when the problem began.  Now, when I change the project name, the sketch compiles OK and flashes OK- but the Sonoff just won't connect to the network.

Since the original post.
I compiled a sketch that I had been using on the Sonoff's- it compiled and flashed OK and connected to the net as always.

SteveRMann, 2018-09-03:
Close the topic and start over since I can't change the title.

closed 2018-09-03 by SteveRMann