arendst/Tasmota #3670: Grove MGS Multichannel Gas Sensor weired readings

edelstahlratte, 2018-09-02:
I got myself a Grove MGS and connected it to a Sonoff Basic running Tasmota 6.1.1.

It works, but the readings are way off and making no sense at all.
The readings started with VERY hight values and are now dropping over the last 12 hours...

So I took a quick look at the source code  and it seems that the heater element is not in use,
but I'm not a C wizard.. (gas.PowerON seems not to be used)

Is that true? Can the sensor even produce correct results without the heating?
The Mics-6814 documentation states that the correct heating (power) is crucial for the measuring process (and should be powered on at least 1s before taking a measurement). The diagrams all show values with heating element constant on...

Just wondering... Maybe someone can tell me more... Thank you!

mike2nl, 2018-09-02:
first i read this to get an idea about that sensor:
http://wiki.seeedstudio.com/Grove-Multichannel_Gas_Sensor/

I don't if i can help, but i will read ;-).

edelstahlratte, 2018-09-02:
@mike2nl 

Thank you for the fast response :)

In lib\Mutichannel_Gas_Sensor\src\MutichannelGasSensor.cpp:

`void MutichannelGasSensor::powerOn(void)
{
    if(__version == 1)
    sendI2C(0x21);
    else if(__version == 2)
    {
        dta_test[0] = 11;
        dta_test[1] = 1;
        write_i2c(i2cAddress, dta_test, 2);
    }
}`

This functions seems not to be used in Tasmota...

Maybe because the heater is drawing up to 150mA?
Btw, how much current is available on the Vcc line?

It is the MiCS-6814 and the documentation states that continious heating works best, like you mentioned... 

But how to achieve that with a Sonoff Basic???

mike2nl, 2018-09-02:
@edelstahlratte
No issue because i was thinkimg about that sensor type for a friend
who has very strong health issues based on his loungs. So he has to
know the air mix and concentration. All industrial sensors are to pricey
for him so you was faster then me with that sensor test. I have still to order 
that sensor to show him the results. So sonoff does more the on/off.

I have to wait until i get that sensor to test it on my basic or sv model.
Possible someone is faster again ;-). I don't know if you can wait as long 
before i get that sensor.

Possible you can add the gas.poweron in the init call and test it again..?

mike2nl, 2018-09-02:
i have more ideas for the wiki but yeah which one give it free to do so.

edelstahlratte, 2018-09-03:
Will try to evaluate further using Arduino board and direct sensor control with a second sensor to compare the results...

ascillato2, 2018-09-04:
Hi

Have you managed to solve your issue?

edelstahlratte, 2018-09-04:
I'm still waiting for the delivery of the second sensor and the Arduino test equipment...

edelstahlratte, 2018-09-05:
Finally the 2nd sensor and the Arduino bopard arrived...

I did a test and the second sensor shows the same behavior:

The first reading shows immense high values... After that the values drop steadily over time until they stabilize... This can take up to 48 hours! Every power loss starts the process again...

I did not find ou what the function gas.powerOn() really does (theoretically it should activate the heating element on the sensor which is needed for the measurement) but the sensor shows the same readings with and without this function call...

I hope this helps someone.

closed 2018-09-05 by edelstahlratte

mike2nl, 2019-01-09:
@edelstahlratte @ascillato2 
The datasheet says that the sensor needs a long burn in time also after start up.

The heating is used well because the led on that breakout board from groove (i have the same one)
- It is flashing 3 or 4 times at startup. 
- When the web ui is open the led flashes in the refresh cycle of the web ui.
- It flashes when the tele is happening based on the teleperiod (case FUNC_PREP_BEFORE_TELEPERIOD)

So the heating is done very well.