arendst/Tasmota #3673: RF Bridge - Dooya DC90 remote - howto program bridge

jk2020, 2018-09-02:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : 433RFBRIDGE R2 V1.0
- [x] Provide the output of command ``status 0`` :
```
11:37:56 CMD: status 0
11:37:56 SRC: WebConsole from 192.168.2.3
11:37:56 RSL: Group 0, Index 1, Command STATUS, Data 0
11:37:56 MQT: stat/sonoff-bridge/STATUS = {"Status":{"Module":25,"FriendlyName":["sonoff-bridge"],"Topic":"sonoff-bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:37:56 MQT: stat/sonoff-bridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:21","StartupUTC":"2018-09-02T10:37:35","Sleep":0,"BootCount":59,"SaveCount":100,"SaveAddress":"F4000"}}
11:37:56 MQT: stat/sonoff-bridge/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-08-31T15:15:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:37:56 MQT: stat/sonoff-bridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Free VirusFI",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
11:37:56 MQT: stat/sonoff-bridge/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
11:37:56 MQT: stat/sonoff-bridge/STATUS5 = {"StatusNET":{"Hostname":"sonoff-bridge-3976","IPAddress":"192.168.2.62","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:A3:6F:88","Webserver":2,"WifiConfig":3}}
11:37:56 MQT: stat/sonoff-bridge/STATUS6 = {"StatusMQT":{"MqttHost":"nodered.jfr.de","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A36F88","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:37:56 MQT: stat/sonoff-bridge/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 02 10:37:56 2018","Local":"Sun Sep 02 11:37:56 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:08","Sunset":"19:31"}}
11:37:56 MQT: stat/sonoff-bridge/STATUS10 = {"StatusSNS":{"Time":"2018-09-02T11:37:56"}}
11:37:56 MQT: stat/sonoff-bridge/STATUS11 = {"StatusSTS":{"Time":"2018-09-02T11:37:56","Uptime":"0T00:00:21","Vcc":3.170,"Wifi":{"AP":1,"SSId":"Free VirusFI","RSSI":62,"APMac":"80:2A:A8:CA:7E:9B"}}}

```
I have a DC90 Dooya Remote Control
https://de.aliexpress.com/item/Original-Dooya-DC90-Single-Channel-Transmitter-for-Dooya-Motors-RF-433MHZ-Remote-Control-Dooya-Automatic-Curtain/32811888324.html
for a rollershutter. It is a simple three button remote UP/STOP/DOWN

Sonoff Rf Bridge flashed with Tasmota V6.2.0 and flashed Portisch RF-Bridge-EFM8BB1-20180816.hex 

sonoff rf bridge is up and running successfully

Using RfRaw 166

11:42:51 CMD: rfraw 166
11:42:51 SRC: WebConsole from 192.168.2.3
11:42:51 RSL: Group 0, Index 1, Command RFRAW, Data 166
11:42:51 MQT: stat/sonoff-bridge/RESULT = {"RfRaw":"ON"}
11:42:51 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}

Does NOT sniff any button presses - NO success!

Using RfRaw 177
11:53:37 CMD: rfraw 177
11:53:37 SRC: WebConsole from 192.168.2.3
11:53:37 RSL: Group 0, Index 1, Command RFRAW, Data 177
11:53:37 MQT: stat/sonoff-bridge/RESULT = {"RfRaw":"ON"}
11:53:37 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}

Press UP
11:54:04 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA B1 08 1270 0631 02A3 0160 01B3 034E 0122 1E46 0123322323233223323232323223323223322323232323232332233223323232233232322323232337 55"}}

Press Stop (several times)
nothing happens

Press DOWN
11:55:06 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 129D 060E 02BA 0167 1E3C 0123322323233223323232323223323223322323232323232332233223323232233232232332322323012332232323322332233232232323233234 55"}}

Press Stop (several times)
nothing happens

Press UP
11:56:20 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 12A2 05FF 02CA 014B 1E32 01233223232332233232323232233232233223232323232323322332233232322332323223323232230123322323233223323232323223323223322323232323232332233223323232233232322332323224 55"}}

Press Stop (several times)
nothing happens

Press UP
11:57:53 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 05F0 02CE 016C 12AC 0122112121221122121212121122121122112121212121212211221122121211221212112212121123 55"}}

Press DOWN
11:58:20 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 1298 0604 02CA 0156 1E3C 0123322323233223323232323223323223322323232323232332233223323232233232232323233234 55"}}

Press UP
11:59:20 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA B1 07 064A 02C7 0110 01C8 038B 0093 127A 0132113121221122121212121122121122112121212121212211221122121211221212112212121136 55"}}

Press DOWN
12:00:25 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 129A 0604 02D1 014F 1E46 0123322323233223323232323223323223322323232323232332233223323232233232232323233234 55"}}

Press UP
12:01:21 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 129F 0601 02B5 0161 1E3C 0123322323233223323232323223323223322323232323232332233223323232233232322323232334 55"}}

When presseing STOP key, the red led on sonoff bridge is blinking
Why can I not see the code while pressing STOP?

How can I program Keys to use UP/DOWN?

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-09-02:
Latest portisch firmware? May this helps:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/23

jk2020, 2018-09-02:
@Jason2866 many thanks, the link definetly helps to understand this better.

My current challenge is that RfRaw 166 does not give me any data when I press the buttons

17:24:40 RSL: Received Topic /rfraw, Data Size 6, Data aaa655
17:24:40 RSL: Group 0, Index 1, Command RFRAW, Data aaa655
17:24:40 MQT: stat/sonoff-bridge/RESULT = {"RfRaw":"ON"}

Re-flashed with latest hex file found under:
https://github.com/Portisch/RF-Bridge-EFM8BB1
but same result, e.g. no succes with rfraw 166

Any special command you need to type in beforehand?

rfraw 177 works but it looks like I need the data captured with rfraw 166

Jason2866, 2018-09-02:
No rfraw 166 and rfraw 177 are different sniff commands see
https://github.com/Portisch/RF-Bridge-EFM8BB1/

Jason2866, 2018-09-03:
Make a issue / question on portisch github. Portisch is the expert!

ascillato2, 2018-09-03:
Issue continue at https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/29

closed 2018-09-03 by ascillato2