arendst/Tasmota #3674: Wrong layout for sonoff basic

mkh595, 2018-09-02:
**Describe the bug**
I upgraded Sonoffs basic from version 5.11 to 5.14 and to 6.2.0 via OTA successfully.
When I flashed version 6.2.0 directly and customized pins, layout changed

[image]

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff basic
- [X] _Provide the output of command_``status 0`` :
```
17:06:46 CMD: Status 0
17:06:46 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff","sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:06:46 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffxxx","OtaUrl":"http://192.168.1.15:80/arduino/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:38","StartupUTC":"2018-09-02T15:06:08","Sleep":0,"BootCount":3,"SaveCount":7,"SaveAddress":"F5000"}}
17:06:46 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-09-02T17:00:47","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:06:46 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"192.168.1.15","LogPort":514,"SSId":["mkh","mkh-samsung"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
17:06:46 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":447,"Free":556,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F582794","24000001","230217A6","00000000"]}}
17:06:46 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7489","IPAddress":"192.168.1.236","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.25.1","Mac":"60:01:94:68:DD:41","Webserver":2,"WifiConfig":5}}
17:06:46 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.14","MqttPort":1883,"MqttClientMask":"ESP_%06X","MqttClient":"ESP_68DD41","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:06:46 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 02 15:06:46 2018","Local":"Sun Sep 02 17:06:46 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:03","Sunset":"19:28"}}
17:06:46 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-02T17:06:46","Switch2":"ON"}}
17:06:46 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-02T17:06:46","Uptime":"0T00:00:38","Vcc":3.062,"POWER1":"OFF","POWER2":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":62,"APMac":"2C:4D:54:83:7F:68"}}}
```

**To Reproduce**
See screenshots

**Expected behavior**

**Screenshots**

After initial flash of version 6.2.0 is layout correct

[image]

Left side is device upgrade from version 5.14, right side flashed directly version 6.2.0

[image]

Same pins are set on the right side

[image]

After save

[image]

I uploade via file back version 6.1.1 and set pins to 00

[image]

layout is OK

[image]

Same pins are set on the right side

[image]

layout wrong also in version 6.1.1

[image]

I returned back to version 5.11.1a, set same pins on right side

[image]

and now is layout OK

[image]

**Additional context**

Could be this behaviour caused by fact that version 5.11.1 has different number for Button1 and PWM1?

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-09-02:
As you said the latest version shuffled some GPIO around.

There is an issue here that explains it too. Can't find it just now.

Doesn't matter. The addition of more switches has shuffled some GPIOs around.

ascillato2, 2018-09-03:
Hi,

Thanks for share a detailed information of this issue. We need to add this information to the wiki as a breaking change from old versions to last.

closed 2018-09-03 by ascillato2

mkh595, 2018-09-03:
I don't think that it is breaking change but it is bug as this happen only if PWM1-5 or PWM1i-5i is selected.

ascillato2, 2018-09-07:
Sorry. Breaking Change means a change in the code that it is backward incompatible. So, those configurations were not ported properly from old version to the new version due to changes in the GPIO definition.

ascillato2, 2018-09-07:
If you need more information, please follow the links. Thanks

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.