arendst/Tasmota #3675: How to separate switch and actor for using rules?

renne, 2018-09-02:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Hardware used : H801
- [x] Provide the output of command ``status 0`` :
```
21:58:23 MQT: stat/Wohnzimmer/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"Wohnzimmer","ButtonTopic":"","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:58:23 MQT: stat/Wohnzimmer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"H801","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:27:59","StartupUTC":"2018-09-02T19:30:24","Sleep":0,"BootCount":31,"SaveCount":598,"SaveAddress":"F6000"}}
21:58:23 MQT: stat/Wohnzimmer/STATUS2 = {"StatusFWR":{"Version":"6.1.1.14","BuildDateTime":"2018.09.02 18:09:32","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
21:58:23 MQT: stat/Wohnzimmer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXX",""],"TelePeriod":30,"SetOption":["00108009","55818000","00000000"]}}
21:58:23 MQT: stat/Wohnzimmer/STATUS4 = {"StatusMEM":{"ProgramSize":468,"Free":532,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0F000794","00000000","40000496","00000002"]}}
21:58:23 MQT: stat/Wohnzimmer/STATUS5 = {"StatusNET":{"Hostname":"Wohnzimmer-5475","IPAddress":"192.168.100.47","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"60:01:94:6F:95:63","Webserver":2,"WifiConfig":5}}
21:58:23 MQT: stat/Wohnzimmer/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6F9563","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:58:23 MQT: stat/Wohnzimmer/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 02 19:58:23 2018","Local":"Sun Sep 02 21:58:23 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:44","Sunset":"20:10"}}
21:58:23 MQT: stat/Wohnzimmer/STATUS10 = {"StatusSNS":{"Time":"2018-09-02T21:58:23","Switch1":"ON","BME280":{"Temperature":23.3,"Humidity":44.0,"Pressure":1009.0,"SeaPressure":1025.1},"MPR121A":{"Button0":0,"Button1":0,"Button2":0,"Button3":0,"Button4":0,"Button5":0,"Button6":0,"Button7":0,"Button8":0,"Button9":0,"Button10":0,"Button11":0,"Button12":0},"TempUnit":"C"}}
21:58:23 MQT: stat/Wohnzimmer/STATUS11 = {"StatusSTS":{"Time":"2018-09-02T21:58:23","Uptime":"0T00:27:59","Vcc":3.481,"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"XXX","RSSI":74,"APMac":"xx:xx:xx:xx:xx:xx"}}}
```

Hi,

I'm trying to link `Switch1` (RCWL-0516 motion detector on GPIO0, `SwitchMode1 1`) with `POWER1` via rules. If I use `SwitchTopic1 0` or `SwitchTopic1 1` the light is always powered off when the RCWL-0516 goes low after 2 seconds. If `SwitchTopic1 xxx` is used I can't get a rule to trigger.

How do I have to configure the H801 to link `Switch1` with `Power1` via rules without internal connections interfering?

Frogmore42, 2018-09-02:
I have not tried rules yet, but you probably want switch mode 4, so it activates on motion.  You need switch topic to be something other than 0 or 1, since they will control the relay either directly or indirectly, respectively.

I would suggest NOT using GPIO 0 for the RCWL-0516, since it might trigger bootloader mode on power up and you probably want a physical button on GPIO to enable bootloader mode manually.  You can also define the pin that the rcwl is using as Switch2 and define another unused pin as Relay2 and then you rule should work.  If it also switches Relay2 it won't matter as nothing is connected to it.

ascillato2, 2018-09-03:
Please, read the wiki at rules examples: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#15-rule-to-enable-a-pir-sensor-only-at-night

closed 2018-09-03 by ascillato2

renne, 2018-09-03:
@Frogmore42 
`Switchmode1 4` with `SwitchTopic1 0`or  `SwitchTopic1 1` toggles the light every time someone moves. If I use `SwitchTopic MOTION` rules don't trigger. The RCWL-0516 doesn't have adjustable delay. It switches on for 2 seconds when movement is detected. So I need something like the [10. Simple Staircase Timer](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#10-simple-staircase-timer) without switching off by the button (= RCWL-0516).

The [H801](https://github.com/arendst/Sonoff-Tasmota/wiki/H801) has only GPIO0, GPIO2 and GPIO3 on solder-pads. As GPIO2 and GPIO3 are used for I2C only GPIO0 is available for the RCWL-0516. If I want to flash the [H801](https://github.com/arendst/Sonoff-Tasmota/wiki/H801) I have to pull out the cable and connect a USB-UART adapter.

Using `Switch1`  with `SwitchTopic1 0` and `Switchmode1 1` on `GPIO0`, dummy `Relay 1` on `GPIO16` and the rule `{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":413,"Rules":"on switch1#state=1 do backlog power2 on; ruletimer1 300 endon on rules#timer=1 do power2 off endon"}` works but is quite complicated and costs a GPIO.

@ascillato2 
[15. Rule to enable a PIR Sensor only at night](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#15-rule-to-enable-a-pir-sensor-only-at-night) does not work as the light is switched off by `Switch1` with `SwitchTopic1 0`or  `SwitchTopic1 1`. `SwitchTopic MOTION` doesn't trigger rules (e.g. event cmnd/motion/power1 = on).

Is there any way to trigger a rule by a MQTT message (e.g. `SwitchTopic motion`)?

Frogmore42, 2018-09-03:
I would try this:
'rule1 on button1#state do backlog power1 on; ruletimer1 600 endon on rules#timer=1 do power1 off endon

Configure the GPIO as a button not a switch.  You can use GPIO 2 as  button.  But it might work okay with GPIO 0 too.

renne, 2018-09-03:
@Frogmore42 
After resetting the configuration of the H801 completely the rule `rule1 on button1#state do backlog power1 on; ruletimer1 600 endon on rules#timer=1 do power1 off endon` works. The [button usage functionalities](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage) of `Button1` cause the H801 to reset after some time. Changing `GPIO0` and the rule to `Button2` doesn't work. What is the difference in configuration between `Button1` and `Button2`?

@benzino77 
`Switch1` works with `SwitchMode1 4` after configuration reset, too.

ilya89vet, 2018-12-23:
I have a Wemos D1 module and 4 relays are connected to it on pins GPIO04, GPIO05, GPIO12, GPIO13 
The RCWL-0516 sensor is connected to GPIO14
On my idea in the absence of the movement during certain time the 2nd relay has to be switched off.
But the problem is that switch 5...8 -> on/off relay 1

ilya89vet, 2018-12-23:
Whether it is possible to remove somehow a binding of switch 5...8 from relay 1
Following rule:
rule1 on switch5#state do ruletimer1 1800 endon on rules#timer=1 do power2 off endon on time#minute=120 do power2 off endon

And I want so:
rule1 on power2#state=1 do rule2 1 endon on power2#state=0 do rule2 0 endon
rule2 on switch5#state do ruletimer1 1800 endon on rules#timer=1 do power2 off endon on time#minute=120 do power2 off endon

benzino77, 2018-12-23:
What is your switchtopic[12345]?