arendst/Tasmota #3677: Openhab Tasmota SR04 live update via MQTT?

hectorasoto, 2018-09-03:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Hardware used : 
- [ x] Provide the output of command ``status 0`` :
```

Hello 

How do a get the value form this sensor SR04 live via MQTT

Im current using the statement below but I receive the update every 10 seconds because I change the Teleperiod to 10 

---Openhab Item---
Number HCSR04 "Tank [%.1f cm] " {mqtt="<[broker:tele/waterwemos/SENSOR:state:JSONPATH($.SR04.Distance)]"}

Hardware Used:
Wemos D1
SR04 Sensor 
Openhab

```

**(Please, remember to close the issue when the problem has been addressed)**

hectorasoto, 2018-09-03:
Garage door closes when the car arrive

ascillato, 2018-09-03:
Inside Tasmota is real time, so you can make a rule for making a mqtt message when your sensor detects a car. You don't need to flush your network with a lot of unused mqtt messages.

ascillato, 2018-09-03:
Please, read the wiki for examples

ascillato2, 2018-09-04:
Hi

Have you managed to solve your issue?

hectorasoto, 2018-09-04:
I check the wiki but he teleperiod can be change form 10 seconds to 3600 seconds 
I current have a SR04 sensor on the Watertank and the sensor send me teledata every 10 seconds that is fine for the watertank, but i was trying to create a another sensor using the SR04 to display live Distance

ascillato2, 2018-09-04:
Have you tried rules?

ascillato2, 2018-09-04:
Sorry, Rules are triggers for defining actions to a sonoff device. Not for what you want.

You can try to make your home automation software to ask tasmota status.

Or also, you can send to Tasmota the command cmnd/yoursonofftopic/teleperiod 10 and will reply the actual value of sensors. If you ask once per second should work.

hectorasoto, 2018-09-04:
thanks  understand this trick 

i will close the the issue thanks a lot

closed 2018-09-04 by hectorasoto