arendst/Tasmota #3678: Openhab 6.20 on sonoff basic do not connect wifi

spad, 2018-09-03:
Hi,
compiling and uploading 6.20 on Sonoff Basic, do not connect wifi network (11N).
The log reports: incorrect password, which is correct (setup on code).

Same setting, re-uploading 5.13.1 works correct.

Jason2866, 2018-09-03:
Without further information we cant provide help.
Please post `Status 0` output.

Jason2866, 2018-09-03:
User_Config.h has changed since 5.xx. Latest used?

Frogmore42, 2018-09-03:
If you compiled it, you are probably using core 2.4.2.  Status 0 will indicate for sure.

I found that I had to reset the password using the serial connection to get it to connect.

ascillato2, 2018-09-03:
Have you tried the precompiled bin from release? May be you are having a bad compilation issue that leads to having wifi issues, please, try to erase all flash, and then flash last release. If you are building by yourself, please use esplib 2.3.0 (if you will use sleep feature) or 2.4.2 lwIP1.4 if you are not going to use sleep.

mike2nl, 2018-09-03:
Have you cleaned the flash?
Including wifi, when you use aeduino IDE...
Or have you used esptool?

spad, 2018-09-04:
Hi All,
I'm waiting for a new sonoff basic and I will try. The current one is used
and I cannot test it.
I compile it using  platform = espressif8266 , I'll try with
espressif8266@1.6.0.

How can I erase the flash?
Thanks.

ascillato2, 2018-09-04:
`esptool.py --port COM5 erase_flash`

Prutsium, 2018-09-05:
Had a similar issue with a Sonoff Basic Wifi.
Running rock solid on 6.1.1 but after upgrading to 6.2.x cant reach the unit anymore.

Updated the unit via the Webinterface (yes it had been flashed with DIO before) and after upgrade device comes up but when i go to it's website it hangs directly so impossible to retrieve a status 0.

Downgraded to 6.1.1 and unit works rock solid again.

ascillato, 2018-09-05:
@Prutsium 

Hi, seems that you OTA has failed. Have you tried to erase all flash and then flash latest release? 

Latest release is working fine, so it is not a problem with the code.

@spad 

Have you tried this solution?

arendst, 2018-09-05:
@Prutsium did you use:
1) 6.2.0/6.2.1 from the github release
2) a development version from http://thehackbox.org/tasmota/
3) compiled yourself

Prutsium, 2018-09-05:
I used the development version 6.2.1 as the 6.2.1 was not (yet) on Github.
Flashed this over the Webinterface with Minimal (6.2.1) in between.

arendst, 2018-09-05:
OK. Now try to use the Github release version v6.2.1.

If that still fails, erase flash and retry to flash the github v6.2.1 release.

Prutsium, 2018-09-05:
Tried the Git version first without erase and result non working.
Cleaned the device with esptool erase and also flashed a 1mb blanco.

Flashed Git version and device comes up but no network visible after 4 clicks.
Serial log:
`00:00:13 RSL: stat/sonoff/RESULT = {"WifiConfig":"WifiManager selected"}
                                                                        00:00:15 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff35c8
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_86A617, GroupTopic sonoffs) Version 6.2.1-2_3_0
                           00:00:00 WIF: WifiManager active for 3 minutes
                                                                         00:00:00 HTP: Web server active on sonoff-1559 with IP address 192.168.4.1
`

arendst, 2018-09-05:
What tool do you use to connect to sonoff-1559?

Prutsium, 2018-09-05:
For flashing ESPtool.py (On Windows) and for logging Putty on the serial port.
(There is no plugin active requiring the Serial port as within 6.1.1 i see all log data passing.)

arendst, 2018-09-05:
But how do you connect to the device when it is in AP mode? Using a phone, tablet or laptop?

Prutsium, 2018-09-05:
AP Mode (4 clicks) with a laptop.

The same laptop i use when i take the device on 6.1.1 and click 4 times i see directly it's network.

The only difference i notice:
In 6.2.1 it reboots directly so likewise it comes never in AP mode where on 6.1.1 it does not need to reboot to get in AP mode.

arendst, 2018-09-05:
Depending on version and firmware selection the default wifi config is different.

Before 6.2.x it would always be WPS.
Since 6.2.x only sonoff-classic will use WPS. All other selections will only use WifiManager.

In my tests it works as expected.

As you have putty connected why don't you use commands to configure ssid and password? No need for any wifi manager...

Prutsium, 2018-09-05:
As somehow i cant send any command (only receive data).

Anyway 6.2.1 wont connect to access point what is configured before:
[code]
00:00:00 Project sonoff Michelle_Aquarium (Topic Aquarium, Fallback DVES_86A617, GroupTopic sonoffs) Version 6.2.1-2_3_0
                                        00:00:00 WIF: Connecting to AP1 404Domotics in mode 11N as Michelle_Aquarium...
                                       00:00:07 WIF: Connect failed as AP cannot be reached
           00:00:07 WIF: Connecting to AP1 404Domotics in mode 11N as Michelle_Aquarium...
          00:00:14 WIF: Connect failed as AP cannot be reached
                                                              00:00:14 WIF: WifiManager active for 3 minutes
                            00:00:14 HTP: Web server active on Michelle_Aquarium with IP address 192.168.4.1
[/code]

Prutsium, 2018-09-05:
And then the webserver isnt visible.

ascillato2, 2018-09-05:
So, but the v6.1.1 works for you?

I could not reproduce this problem, sorry.

If you have already erased the flash and then used the precompiled bins and you can not see the wifi AP made by your sonoff, I don't know what is happening in your case. Sorry.

I have done exactly that and works in my devices.

razer4908, 2018-09-05:
I have exactly the same problem as @Prutsium

fillwe, 2018-09-05:
Having the same problems on a couple of Sonoff Basics, gonna try erasing flash. Saying that its setting upp wifi manager after failing to connect but nothing happening

Jason2866, 2018-09-05:
On new sonoff (stock firmware on it)  do always before flashing TASMOTA a whole erase.
If you have problems with self compiled firmware. Do a whole flash erease! Flash a ready compiled bin from release path from here.

osvaldoasn, 2018-09-05:
Same problem here after update to 6.2.1(OTA) and upload by serial and does not work.

KPK75, 2018-09-06:
Just for reference : 

1) one of mine ESP-01M's won't connect to any password protected WiFi ( even using simple arduino sketch ) but no problems with open WiFis. And flashes from arduino without any problems, AP mode works fine.
2) When I used flash download tool 3.4.4 from AI-Thiker I got EFUSE error so I think it's bricked.

But it was working fine earlier so it got bricked somehow during tasmota tests.

KPK75, 2018-09-06:
I should said not bricked but damaged. And you can damage EEPROM :) I used DOUT.
Tasmota firmware works OK via serial but funny thing - it just wont't connect to any password proteced WiFi. I created open WiFi for tests and it's working fine.

arendst, 2018-09-06:
@KPK75 try a pre-compiled RELEASED 6.2.1 version and see if that works.

KPK75, 2018-09-06:
I did :) works fine except WPA2. Maybe others have similar problems so I shared my findings.

'0:05:16 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}'
'10:05:16 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:26:55","StartupUTC":"2018-09-06T07:38:21","Sleep":0,"BootCount":7,"SaveCount":11,"SaveAddress":"F9000"}}'
'10:05:16 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-05T10:40:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}'
'10:05:16 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DI","xxxxx "],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}'
'10:05:16 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}'
'10:05:16 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5781","IPAddress":"172.17.4.117","Gateway":"172.17.4.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"DC:4F:22:C5:B6:95","Webserver":2,"WifiConfig":5}}'

ascillato, 2018-09-06:
Hi to all,

If you are having wifi issues, please:
* erase all flash using `esptool.py --port COM5 erase_flash`
* and flash again by cable using the precompiled bins from release

If you want to compile by yourself please, remember that platformio sometimes makes a bin with a misaligment that give as result a firmware with wifi issues. In that case, just recompile and reflash again. The source code in this repository is working fine (v6.2.1.2). Remember to use esplib v2.3.0 if you need sleep command or esplib v2.4.2 if you don't.

If you did a OTA and it fails and you can not reconnect, you will need to flash again by wire.

In the case you still having problems, please open a new troubleshooting issue with the template complete, the logs, screenshots, the symptoms you found and all the information you can share.

GHPS, 2018-09-07:
I can confirm the version 6.2.1 still doesn't work with the Sonoff Basic. As in version 6.1.1 the wifi connection to the router and the direct connection to the device fail. I tried both the OTA and the
direct flash update (with erasing the flash first). Interestingly enough the minimal version 6.2.1 works fine.

I had to return to 5.14...

raymajor, 2018-09-07:
Same problem with V 6.2.1 and Sonoff Basic
Update by Web interface, update by USB to Serial.
First update to Sonoff-minimal. That's works fine, than to Sonoff-NL. Device wil not start, only the LED is flahing and no connection with Wifi manager.
Flash it back to V6.1.1 end everything works fine.

ascillato, 2018-09-07:
Hi,

We are trying to find why you are having this problem. Please, test the precompiled bins from development at http://thehackbox.org/tasmota/ and report back if that works for you.

Thanks

onkelbeh, 2018-09-08:
Hi @ascillato,

me again (case #3658) :-)
after getting my Basics to work with the precompiled binary from release folder i first believed it was my build environment's fault. But then: yesterday i got some Sonoff 4CH showing the same symptoms. But something was different, these guys refused to connect to wifi even with the release binary which made the Basics work. So i made some other tests. Had a couple of older acces points here, and i found out that it depends very on the accesspoint the have to connect to. My first tests were all made eith an ASUS RT-AC66U (wifi is a Broadcom BCM4360). Then i tried a Mikrotik (Qualcomm QCA9531), which cased a bit more trouble. Even the Basics with the release binary refused to. All tasmotas with firmware < 6.2 worked with it.

Next: AVM Fritz 7390 (imho QCA9558, Atheros?): All (test-)devices **work immediately properly**, even my self compiled ones (current dev tree, v6.2.1.2). For today my plan is to check if this depends on the router firmware, will try it with freetz. next to upgrade my productive Sonoffs one by one and move them to the Fritz.  Will keep you informed.

raymajor, 2018-09-08:
Hi ascillato,

I flasht it with http://thehackbox.org/tasmota/sonoff.bin
After rebooting, LED flashing. Trying to connect with WIfimanager. (4 short presses)
I'll see the Sonoff-4085 can connect to it.
Go to Safari, Ip adress: 192.168.4.1 and i'm in. (or it's goes automatic)
I'll put in the SSId and the password. after an reboot the sonoff will change to ESP_066FF5
Save everything but can't connect to it. (I'll have the MAC adress set to the IP adress 192.168.10.55)
But LED is still flashing!!
After a reboot, ESP_066FF5 will come in to Wifimanager modus...
Again SSId and Password....
Yes it's work.. Sonoff-Tasmota 6.2.1.2
Change configuration for Domoticz, Save it and reboot.
Then he goes to Wifimanager mode (Access point), Sonof 4085... LED Flashing

I'll hope this wil help you

Cheers, Ray

RIKWEBB, 2018-09-08:
I too can confirm that my Sonoff basic's have worked with 6.2.1 classic and minimal, but will not connect to Wifi with the standard 6.2.1 sonoff.bin. 
I have now performed an OTA from 6.1.1 to 6.2.1.2 from http://thehackbox.org/tasmota/sonoff.bin and its working fine.

Rik

Prutsium, 2018-09-08:
Here running an Asus Ac68 and as described before had issues with the previous versions.
Flashed now 6.2.1.3 via OTA and all works ok.

Did you guy's find any issue?

raymajor, 2018-09-08:
I'll try also OTA to http://thehackbox.org/tasmota/sonoff.bin and it's work also fine...
6.1.1 to 6.2.1.3

ascillato, 2018-09-08:
Great!

Thanks a lot to everyone for the testing and the information. Super useful. :+1: 

So, we can confirm that all devices work fine with the precompiled BINs from http://thehackbox.org/tasmota 

And that the **sonoff.bin** from release [v6.2.1](https://github.com/arendst/Sonoff-Tasmota/releases/tag/v6.2.1) is having wifi issues on devices with esp8266 1M. This was due to a misaligment problem in platformio compilation.

Please, avoid **sonoff.bin** file from release page v6.2.1 and use the precompiled BINs from http://thehackbox.org/tasmota until Theo reupload them or make a new release.

Thanks again and sorry for the inconvenience. Platformio's misalignment problem is randomly affecting projects based on esp8266.

closed 2018-09-08 by ascillato2

arendst, 2018-09-09:
Released a newly compiled **sonoff.bin** just now.

fotiDim, 2018-09-09:
@arendst where do we download that from? Under the releases pages it says it was released 4 days ago.

arendst, 2018-09-09:
And when you scroll down you'll see a date telling you that today one of the binaries has changed....

raymajor, 2018-09-09:
@arendst Sorry, its not working Theo.
Sonoff basic. goes into Wifimanager modus. LED is flashing.
I did a update by file upload.

razer4908, 2018-09-09:
not working for me either. Updated via new provided file via FTDI & via OTA. LED is blinking. No Wifi Response

RIKWEBB, 2018-09-09:
New sonoff.bin file not working for me either. No Wifi again. Flashed back to 6.2.1.3 and all is well.

arendst, 2018-09-09:
Hold on. I'll try @wasi54 option....

arendst, 2018-09-09:
According to the Xtensa docs target-align is the default and indeed if I compile with option no-target-align the resulting binary is 1k smaller due to un-alignment.

Conclusion is that this option is needed on Linux but not needed on windows. I'll add it anyway to the repository.

So I'm still fighting over the fact that the latest compiled image is not solving the problems mentioned earlier.

Will continue to investigate.

raymajor, 2018-09-09:
Good luck, You doing a great Job!!

raymajor, 2018-09-09:
Flash it back to 6.1.1 USB/Serial
OTA  https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.2.1/sonoff.bin
Stays @ 6.1.1 nothing happened.

flash to 6.2.1 minimum by file upload. Then flash it to Sonoff.bin by file upload.
LED is flashing and WIFI manager modus.

arendst, 2018-09-09:
First, you cannot OTA a Github release as Github only supports HTTPS and OTA only supports HTTP. That's why Maddox provides the release binaries on HTTP. I guess they won't be available within an hour after I release them on Github.

Second, I don't have a clue.....

raymajor, 2018-09-09:
Hi Theo, If you want we can make a Teamviewer session so you can look?
With the Https i'll no that was my mistake...

arendst, 2018-09-09:
I have teamviewer started. What next?

raymajor, 2018-09-09:
891 621 526
6644

raymajor, 2018-09-09:
Theo Good Luck, I'll hope i'll help you with it....

arendst, 2018-09-09:
@raymajor I managed to get a sonoff.bin image working. Could you be so kind to download the latest sonoff.bin from the release page and give it a try?

arendst, 2018-09-09:
Great it works. Problem is that it's still a miracle as it only seems to work if I disable the LM75AD sensor which is still enabled in 6.2.1.3

Need further investigation but for now I call it a day.

raymajor, 2018-09-09:
You do a great job.. When you need to test let me hear it!
Thank you! 
i'll hope more guys test is just before...

tobox, 2018-09-09:
Thank you so much for tracking down this nasty bug. I was already getting frustrated with Tasmota because none of my flashed devices was running properly. I would never have guessed that the problem was caused by compiling on linux. 

Maybe we should try to get reproducible builds to find bugs like this one more easily?

raymajor, 2018-09-09:
Also "update" from V6.2.1.3 (thehackbox) to V6.2.1(Github) works fine!
@ascillato Close ticket #3678 ?

raymajor, 2018-09-09:
@arendst Sonoff-NL.bin works NOT!!
I'ts comming into accespoint mode....

[14 more replies]