arendst/Tasmota #3679: Disable/Turn Off Wifi LED

benmprojects, 2018-09-03:
**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
No, The flashing LED is rather annoying
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
An option to disable the Wifi LED from the web dashboard
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Tape over the LED :)
**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

wasi54, 2018-09-03:
See https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
LedPower | 0 / off | Turn led AND LedState Off
Please close the Issue if your Problem ist solved

Frogmore42, 2018-09-03:
You can send that command from the web UI.

ascillato2, 2018-09-03:
Closing as your question has been addressed. Please, ask to reopen if needed.

closed 2018-09-03 by ascillato2

vel2000, 2019-01-21:
>  LedPower | 0 / off | Turn led AND LedState Off

does NOT solve the problem!
The wifi LED is still flashing, if there is no Wifi-connection.
"LedState 0" does not help either...

andreas-bulling, 2019-01-25:
Same here. Also `LEDPower 0; SetOption31 1` doesn't help.

Maybe https://github.com/arendst/Sonoff-Tasmota/issues/488 explains why?

andreas-bulling, 2019-01-26:
Didn't help in my case unfortunately :(

john-arvid, 2019-02-11:
Same problem here.
@ascillato2 Could you open this again or should it be created a new issue?

ascillato, 2019-02-11:
Hi, please address your issue in the Tasmota Support Chat

topperwh, 2019-02-14:
1) go to your Sonoff IP Address
2) click "console"
3) LedPower 0  (Copy, paste & no semicolon) 

--- That MQTT protocol will stop the LED blink light.

john-arvid, 2019-02-14:
> 1. go to your Sonoff IP Address
> 2. click "console"
> 3. LedPower 0     (Copy, paste & no semicolon) press enter
> 4. LedState 0      press enter
> 
> --- That MQTT protocol will stop the LED blink light.

Does not work on the board I am using. D1 Mini

I asked on Tasmota Support Chat and the only suggestion was to remove the led. (Or cut the trace)

8osman, 2019-02-14:
Led Power 0

Then

SetOption31 1

From firmware 6.2.0

Worked on mine.

8osman, 2019-02-14:
This was from the wiki commands

john-arvid, 2019-02-14:
Not working for me, using firmware 6.4.1(sensors)

john-arvid, 2019-02-15:
I have tested a bit more and found that when I was using GPIO2 with my AM2301 sensor the led blinked no matter what commands I used.
But when I set the configuration on GPIO2 to none, the led stopped blinking.
Also when I use GPIO4 with the AM2301 sensor, the led is not blinking.
Even if I am sending MQTT messages every 10 seconds there is no blinking from the led.

Is this a hardware or software problem?

arendst, 2019-02-15:
Your D1 mini led is hardwired to GPIO02. So anything you connect to GPIO02 will also blink the Led. The chat also told you that the led is hardwired. So...

john-arvid, 2019-02-15:
Ok, I understand.
Hope this can be of help to others also. Try the commands or another gpio.

84wb, 2019-06-25:
> Hope this can be of help to others also.
It was. Thanks!

HenniC4, 2019-07-01:
That blinking seems to be Tasmota desperately crying for its MQTT Server. Setoption3 0 did the trick for me.

JamFfm, 2021-01-25:
MQTT seems to have impact on the green LED. Turning MQTT off results in LED on all the time.

LedPower 0
AND
Setoption3 0
in the console did the trick for me. 

Thanks all for your help.

dachshund-digital, 2021-05-15:
LedPower 0 and Setoption3 0 did not work for me on firmware 9.4.  Any suggestions?  Trying to explicitly control the onboard led.

MikeTraceur, 2024-04-07:
I could help myself going into the template and set the pin containing Led* to None. I know it is quick and dirty