arendst/Tasmota #3680: Wifi connectivity

jpadie, 2018-09-03:
Hi

is there a threshold below which the wifi will not connect?  I have about fifteen tasmota-sonoff devices in the house; all flashed to the latest release on v2.3.0.  They all work fine save one.  

In one location (about 20m from the house, clear line of sight) there is one device that won't connect.  The interface says 36% RSSI.  I've tried a number of different laptops there, and phones.  They all connect just fine.   But this one doesn't.  I've tried two other sonoff basics from the same batch and they all fail to connect.   I'm aware of the issues with v2.4+, hence flashing on 2.3.0.  

is there a threshold below which tasmota or the wifi lib just doesn't bother finalising the connect process (the earlier logs I have on the router (I stupidly deleted them) show the tasmota devices starting to handshake then just going silent).

mike2nl, 2018-09-03:
@jpadie
Hi, yes there is a minimum value but don't ask which value that is.

Adviese, When all other connections are ok:
- Take (20meter sonoff) device near the AP for a test.

Can you do that and please..

Frogmore42, 2018-09-03:
I deployed a couple of devices to a new location.  While they both worked fine on my bench neither one worked at the intended location.  Here is what I did to figure it out.

One device is a Wemos D1 mini light, so it was easy to connect to my PC and see the logs.  Once I changed the WiFi configuration so that it would switch to the other network, it would at least try and connect.  The reported signal was around 30-ish%.  It said it had a bad password, which seemed weird, since I had used the same configuration before.  I then tried changing the password using the serial console.  When I did that, it tried to connect.  Sometimes it did connect, but mostly it didn't work right.

I then remembered that my previous device there, using a nano router running as a client, had a difficult time connecting to the network there.  It would typically connect, but might take 35-40 seconds to get an IP address, instead of the few seconds on the bench.  So, I had a spare router.  I configured it as an access point and replaced the old one.  With some adjustment to placement and antenna orientation, I was able to get the signal up to 50-ish%.  With that change, it has been very reliable.

jpadie, 2018-09-03:
Good info 

I can't move the device easily to test.  The location is fixed.  Sometimes it connects to a mobile access point and sometimes it doesn't (I add the mobile ap as a second ssid in the config for just this kind of issue). 

Until it does choose to connect I can't report back on the status.  However the only useful data in the status will be there firmware version which is the version available from the device branch this morning and compiled on board version 2.3.0.  

The essential question I'm asking really is not how to remedy this situation. But what the rssi threshold is coded at.  Or the timeout threshold for connection.  If it is too low then I will need to move to alternative radios (rfm69) for data transfer.  Annoying to be using a mixed environment but I don't have the time until November to do any material troubleshooting of WiFi boxes.

Frogmore42, 2018-09-03:
You will have to ask that question at the espressif forum as they are probably the only ones who will know the answer in any detail.

The simple answer is that you have already proven that the WiFi is not good enough at the location where you have your device.  You have some choices to make.

jpadie, 2018-09-03:
> Hi,
> You said in the first post that this issue is with just one device. May be that device has a hardware defect and it shows signal but may be the noise do not allows the connection. I have one device with that issue but it is hardware not software. Try to change you sonoff for another one.
> Tasmota do not have any threshold configuration. That is managed by the wifi libraries and the SDK that are outside the control and the scope of this project.

It is with one device in my network.  As in at that location and with the functionality.   I have swapped in a couple of sonoffs to test.   All the same.  It is a very difficult location to service though.  So I would rather drop the rssi threshold for connection and use UDP for comms than swap out again and/or install a different radio altogether. Thanks for the last information.  I will review the WiFi library.  I had felt sure that someone competent here would have already done so.

ascillato2, 2018-09-03:
You are welcome

closed 2018-09-03 by ascillato2