arendst/Tasmota #3681: TSL2561 sensor vanishes from Web and MQTT when 0 lux are reached

edelstahlratte, 2018-09-03:
**Describe the bug**
The TSL2561 sensor vanishes from Web UI and MQTT when 0 lux are reached.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ :  Sonoff Basic
- [x] _Provide the output of command_``status 0`` :

MQTT and status 0 with light:

12:56:21 MQT: tele/sensor_og/SENSOR = {"Time":"2018-09-03T12:56:20","Switch1":"OFF","BME280":{"Temperature":26.5,"Humidity":46.2,"Pressure":1013.6},"TSL2561":{"Illuminance":142.432},"TempUnit":"C"}

13:00:40 MQT: stat/sensor_og/STATUS = {"Status":{"Module":1,"FriendlyName":["Sensor_OG"],"Topic":"sensor_og","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:00:40 MQT: stat/sensor_og/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:32","StartupUTC":"2018-09-03T11:00:08","Sleep":0,"BootCount":14,"SaveCount":47,"SaveAddress":"FA000"}}
13:00:40 MQT: stat/sensor_og/STATUS2 = {"StatusFWR":{"Version":"6.2.0.1","BuildDateTime":"2018-09-02T19:07:38","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:00:40 MQT: stat/sensor_og/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EasyBox-72EC66-2.4","EasyBox-72EC67-2.4"],"TelePeriod":300,"SetOption":["000A8009","55818000","00000000"]}}
13:00:41 MQT: stat/sensor_og/STATUS4 = {"StatusMEM":{"ProgramSize":427,"Free":576,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0D082794","240003A1","014814C6","00000000"]}}
13:00:41 MQT: stat/sensor_og/STATUS5 = {"StatusNET":{"Hostname":"sensor_og-4898","IPAddress":"192.168.1.174","Gateway":"192.168.1.128","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.128","Mac":"68:C6:3A:A3:53:22","Webserver":2,"WifiConfig":5}}
13:00:41 MQT: stat/sensor_og/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.199","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A35322","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:00:41 MQT: stat/sensor_og/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 03 11:00:41 2018","Local":"Mon Sep 03 13:00:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
13:00:41 MQT: stat/sensor_og/STATUS10 = {"StatusSNS":{"Time":"2018-09-03T13:00:41","Switch1":"OFF","BME280":{"Temperature":26.5,"Humidity":46.0,"Pressure":1013.6},"TSL2561":{"Illuminance":144.400},"TempUnit":"C"}}
13:00:41 MQT: stat/sensor_og/STATUS11 = {"StatusSTS":{"Time":"2018-09-03T13:00:41","Uptime":"0T00:00:33","Vcc":3.157,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EasyBox-72EC66-2.4","BSSId":"E0:28:6D:CC:3F:9D","Channel":6,"RSSI":76}}}

MQTT and status 0 without light:

13:49:47 MQT: tele/sensor_og/SENSOR = {"Time":"2018-09-03T13:49:42","Switch1":"OFF","BME280":{"Temperature":28.0,"Humidity":43.7,"Pressure":1013.2},"TempUnit":"C"}

13:51:28 MQT: stat/sensor_og/STATUS = {"Status":{"Module":1,"FriendlyName":["Sensor_OG"],"Topic":"sensor_og","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:51:28 MQT: stat/sensor_og/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:51:20","StartupUTC":"2018-09-03T11:00:08","Sleep":0,"BootCount":14,"SaveCount":47,"SaveAddress":"FA000"}}
13:51:28 MQT: stat/sensor_og/STATUS2 = {"StatusFWR":{"Version":"6.2.0.1","BuildDateTime":"2018-09-02T19:07:38","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:51:28 MQT: stat/sensor_og/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EasyBox-72EC66-2.4","EasyBox-72EC67-2.4"],"TelePeriod":300,"SetOption":["000A8009","55818000","00000000"]}}
13:51:28 MQT: stat/sensor_og/STATUS4 = {"StatusMEM":{"ProgramSize":427,"Free":576,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0D082794","240003A1","014814C6","00000000"]}}
13:51:28 MQT: stat/sensor_og/STATUS5 = {"StatusNET":{"Hostname":"sensor_og-4898","IPAddress":"192.168.1.174","Gateway":"192.168.1.128","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.128","Mac":"68:C6:3A:A3:53:22","Webserver":2,"WifiConfig":5}}
13:51:28 MQT: stat/sensor_og/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.199","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A35322","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:51:28 MQT: stat/sensor_og/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 03 11:51:28 2018","Local":"Mon Sep 03 13:51:28 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
13:51:33 MQT: stat/sensor_og/STATUS10 = {"StatusSNS":{"Time":"2018-09-03T13:51:28","Switch1":"OFF","BME280":{"Temperature":28.0,"Humidity":43.6,"Pressure":1013.1},"TempUnit":"C"}}
13:51:33 MQT: stat/sensor_og/STATUS11 = {"StatusSTS":{"Time":"2018-09-03T13:51:33","Uptime":"0T00:51:25","Vcc":3.157,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EasyBox-72EC66-2.4","BSSId":"E0:28:6D:CC:3F:9D","Channel":6,"RSSI":80}}}

**To Reproduce**
Cover the sensor until completely dark and undo to see reaction in WEB UI...

**Expected behavior**
Show 0 lux in Web UI and MQTT

**Screenshots**

With light:

[image]

Without light:

[image]

**Additional context**

Problem has been observed in 6.1 and 6.2.
Since 6.2 all sensors vanish...

**(Please, remember to close the issue when the problem has been addressed)**

edelstahlratte, 2018-09-03:
Maybe it is related to #2415 ??? Seems about the same... But changes had been made and added... So maybe the TSL2561 really is kind of a diva... I have several sensors running without any problems... Up to 4 on an I2C Bus, but the TSLs (I have 5 of them, 2 different suppliers) always cause problems...

edelstahlratte, 2018-09-05:
Seems that I am alone with this problem?

ascillato, 2018-09-05:
@edelstahlratte 

When you post this issue, we add the label Bug to work on it.

Please, be patient or try to find a solution and propose a Pull Request.

edelstahlratte, 2018-09-05:
@ascillato

Hello @ascillato!

I literally meant what I wrote! My post was not about impatience...

I just wanted to know if someone else is experiencing the same behaviour...

KPK75, 2018-09-07:
Yes. Confirmed.
Status 10 reports "Illuminance":0.028" but it takes bit longer to display result comparing to daylight.
And it doesn't happen with BH1750 light sensor so maybe bug in the library ? Timeout ?

edelstahlratte, 2018-09-07:
@KPK75

BH1750 works fine for me as well... No problems at all.

What happens if you cover your sensor complete?

edelstahlratte, 2018-09-07:
@KPK75

Okay, so this confirmes my observation...

Staars, 2018-09-15:
Can You try the following?:
In /lib/Joba_Tsl2561-2.0.7/src/Tsl2561Util.cpp change in line 191 and 205 "return false;" to "return true;". This would be more similar to the data sheet of the chip on page 25, where a ratio of the channels above 1.3 is 0 Lux but not defined as "false".

edelstahlratte, 2018-09-17:
Mhh, I tried, but with the same result... Value is not shown in WebUI...

Staars, 2018-09-17:
Okay, that is not what I wanted to hear :grin:.

In /sonoff/xsns_16_tsl2561.ino thera are 6 chances to fail from line 59 - 64 and to prevent the showing of the regarding WebUI/TELE-message.

By best guess is, that one of the functions in line 62-64 returns a "false"  when lux is 0 and prevents line 70 and later.

You could try to isolate/nest these in an extra "if-branch" and "else" set "millilux" to 0, but leave the code beginning with line 70 outside.

edelstahlratte, 2018-09-17:
I don't get it...

TELE is fine... But WebUI fails...

If one of the statements in line 62-64 fails, there should be no TELE as well...

`tele/sonoff/SENSOR = {"Time":"2018-09-17T19:47:54","Switch1":"ON","TSL2561":{"Illuminance":0.456}}`

[image]

Staars, 2018-09-17:
Yes, both should fail or work  ... at least I can not explain it.
I‘m sorry, but without a sensor I do not have an additional idea (beside the usual stuff like serial.print everything and temporarily disable some conditions) at the moment.

KPK75, 2018-09-18:
Javascript timeout on web page ?

petez69, 2018-09-22:
I've just compiled 6.2.1.4 

#define USE_TSL2561

stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x39"}

status 10
08:48:48 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-22T08:48:48"}}

No sign of the sensor on the web page nor on the console.....

petez69, 2018-09-22:
OK, weird.  When the MQTT server was not defined, the sensor did not show on the web page....When I defined the MQTT server, 

stat/sonoff-development/STATUS10 = {"StatusSNS":{"Time":"2018-09-22T08:54:03","TSL2561":{"Illuminance":178.617}}}

petez69, 2018-09-22:
Theo

Can confirm the sensor was correctly detected with this last fix (without mqtt server defined).

Thankyou

closed 2018-09-22 by arendst