arendst/Tasmota #3683: Function Timers via webpage don't send mqtt

witek1308, 2018-09-03:
Hi,
I needed new feature to globally arm/disarm timers (Timers ON/OFF) and I needed to do this via mqtt. Initially I thought it is problem with my mqtt but then I realised that each time when I send change of state via command (e.g. browser cm?cmnd=Timer ON/OFF) it gets published via mqtt is my server captures it. When I do it however via website (tick the box and press Save) it doesn't get to mqtt, neither console. 

BTW (not sure if it is intended or side effect but very nice - although documentation says Timers 0/1 it works also when used Timers OFF/ON.
Not sending any logs yet as I believe issue is clearly described and when requested I'll support it with pointed logs

ascillato, 2018-09-03:
Your can use all Tasmota commands by mqtt. Cmnd/yoursonofftopic/timers 1

Please, read the wiki for more infomation

witek1308, 2018-09-03:
hi ascillato2,
yes this I know and this is what I'm doing (sending from OpenHab2 commands via mqtt and receiving them back). Where I see issue is that changing timers setting (tickbox) from website doesn't send mqtt. This is causing a desync issue that someone changes setting via website and central portal which relies on mqtt doesn't see it. Normally I was used to the fact that any change via website (e.g. relay toggle) was resulting in mqtt and therefore whichever method I used to toggle (website, mqtt, etc) was keeping the whole environment in sync.

arendst, 2018-09-03:
I understand the issue and will evaluate.

Yes, when 0/1/2 can be entered you can also enter No/Yes/Toggle

arendst, 2018-09-03:
Evaluated and sorry but most settings changed via the web interface are not being signaled via MQTT as there are often many settings changed at once what would lead in massive amount of MQTT messages.

So you just have to learn not to use the web interface for make important changes ;-)

witek1308, 2018-09-03:
easy to say - I'm married :) but I'll explain somehow... ;)

Jason2866, 2018-09-03:
@witek1308 
The hardcore way -> disable web interface

closed 2018-09-03 by ascillato2