arendst/Tasmota #3685: Wrong sunrise and sunset

mkh595, 2018-09-03:
I set LATITUDE and LONGTITUDE for my location, APP_TIMEZONE=99. Time is displayed correctly, but  sunrise and sunset is shifted +1 hour.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff basic
- [X] _Provide the output of command_``status 0`` :
```
19:18:39 MQT: stat/home238/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 03 17:18:39 2018","Local":"Mon Sep 03 19:18:39 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:10","Sunset":"20:29"}}
```

**Expected behavior**
Correct sunrise should be 6:10 and correct sunset 19:29, like return from http://api.wunderground.com for my location
```
{
  "response": {
  "version":"0.1",
  "termsofService":"http://www.wunderground.com/weather/api/d/terms.html",
  "features": {
  "astronomy": 1
  }
	}
		,	"moon_phase": {
		"percentIlluminated":"43",
		"ageOfMoon":"23",
		"phaseofMoon":"Last Quarter",
		"hemisphere":"North",
		"current_time": {
		"hour":"19",
		"minute":"20"
		},
		"sunrise": {
		"hour":"6",
		"minute":"05"
		},
		"sunset": {
		"hour":"19",
		"minute":"26"
		},
		"moonrise": {
		"hour":"23",
		"minute":"42"
		},
		"moonset": {
		"hour":"14",
		"minute":"32"
		}
	},
	"sun_phase": {
		"sunrise": {
		"hour":"6",
		"minute":"05"
		},
		"sunset": {
		"hour":"19",
		"minute":"26"
		}
	}
```
**Screenshots**

[image]

Jason2866, 2018-09-03:
What is your LATITUDE and LONGTITUDE?

ascillato, 2018-09-03:
Please, provide us the output of the `status 0` command. We need all your information in order to reproduce what are you claiming.

mkh595, 2018-09-03:
Device was updated via OTA and therefore original Longtitude and Latitude was not updated (CFG_HOLDER not changed) to my location which I updated for version 6.2.0

```
21:18:06 CMD: Longitude
21:18:06 MQT: stat/home238/RESULT = {"Longitude":"2.294442"}
21:18:12 CMD: latitude
21:18:12 MQT: stat/home238/RESULT = {"Latitude":"48.858360"}
```
I tested with device flashed from scratch and Sunrise and Sunset is correct.

closed 2018-09-03 by mkh595

honggiwei, 2018-09-07:
I setting timezone is 8
Then I am also have this bug
[image]

arendst, 2018-09-07:
@honggiwei if you do not set your latitude and longitude correctly do not expect valid results.

honggiwei, 2018-09-07:
Dear arendst:
   Thanks. It is correct after setting  latitude and longitude.
I forget sonoff have no GPS module.
I think another way,  if the latitude and longitude information can be fetched by the world-wide Ethernet network information ?

pauldogg, 2020-11-14:
I have set Longitude.
I have set Latitude.
I have set Timezone 99
I have setup TimeDST
I have setup TimeSTD

Sunset/sunrise information is incorrect.

Please advise.

Is there a way to omit the need for GPS info? Seems unnecessary...

angelgz123, 2021-01-21:
The correct sunset and sunrize depend on 3 parameters.
At the TASMOTA CONSOLE enter:
latitude XX.XXXX
longitude YY.YYYY
timezone +/- X

wuala... it works charm.

ricardok-sp, 2022-02-19:
Q11

> We