arendst/Tasmota #3686: Username size is too short for Microsoft Azure IoT Hub

InsomniumBR, 2018-09-03:
First thanks for the great job.
I am trying to use the MQTT feature with Microsoft Azure IoT Hub that has very specific configurations as described here: https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support#using-the-mqtt-protocol-directly

I am not having success on sending MQTT packages and I think that the problem is that the username and password are being truncated by the configurations. I took a look at your documentation and you say that the username and password both can have 32 chars, but the IoT hub uses a very large username and password...

for instance:
username: "iothub-brw.azure-devices.net/TH10_01/api-version=2016-11-14"
password: "SharedAccessSignature sr=iothub-brw.azure-devices.net%2Fdevices%2FTH10_01&sig=2############################MDYeLIaiQcso%3D&se=1599999991"

Is it possible to set directly by a command a bigger username/password?
I can see both truncated on this message when using miniterm.py.

[image]

I am using a TH10 device, latest firmware (sonoff.bin).

Thanks for your help.

InsomniumBR, 2018-09-04:
> Hi,
>
> So as to increase the 32 characters limit for mqtt username/password, you
> need to change:
>
> File: *settings.h*
> Line 209:char mqtt_user[150];
> Line 210:char mqtt_pwd[150];
>
> But this will scramble ALL the internal saved settings.
>
> So, you will need to erase all flash and then compile your modified
> Tasmota version and flash to your device.
>
> Sorry, but I don't think this change will be implemented in Tasmota due to
> backward incompatibility and too much use of memory.
>
> Hope this information help you to achieve what you want.

Adrian,

Thanks for the directions.
Ok, not a problem, I will compile my modified version.

Thanks a lot.

InsomniumBR, 2018-09-04:
@ascillato2 I have changed the header accordingly:

```
  char          mqtt_host[33];             // 1E9
  uint16_t      mqtt_port;                 // 20A
  char          mqtt_client[33];           // 20C
  char          mqtt_user[80];             // 22D
  char          mqtt_pwd[150];              // 24E
  char          mqtt_topic[33];            // 26F
  char          button_topic[33];          // 290
  char          mqtt_grptopic[33];         // 2B1
```

But, I am still having problems with the password, it is truncating to 99 chars. =/
The username now its OK.

ascillato, 2018-09-04:
So, memory is not enough. Sorry

ksaye, 2021-04-17:
@arendst @TBraune I am trying the same thing -- to get it to talk to Azure IoT Hub using an SAS token.  When I make one simple change 
``
char mqtt_pwd[150]; // 24E
``
in the settings.ino, it restarts when I try set it up for the first time -- connecting it to the network 'iotdevices'.

[image]

While that change is just defining the size of a variable, there seems to be something else causing an issue.  I have tried this with the master branch and the v9.3.1 branch, which has some of the fixes required but still has has the limited mqtt_user and mqtt_pwd set to char[33].  Any help would be appreciated, as we are trying to use the platform, but use it talking to IoT Hub.

arendst, 2021-04-17:
@ksaye you're a bit late to this party! There is a reason why ``char ex_mqtt_pwd[33];           // 24E`` is called `` ex_mqtt_pwd``.

Things have changed massively since 2018. In theory you can have any length you want as long as you stay within the character pool of 699 characters. The ratio is displayed with command ``status 2`` at the end of the message.

Your issue is likely caused by internal char arrays handling the "longer" strings. What is the max string length you need for one or both these parameters?

ksaye, 2021-04-17:
Story of my life, always late.  :)  I did not notice that the ``ex_`` meant no longer used.

So I guess (based on https://tasmota.github.io/docs/TLS/) I can just set the "#define USE_MQTT_TLS" in user_config_override.h and compile [Azure requires TLS] and it should work.

Thanks for the rapid response.

ksaye, 2021-04-17:
@arendst Thanks for the direction and support.  I was able to get it to connect with the following settings.  Posting info here for others who may have the same issue.

Step 1: In user_config_override.h, define: ``#define USE_MQTT_TLS``, compile and flash to the device with new bin and perform normal setup.
Step 2: Create a device in your Azure IoT Hub, my device was named ``tasmota`` and generate a SAS token for the device.
Step 3: Configure the MQTT Parameters, simular to what is shown below, replacing your hostname, device id and SAS token.

Setting | Example Value
--- | ---
 Hostname: | kevinsay.azure-devices.net
 Port: | 8883
 Use TLS: | checked
 Client: | tasmota
 User: | kevinsay.azure-devices.net/tasmota/?api-version=2018-06-30
 Password: | SharedAccessSignature sr=kevinsay.azure-devices.net%2Fdevices%2Ftasmota&sig=8nXrEAH*********ES6M%3D&se=1650146410
 Topic: | anything
 Full Topic: | devices/tasmota/messages/events/

[image]

As you can see below, it is connected and sending messages:

[image]

There is an issue of 'flattening' the topics, but I will post a separate issue.