arendst/Tasmota #3687: Rule switch state domoticz publish 6.2

Dickesplumpel, 2018-09-04:
Maybe I oversee something so I report it not as a bug:

Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":438,"Rules":"on Switch2#state do publish domoticz/in {"idx":61,"nvalue":%value%} endon

works fine on 6.0.0a but on 6.1.1.13 and 6.2 nothing is published - SwitchTopic=0

ascillato2, 2018-09-04:
Hi,

You did not complete the troubleshooting template. Please, complete it in order to help you properly.

Dickesplumpel, 2018-09-04:
Hi! 
With Firmware 6.1.13 and 6.2.0  the sonoff doesn't publish the command 
RUL: SWITCH2#STATE performs "publish domoticz/in {"idx":61,"nvalue":1}"
  to domoticz. when switch2 (Motion sensor) changes state.

ascillato, 2018-09-04:
Hi,

You did not complete the troubleshooting template. Please, complete it in order to help you properly.

That give us a lot of information in order to help you. Do you have check that you have a switch2 defined in the module configuration when you are using v6.2.0 ?

Dickesplumpel, 2018-09-04:
Hi!

Yes. Switch2 is defined switch2 idx is set to 61 and the motion sensor (switch2) switches POWER1, the same when I define relay2 on an unused gpio POWER1 is switched and no rule is performed.

It is an ESP8266 Wifi RGB light bulb  (Ailight) on which I have plugged an HC-SR501 at TX (gpio1)

stat/Lampe1/STATUS = {"Status":{"Module":27,"FriendlyName":["Sonoff"],"Topic":"Lampe1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
tele/Lampe1/STATE = {"Time":"2018-09-04T17:33:20","Uptime":"0T09:41:58","Vcc":3.178,"POWER1":"OFF","Dimmer":90,"Color":"00E51F00","HSBColor":"128,100,90","Channel":[0,90,12,0],"Scheme":3,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"arminseins","RSSI":54,"APMac":"28:C6:8E:B3:C8:9C"}}
tele/Lampe1/SENSOR = {"Time":"2018-09-04T17:33:20","Switch2":"OFF"}

ascillato, 2018-09-04:
Please, provide the status 0. There is no password nor sensible information there. Just the config and features enabled.

Dickesplumpel, 2018-09-04:
18:31:38 CMD: status 0
18:31:38 MQT: stat/Lampe1/STATUS = {"Status":{"Module":27,"FriendlyName":["Sonoff"],"Topic":"Lampe1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:31:38 MQT: stat/Lampe1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:20","StartupUTC":"2018-09-04T16:31:18","Sleep":1,"BootCount":141,"SaveCount":4432,"SaveAddress":"FB000"}}
18:31:38 MQT: stat/Lampe1/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018.09.02 07:59:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:31:38 MQT: stat/Lampe1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["arminseins",""],"TelePeriod":300,"SetOption":["04008009","55818000","00000001"]}}
18:31:38 MQT: stat/Lampe1/STATUS4 = {"StatusMEM":{"ProgramSize":456,"Free":544,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0F50A794","24000000","020B979E","00000000"]}}
18:31:38 MQT: stat/Lampe1/STATUS5 = {"StatusNET":{"Hostname":"Lampe1-1681","IPAddress":"192.168.1.11","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:BD:A6:91","Webserver":2,"WifiConfig":5}}
18:31:38 MQT: stat/Lampe1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BDA691","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:31:38 MQT: stat/Lampe1/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 04 16:31:38 2018","Local":"Tue Sep 04 18:31:38 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:11","Sunset":"20:27"}}
18:31:38 MQT: stat/Lampe1/STATUS10 = {"StatusSNS":{"Time":"2018-09-04T18:31:38","Switch2":"OFF"}}
18:31:38 MQT: stat/Lampe1/STATUS11 = {"StatusSTS":{"Time":"2018-09-04T18:31:38","Uptime":"0T00:00:20","Vcc":3.167,"POWER1":"OFF","Dimmer":90,"Color":"00E51F00","HSBColor":"128,100,90","Channel":[0,90,12,0],"Scheme":3,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"arminseins","RSSI":58,"APMac":"28:C6:8E:B3:C8:9C"}}}

Dickesplumpel, 2018-09-04:
I have flashed with 6.2 for status 0 and now it woks as it should - don't know why - it looks as if the things have a will of their own ;-))

Thank you very much!

ascillato, 2018-09-04:
Hi,

From your `status 0`:
```
{"Status":{"Module":27,"FriendlyName":["Sonoff"],"Topic":"Lampe1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:20","StartupUTC":"2018-09-04T16:31:18","Sleep":1,"BootCount":141,"SaveCount":4432,"SaveAddress":"FB000"},"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018.09.02 07:59:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["arminseins",""],"TelePeriod":300,"SetOption":["04008009","55818000","00000001"]},"StatusMEM":{"ProgramSize":456,"Free":544,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0F50A794","24000000","020B979E","00000000"]},"StatusNET":{"Hostname":"Lampe1-1681","IPAddress":"192.168.1.11","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:BD:A6:91","Webserver":2,"WifiConfig":5},"StatusMQT":{"MqttHost":"192.168.178.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BDA691","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Tue Sep 04 16:31:38 2018","Local":"Tue Sep 04 18:31:38 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:11","Sunset":"20:27"},"StatusSNS":{"Time":"2018-09-04T18:31:38","Switch2":"OFF"},"StatusSTS":{"Time":"2018-09-04T18:31:38","Uptime":"0T00:00:20","Vcc":3.167,"POWER1":"OFF","Dimmer":90,"Color":"00E51F00","HSBColor":"128,100,90","Channel":[0,90,12,0],"Scheme":3,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"arminseins","RSSI":58,"APMac":"28:C6:8E:B3:C8:9C"}}}
```
Theo's _decode-status.py_ gives the output:
```
*** decode-status.py v20180730 by Theo Arends ***
Decoding information for device Sonoff from status report taken at 2018-09-04T18:31:38

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT enabled
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) MQTT serial binary
  24 (OFF) Rules once mode until 5.14.0b
  25 (OFF) KNX enabled
  26 (ON ) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led
  50 (ON ) Timers enabled

Features
  Language LCID = 1031
  MQTT_HOST_DISCOVERY
  MQTT_PUBSUBCLIENT
  PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
  USE_ADC_VCC
  USE_ADS1115
  USE_BH1750
  USE_BMP
  USE_DHT
  USE_DISCOVERY
  USE_DOMOTICZ
  USE_DS18B20
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_HOME_ASSISTANT
  USE_HTU
  USE_I2C
  USE_INA219
  USE_PZEM004T
  USE_RULES
  USE_SHT
  USE_SHT3X
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_TSL2561
  USE_WEBSERVER
  USE_WS2812
  VTABLES_IN_FLASH
  WEBSERVER_ADVERTISE
```

So, Domoticz is enabled.

Please, use the command weblog4, then press your switch and paste your console output.

Dickesplumpel, 2018-09-05:
CMD: weblog4
MQT: stat/Lampe1/RESULT = {"WebLog":2}
MQT: domoticz/in = {"idx":60,"nvalue":1,"svalue":"90","Battery":56,"RSSI":6}
MQT: stat/Lampe1/RESULT = {"POWER1":"ON"}
MQT: stat/Lampe1/POWER1 = ON
MQT: domoticz/in = {"idx":60,"nvalue":0,"svalue":"90","Battery":56,"RSSI":6}
MQT: stat/Lampe1/RESULT = {"POWER1":"OFF"}
MQT: stat/Lampe1/POWER1 = OFF

ascillato, 2018-09-05:
hi,

Sorry

It is `weblog 4`
with a space

Dickesplumpel, 2018-09-05:
Hi - no Problem. 
The whole domoticz/out traffic  is shown, I think you are looking for this:

20:24:08 RSL: empfangenes topic domoticz/out, Datengröße 261, Daten {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "0001408E",
   "idx" : 62,
   "name" : "Dummy Ailight1",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}

20:24:08 DOM: idx 62, nvalue 0
20:24:09 SRC: MQTT
20:24:09 RSL: empfangenes topic domoticz/out, Datengröße 272, Daten {
   "Battery" : 255,
   "Level" : 90,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "0001408C",
   "idx" : 60,
   "name" : "AiLight1",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "Dimmer",
   "unit" : 1
}

20:24:09 DOM: idx 60, nvalue 0

Light bulb is only 3 seconds on (hold time motion sensor) when motion detected.

With correct working firmware 6.0.0a (motion sensor triggers domoticz dummy switch) the output is:

20:35:10 DOM: idx 61, nvalue 1
20:35:11 SRC: MQTT
20:35:11 RSL: empfangenes topic domoticz/out, Datengröße 273, Daten {
   "Battery" : 255,
   "Level" : 90,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "0001408C",
   "idx" : 60,
   "name" : "AiLight1",
   "nvalue" : 2,
   "stype" : "Switch",
   "svalue1" : "90",
   "switchType" : "Dimmer",
   "unit" : 1
}

20:35:11 DOM: idx 60, nvalue 2

ascillato2, 2018-09-05:
Please, can you post the output of `gpio` command?

Dickesplumpel, 2018-09-06:
Same for 6.2 and 6.0 :
 stat/Lampe1/RESULT = {"GPIO1":"10 (Switch 2)","GPIO2":"0 (None)","GPIO3":"0 (None)"}

ascillato2, 2018-09-06:
Ok,

All seems ok.

Have you tried another rule like `rule on switch2#state do power1 2 endon` to see if rules are being triggered?

I tested your rule with last version and works fine.

ascillato2, 2018-09-06:
Ok, let's see what we are missing in your config. I'm gathering all the information you gave in the troubleshooting template. Please, check if all the following is correct.

> `Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":438,"Rules":"on Switch2#state do publish domoticz/in {"idx":61,"nvalue":%value%} endon`
> 
> works fine on 6.0.0a but on 6.1.1.13 and 6.2 nothing is published - SwitchTopic=0

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Hardware used : **AILIGHT**
- [X] Provide the output of command ``status 0`` :
```
18:31:38 CMD: status 0
18:31:38 MQT: stat/Lampe1/STATUS = {"Status":{"Module":27,"FriendlyName":["Sonoff"],"Topic":"Lampe1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:31:38 MQT: stat/Lampe1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:20","StartupUTC":"2018-09-04T16:31:18","Sleep":1,"BootCount":141,"SaveCount":4432,"SaveAddress":"FB000"}}
18:31:38 MQT: stat/Lampe1/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018.09.02 07:59:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:31:38 MQT: stat/Lampe1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["arminseins",""],"TelePeriod":300,"SetOption":["04008009","55818000","00000001"]}}
18:31:38 MQT: stat/Lampe1/STATUS4 = {"StatusMEM":{"ProgramSize":456,"Free":544,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0F50A794","24000000","020B979E","00000000"]}}
18:31:38 MQT: stat/Lampe1/STATUS5 = {"StatusNET":{"Hostname":"Lampe1-1681","IPAddress":"192.168.1.11","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:BD:A6:91","Webserver":2,"WifiConfig":5}}
18:31:38 MQT: stat/Lampe1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BDA691","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:31:38 MQT: stat/Lampe1/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 04 16:31:38 2018","Local":"Tue Sep 04 18:31:38 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:11","Sunset":"20:27"}}
18:31:38 MQT: stat/Lampe1/STATUS10 = {"StatusSNS":{"Time":"2018-09-04T18:31:38","Switch2":"OFF"}}
18:31:38 MQT: stat/Lampe1/STATUS11 = {"StatusSTS":{"Time":"2018-09-04T18:31:38","Uptime":"0T00:00:20","Vcc":3.167,"POWER1":"OFF","Dimmer":90,"Color":"00E51F00","HSBColor":"128,100,90","Channel":[0,90,12,0],"Scheme":3,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"arminseins","RSSI":58,"APMac":"28:C6:8E:B3:C8:9C"}}}
```
From your `status 0`:
```
{"Status":{"Module":27,"FriendlyName":["Sonoff"],"Topic":"Lampe1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:20","StartupUTC":"2018-09-04T16:31:18","Sleep":1,"BootCount":141,"SaveCount":4432,"SaveAddress":"FB000"},"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018.09.02 07:59:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["arminseins",""],"TelePeriod":300,"SetOption":["04008009","55818000","00000001"]},"StatusMEM":{"ProgramSize":456,"Free":544,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0F50A794","24000000","020B979E","00000000"]},"StatusNET":{"Hostname":"Lampe1-1681","IPAddress":"192.168.1.11","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:BD:A6:91","Webserver":2,"WifiConfig":5},"StatusMQT":{"MqttHost":"192.168.178.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BDA691","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Tue Sep 04 16:31:38 2018","Local":"Tue Sep 04 18:31:38 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:11","Sunset":"20:27"},"StatusSNS":{"Time":"2018-09-04T18:31:38","Switch2":"OFF"},"StatusSTS":{"Time":"2018-09-04T18:31:38","Uptime":"0T00:00:20","Vcc":3.167,"POWER1":"OFF","Dimmer":90,"Color":"00E51F00","HSBColor":"128,100,90","Channel":[0,90,12,0],"Scheme":3,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"arminseins","RSSI":58,"APMac":"28:C6:8E:B3:C8:9C"}}}
```
Theo's _decode-status.py_ gives the output:
```
*** decode-status.py v20180730 by Theo Arends ***
Decoding information for device Sonoff from status report taken at 2018-09-04T18:31:38

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT enabled
[... cut ...]
  31 (OFF) Do not show Wifi and Mqtt state using Led
  50 (ON ) Timers enabled
[... 13 lines cut ...]
  USE_DS18B20
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_HOME_ASSISTANT
  USE_HTU
  USE_I2C
  USE_INA219
  USE_PZEM004T
  USE_RULES
  USE_SHT
  USE_SHT3X
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_TSL2561
  USE_WEBSERVER
  USE_WS2812
  VTABLES_IN_FLASH
  WEBSERVER_ADVERTISE
```

So, Domoticz and Rules are enabled.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-09-06:
Please, re do the following test with last version:

* in the console type `weblog 4`
* then activate the switch 2 (a PIR?)
* Copy and paste the output in the console

Dickesplumpel, 2018-09-07:
Test with Version - 6.2.1.2:

- bulb doesn't restart correctly after firmware update (have to turn off fuse - same with version 6.2)
- weblog 4 same as version 6.2
- rule is on but not triggered, switch2 (PIR) switches relais1

Thank you for spending your time and patience, but mine is consumed. I switched back to working 6.0 , maybe I try again at some point! ;-))

closed 2018-09-07 by Dickesplumpel