arendst/Tasmota #3689: Sonoff POW-R2 losing it's total value

Prutsium, 2018-09-04:
After upgrading to 6.2.0 (from 6.1.1) on 3 units Sonoff POW-R2 i lost my total value after a power down.
When adding it back with energyreset3 #Value it's back but after a power cycle it's gone again.

Reverted back to 6.1.1 and no issue found.

ascillato2, 2018-09-04:
Hi,

Please, share your `status 0` from your device with this issue. Thanks

Prutsium, 2018-09-04:
I had to revert back to 6.1.1 as the switch is used in a quite important area so the status 0 below is from 6.1.1 (with manual energyreset to get my current Total usage back)

14:17:04 MQT: domoticz/in/STATUS = {"Status":{"Module":43,"FriendlyName":["POW-R2-3-Washing"],"Topic":"washing","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
14:17:04 MQT: domoticz/in/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:03:56","StartupUTC":"2018-09-04T13:13:08","Sleep":0,"BootCount":10,"SaveCount":35,"SaveAddress":"FB000"}}
14:17:05 MQT: domoticz/in/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-23T11:19:23","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:17:05 MQT: domoticz/in/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["404Domotics",""],"TelePeriod":300,"SetOption":["00008029","55818000"]}}
14:17:05 MQT: domoticz/in/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDEE794","0C000004","37FFBFCE","00000000"]}}
14:17:06 MQT: domoticz/in/STATUS5 = {"StatusNET":{"Hostname":"POW_3","IPAddress":"192.168.99.63","Gateway":"192.168.99.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.99.1","Mac":"84:F3:EB:B1:D7:32","Webserver":2,"WifiConfig":2}}
14:17:06 MQT: domoticz/in/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.99.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B1D732","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:17:07 MQT: domoticz/in/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 04 13:17:06 2018","Local":"Tue Sep 04 14:17:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:11","Sunset":"19:27"}}
14:17:07 MQT: domoticz/in/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:17:08 MQT: domoticz/in/STATUS10 = {"StatusSNS":{"Time":"2018-09-04T14:17:07","ENERGY":{"Total":50.873,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":234,"Current":0.000}}}
14:17:13 RSL: domoticz/in/STATUS11 = {"StatusSTS":{"Time":"2018-09-04T14:17:08","Uptime":"0T00:04:00","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"404Domotics","RSSI":100,"APMac":"40:16:7E:31:A2:D2"}}}

ascillato, 2018-09-04:
Hi,

Issue fixed By Theo. Thanks :+1: 

I have tested v6.2.0.2 and now it works !

closed 2018-09-04 by ascillato2

Prutsium, 2018-09-04:
Will be 6.2.0.2 as a build be available as i cant built from sources?

ascillato2, 2018-09-04:
Yes,

Wait about an hour and download new development bins at http://thehackbox.org/tasmota/

Prutsium, 2018-09-04:
Confirmed flashed 6.2.0.2 and values are kept after power cycle.

Thanks for the quick solution!

orimate, 2019-01-27:
I have this problem wih 6.4.1 master, although the error was solved. Does anyone have experience? I'm running a test now. At orange point, the sensor was unplugged from the current. My sensor forgets the value of today.
[image]

zox-gh, 2019-04-30:
> I have this problem wih 6.4.1 master. Does anyone have experience? I'm running a test now. At orange point, the sensor was unplugged from the current. My sensor forgets the value of today.
Update:
I tested with 6.1.1, 6.2.1, 6.4.1. Looks like the value is written on flash every day 00:00 or befor rebooting. This is normal.

But after your update 'This is normal" I'm not sure any more if this is by design or a bug?
IMHO all  3 values should be immediately written on flash, not waiting till midnight.
I've  read arendst's remark about sparing flash and writing permanently only at 00:00 but cannot vote for it.  It makes use of this appliance not reliable and suitable in many applications. Just one hard power off and years of data acquisition are simply gone.

joba-1, 2019-05-01:
if you write data to flash immediately, your flash will die soon. You loose your device AND your data. What is gained?

If the data is really important, send it to a reliable database over a network and make a lot of backups of that.

vk496, 2021-02-18:
Hi,

In Tasmota 9.2.0 I have the same issue. After long power off, the Tasmota device starts with all values at 0

ascillato, 2021-02-18:
@vk496 

As designed. 

Tasmota saves today Energy values into flash ONLY at midnight so as to not wear the flash. If you de-energize your device, it will lose the today values if before you didn't trigger any event that saves the data like a software restart, toggle a relay, etc.

vk496, 2021-02-18:
@ascillato 

So why is the Yesterday and Total entries then? If only _Today Energy_ is saved to flash...

Honestly, I prefer to solder a new flash memory (1€? less?) rather than doing tricky stuff with MQTT & Home Assistant to solve this issue:

[image]

Maybe I should open a new issue to discuss a possible solution? I have a lot of Tasmota devices and this is the only issue which I couldn't find a solution

ascillato, 2021-02-18:
There is no bug about this. If your device is loosing total energy too, you have another problem. May be your device is resetting, or your power measurement chip is faulty or you have a retained mqtt message. 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

tammycutzet, 2021-03-21:
> Hi,
> 
> In Tasmota 9.2.0 I have the same issue. After long power off, the Tasmota device starts with all values at 0

I have the same issue with 9.2! There is a bug in the code. Energy today is persistent, but not EnergyTotal

ascillato, 2021-03-21:
> I have the same issue with 9.2! There is a bug in the code. Energy today is persistent, but not EnergyTotal

Please update to latest Tasmota and try again. Thanks.

vk496, 2021-03-28:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAFQAAABUCAYAAAAcaxDBAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAAySURBVHic7cEBDQAAAMKg909tDjegAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAeDVulAABbzDScQAAAABJRU5ErkJggg==###24x24:true###][Imagen de rastreo][https://github.com/notifications/beacon/ABJ7L24KQM3WTH4HVMRT6HTTEZ6GPA5CNFSM4FTDXAB2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOF7TS3DY.gif]

Hello,

I confirm that version 9.3.1 fix this issue.

After 12h of power off, the device starts with the last Total value.

Thank you for you help! Amazing project! :)

BR,
Valentín

Mikhail5555, 2021-11-23:
Sorry for reopening this issue, but i am experiencing this issue(/feature?) with 10.0.0,
I have a Sonoff POW-R2 behind a power switch which i usually turn off for the night, leads to the same resetting/losing of energy history.
[image]

Which leads to fun artefacts in the energy dashboard:  
[image]

Is there a way to enable saving to flash memory more often than only at midnight? (Like every x change, or every x minutes?)

barbudor, 2021-11-23:
There is no issue, working as designed. So please open a Github Discussion in the 'need help' section.