arendst/Tasmota #3690: iFan2 connections

jpadie, 2018-09-04:
Hi

I am hoping others have had some success in connecting an iFan2 to a westinghouse fan. 

**The Physical fan**
the fan motor of a westinghouse presents two sets of cables, one set for each winding.  

One set is connected to a switched hot and neutral.
the other is connected to one or both of a dual capacitor (0.75/1.5uF) which gets its juice from a rotary switch.

The rotary switch has four positions:  off, single hot (the first motor winding is powered), then the single hot plus one cap, then plus both caps.  

there is never a scenario in which the second windings are connected directly to the hot. 

**the iFan controller** 
the iFan just presents a single pair of wires.

Inside the iFan I can see that the first three relays link in to the fan output.  The states seem to be the following:

```
State/speed         Ry1                Ry2                Ry3
0                x                    x                     x
1                 y                    *                     x
2                y                    y                     x
3                *                    *                      y
```

x   = off
y  = on
*  = on briefly then off.
Ry1 and Ry2 are connected to hot via 3uF caps (each in an RC circuit with 200k Ohm resistors).  
Ry3 is directly switched hot.  

is there any easy solution to match this fan controller with the wiring of the westinghouse fan?  

potentially I could wire the old dual cap in place of the current two caps.  But I still don't have a switched hot for the primary winding.

I'm wondering also whether the primary winding is the started coil.  and may be disengaged by a physical centrifugal switch inside the motor?  Does anyone else have one of these that they have taken to bits?  

thanks in advance.
Justin

ascillato2, 2018-09-04:
Hi,

Please also ask this connection troubleshooting to Itead directly. They should know.

jpadie, 2018-09-04:
I have done so. 

however I'm more interested in the views of people with enquiring minds.

jpadie, 2018-09-04:
The two 3uF caps are going to result in significantly different rms AC being passed to the fan than the manufacturer specified 0.75/1.5/3.25uF for low medium and high speed.  

Compare with 3/6uF on the ifan.  And no capacitor dropping on the fast speed.  Westinghouse does not provide for the full voltage to go into the windings except for the secondary. Which I imagine to be a start  winding and would end up placing a cap in series to the output.  I've no idea whether it would be enough to start up the motor.  

If I mod the hardware I can use my own caps in lieu of the ifans.  And I can mod the code easily to use relay 1 and 2 for the fan speeds.  But I cannot see any easy way to mod the hardware to use relay 3 to energise the primary windings.  I'm almost certain that there is a centrifugal switch in the motor as if I spin the blades I can see the resistance going open-line every now and then. 

My current plan is to check the fan speed and if it is zero, to open relay 3 and allow that to energise the windings for a few seconds before cutting back to the right fan speed.  Then cross wiring the primary and secondary windings. 

I can see how to adjust the fan speed but I'm wondering what the best way in the code is to place a flag that can be tested in a loop every few seconds so that the fan speed can be reduced as required. 

Any help on this would be gratefully received.

mike2nl, 2018-09-05:
I can't help with your issue..
I use a table ventilator with 4 wires. Ground and three phases, one for every speed.
That works without any issue.
I use stwich position 3 on the table ventilator. There are four 0, 1, 2, 3.
The speed control comes then over sonoff. That's it.

When something will bechanged in iFan02 i must test it again i think
before i switch over to a new driver.

mike2nl, 2018-09-05:
@jpadie 
yes please test it, messaure again and the report back.

Otherwise you have to solder out the caps and solder different cables on the relais and connect to your fan caps. Not the best solution when your test does not work.

jpadie, 2018-09-05:
in fact I did remove the two big caps as I was going to use a bypass method originally (and I wanted to see what was happening with the relays).  

I've got one back, but I'm having trouble cleaning out the solder from the through-hole on the second.  Combination of being on the power line (lots of heat sinking) and a thick board.  I think I will have to drill it out unfortunately.

ascillato, 2018-09-07:
Hi,

Have you managed to solve your issue?

jpadie, 2018-09-07:
I wired the fan as per my last message. So the two 3uF are in series (in speed q and 2) with the two parallel caps in the fan body.  The starter coil is directly wired to the output of the ifan2. The fan works and speed control is ok. But there is a humming that I have not encountered on any Westinghouse fan before.  Even on speed 3 when the voltage should be being transmitted directly from the relay.   This is not present when the ifan2 is not in the circuit.  I'm at a loss to explain it.   But yes the issue is closed save that keeping it open and in the default list may well help others in the same situation Itead have not responded to the trouble ticket opened. Their support has been execrable over the six months I have been playing with their devices.