arendst/Tasmota #3691: Option to send cmnd when blue 'Toggle' button is clicked on the Web UI

ashmckenzie, 2018-09-04:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : sonoff basic
- [x] Provide the output of command ``status 0`` :
```
17:21:40 CMD: status 0
17:21:40 MQT: sonoff/garage/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff-garage"],"Topic":"garage","ButtonTopic":"garage","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:21:40 MQT: sonoff/garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"garage","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff-classic.bin","RestartReason":"Software/System restart","Uptime":"0T01:01:55","StartupUTC":"2018-09-04T15:19:45","Sleep":0,"BootCount":124,"SaveCount":237,"SaveAddress":"F5000"}}
17:21:40 MQT: sonoff/garage/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-08-31T15:15:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:21:40 MQT: sonoff/garage/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"x.x.x.x","LogPort":514,"SSId":["XXXX_plus","XXXX"],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
17:21:40 MQT: sonoff/garage/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
17:21:40 MQT: sonoff/garage/STATUS5 = {"StatusNET":{"Hostname":"sonoff-garage","IPAddress":"x.x.x.32","Gateway":"x.x.x.254","Subnetmask":"255.255.255.0","DNSServer":"x.x.x.10","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":3}}
17:21:40 MQT: sonoff/garage/STATUS6 = {"StatusMQT":{"MqttHost":"x.x.x.x","MqttPort":1883,"MqttClientMask":"sonoff_garage","MqttClient":"sonoff_garage","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:21:40 MQT: sonoff/garage/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 04 16:21:40 2018","Local":"Tue Sep 04 17:21:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:11","Sunset":"19:27"}}
17:21:40 MQT: sonoff/garage/STATUS10 = {"StatusSNS":{"Time":"2018-09-04T17:21:40"}}
17:21:40 MQT: sonoff/garage/STATUS11 = {"StatusSTS":{"Time":"2018-09-04T17:21:40","Uptime":"0T01:01:55","Vcc":3.138,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXX_plus","RSSI":100,"APMac":"XX:XX:XX:XX:XX:XX"}}}
```

When I click the blue 'Toggle' button on the Web UI, a cmnd is not sent to my MQTT broker.  Is it possible to configure this?

ascillato, 2018-09-04:
Why you want that?
Tasmota is already sending its stat to the MQTT broker

ashmckenzie, 2018-09-04:
I have a sonoff basic in the garage that I wish to control via a wemos D1
mini inside the house (both running tasmota).  When issuing a power ON/OFF
cmnd via home-assistant and using the switch (on the wemos) it works
beautifully.  Both devices are in sync in terms of displaying the same
ON/OFF state in the Web UI.

But when I click 'Toggle' via the Web UI, the two devices are then out of
sync.

If it's not possible to send a cmnd, is there any other way to achieve sync
between the two?

ascillato2, 2018-09-04:
Hi,

I see that your asking this also in Tasmota Chat.

If you need more information let's chat there.

closed 2018-09-04 by ascillato2

ascillato, 2018-09-05:
From the Tasmota Chat:

Solution taken was use of rules in both devices:

`rule on power1#state do publish cmnd/your_other_device_topic/power1 %value% endon`