arendst/Tasmota #3693: Compile error on 6.2.0.2

jpadie, 2018-09-04:
**Describe the bug**
6.2.0.2 will not compile on MacOS using core 2.3.0

**To Reproduce**
try to compile 6.2.0.2 on MacOS using core 2.3.0

**Expected behavior**
successful compulation

**Additional context**
the compile errors included

In function 'void SettingsDefaultSet2()':
settings:502: error: 'MQTT_BUTTON_TOPIC' was not declared in this scope
   strlcpy(Settings.button_topic, MQTT_BUTTON_TOPIC, sizeof(Settings.button_topic));
                                  ^
settings:503: error: 'MQTT_SWITCH_TOPIC' was not declared in this scope
   strlcpy(Settings.switch_topic, MQTT_SWITCH_TOPIC, sizeof(Settings.switch_topic));
                                  ^
settings:633: error: 'KEY_DEBOUNCE_TIME' was not declared in this scope
   Settings.button_debounce = KEY_DEBOUNCE_TIME;
                              ^
settings:634: error: 'SWITCH_DEBOUNCE_TIME' was not declared in this scope
   Settings.switch_debounce = SWITCH_DEBOUNCE_TIME;
                              ^
/Users/jpadie/OneDrive/Electronics/Arduino/sonoff/settings.ino: In function 'void SettingsDelta()':
settings:835: error: 'KEY_DEBOUNCE_TIME' was not declared in this scope
       Settings.button_debounce = KEY_DEBOUNCE_TIME;
                                  ^
settings:836: error: 'SWITCH_DEBOUNCE_TIME' was not declared in this scope
       Settings.switch_debounce = SWITCH_DEBOUNCE_TIME;
                                  ^
/Users/jpadie/OneDrive/Electronics/Arduino/sonoff/xdrv_01_mqtt.ino: In function 'bool MqttCommand()':
xdrv_01_mqtt:700: error: 'MQTT_BUTTON_TOPIC' was not declared in this scope
         case SC_USER: strlcpy(Settings.button_topic, MQTT_BUTTON_TOPIC, sizeof(Settings.button_topic)); break;
                                                      ^
xdrv_01_mqtt:713: error: 'MQTT_SWITCH_TOPIC' was not declared in this scope
         case SC_USER: strlcpy(Settings.switch_topic, MQTT_SWITCH_TOPIC, sizeof(Settings.switch_topic)); break;`
`

ascillato, 2018-09-04:
Hi, mine is compiling under Arduino and Platformio.

Which one are you using?

jpadie, 2018-09-04:
Arduino IDE.

ascillato, 2018-09-04:
Hi, v6.2.0.2 is compiling in my case under Arduino and Platformio. (windows 7 and linux).

Also, check the on-line travis flag: [image]

Have you copy all the files and libraries?

ascillato2, 2018-09-04:
Hi, searching in closed issues, this problem was just file missing. Please, copy all files again from github. Thanks

closed 2018-09-04 by ascillato2

jpadie, 2018-09-04:
Yes. All files and libraries copied except for user_config (brought in from an earlier version ).

jpadie, 2018-09-04:
I downloaded the files about 60 minutes ago!

ascillato2, 2018-09-04:
Ok, that is the problem. The file user_config.h had changed. You will need to add your config to the new user_config.h file

jpadie, 2018-09-04:
That is awkward.  Is that documented anywhere?  

Thanks

ascillato2, 2018-09-04:
In the changelog.

Great that is solved now for you