arendst/Tasmota #3694: DC meter from Peacefair with RS485

eazuka, 2018-09-04:
Having just read about the support for PZEM004T Energy Monitor from Peacefair. I'm just wondering if it will be possible to add support for PZEM-003 and PZEM-017 DC meter with RS485 from same manufacturer (Peacefair). I have use cases where Sonoff 4CH Pro and Sonoff SV will be used with PZEM-017 DC meter for DC (Solar) power metering and switching?

The PZEM-003 and PZEM-017 DC meter uses RS485 to communicate.

@arendst i can donate the PZEM-003 or PZEM-017 DC meter from Peacefair if you don't already have but think its possible to support it.

arendst, 2018-09-06:
I have a PZEM-014 and a PZEM-017 on order. They both use Modbus communication using the serial interface as did the PZEM-004T. The default interface is 5V. Hopefully it can be adjusted to 3V3 as I used on the PZEM-004T.

Let's see.

arendst, 2018-09-06:
Just release experimental support for PZEM-003,014,016,017.

If you manage to connect the PZEM device to a esp8266 device then configure the GPIOs as below

[image]

Pay attention to the correct numbers: 99 PZEM Rx and 98 PZEM Tx. Do not select 63 PZEM Rx and 62 PZEM Tx as they will service the PZEM-004T. This naming scheme may change in the future but I also may want to incorporate all PZEM devices in one driver. Time will tell.

If you have connected the device try to gather some debug information using the web console (do not use a serial connection as that is being used by the PZEM unless you configure other GPIOs but that may lead to timeouts). Pls provide some output here (if any ;-))

This is experimental as I do not own a device yet but the info is from the documentation available from PeaceFair.

eazuka, 2018-09-06:
@arendst you are awesome, i truly appreciate you and all you have done so far. Thanks alot.
My PZEMs are on there way also. not gotton to me yet but hopefully i receive mine before your arrive.

Thanks

mkosta74, 2018-09-18:
Hi, can I test this release? I got PZEM-016 with TTL to RS485 converter.

where I can download this release?

Regards.

ascillato, 2018-09-18:
http://thehackbox.org/tasmota/

mkosta74, 2018-09-20:
Hi, I install this version and try to get communication on modbus but keep getting errors.
after hour of troubleshooting I notice strange clicking sound from PZEM-016. When I open the case and connect it to mains I notice sparks on one resistor https://photos.app.goo.gl/BgPtHNuFHiBLsbvz7 
😄 

Looks my PZEM device is broken...

arendst, 2018-09-26:
Still waiting for my RS485toTTL converters…

[image]

mkosta74, 2018-09-29:
I use this one

MAX3485 Module TTL To RS485 Module MCU Development Accessories
 https://s.click.aliexpress.com/e/hKziY1K

But as I post earlyer my PZEM is broken.

@arendst, is it normal that tasmota gets errors from this TTL2RS485 even if there is nothing connected to 485 terminal?

vicfergar, 2018-10-03:
Hi! I ordered a Pzem-014 too. I would like to connect the ESP directly to the serial TTL lines on-board. Have someone tried it?

arendst, 2018-10-15:
With the latest release the PZEM-014/016 works when connected to the esp8266 using the interface below:

[image]

[image]

[image]

Notice the correct GPIO naming (98 PZEM TX and 99 PZEM Rx).

vicfergar, 2018-10-16:
Hi! I finally got it working without the RS485 interface!

Here are the modifications that I made:

1. Desolder the TX pin from the MAX485 to avoid interferences. (The whole chip can be removed, but I want to reduce the necessary modifications)
[image]

2. Put two pull-down resistor of 2K ohm one for RX and other to TX. This is needed to reduce the voltage to 3.3v ([Simulation circuit](http://tinyurl.com/ybqb358r))
[image]

3. Solder wires for the RX and TX lines.

Here is the final assembly adding a ESP8266 board inside the case:
[image]
[image]

I use the hardware serial with the latest release and it works like a charm. Thanks @arendst for your amazing work 👍

ascillato2, 2018-10-16:
Hi,

Closing issue as now it is implemented in Tasmota.

Thanks all for sharing their work and ideas.

closed 2018-10-16 by ascillato2

arendst, 2018-10-17:
Important PZEM module configuration change as of version 6.2.1.17:

To distinguish between the PZEM-014/016 (AC) and PZEM-003/017 (DC) I had to split the drivers and redesign the pin configuration for all supported PZEM modules.

For PZEM-004T use PZEM0XX Tx and PZEM004 Rx
For PZEM-014/016 use PZEM0XX Tx and PZEM016 Rx
For PZEM-003/017 use PZEM0XX Tx and PZEM017 Rx

SiliconAvatar, 2018-10-25:
I just got my converter in and tried it with the PZEM-016 I have, and it seems to work great! (Using 98 & 61)

I do have a feature request though:

Would it be possible to talk to more than one? Since I have a split phase power system, I need 2 modules to monitor total power consumption, and these are modbus devices so their meant to have many devices on one bus.

The PZEM-016 manual says the node address is selectable in register 0x0002, and is selectable from 1-127, so there should be no technical limitation to having multiple PZEM devices on one bus.

As to setting the device node address within tasmota, if it's not the simplest thing to do, or not desirable for tasmota to modify the devices' node address, could it be selectable to either read a single device at the default address, or to select how many devices are on the bus and choose which address they are at, to allow for more advanced multi-node configurations?

arendst, 2018-10-25:
It's not only a matter of connecting but also of representation either thru GUI or JSON. Currently only ONE energy monitoring module is supported.

SiliconAvatar, 2018-10-25:
I understand that displaying the data would be difficult, and I don't have a good solution to that; but having to configure it over the command interface and just reporting the data over MQTT would be plenty satisfactory.

mkosta74, 2018-11-17:
How can I reset kWh couter to 0?

arendst, 2018-11-17:
Search wiki for command ``EnergyReset``

mkosta74, 2018-11-17:
@arendst, thanks

I was trying to do it over serial bridge using "SerialSend5 01421180"

Is it possble to comunicate with PZEM 016 like that?

vicfergar, 2018-12-02:
Hi @hteggi!

I glad that this work helps you. You are right, 2 resistors are needed both with the same value of 2K ohms. Because of the distance between the pads, I used one through hole and another SMD. That last one is visible in the second picture, you can see an small SMD resistor labeled with 202 between the pads of the TX optocoupler.

Hope it works for you too!

lisergio, 2018-12-11:
where have you connected the power of the  esp8266?

vicfergar, 2018-12-11:
Hi! I used an ESP8266 board with an embedded 3.3v power regulator, so I connect it directly to the 5v output provided by the PZEM

vicfergar, 2018-12-12:
> Thanks @vicfergar.
> 
> Works well with a Wemos D1 mini and home assistant.

Great! I'm glad to hear that.

mkosta74, 2018-12-13:
It was working fine until today when I get this reading:[image]

jkay90, 2018-12-18:
> 2. Put two pull-down resistor of 2K ohm one for RX and other to TX. This is needed to reduce the voltage to 3.3v ([Simulation circuit](http://tinyurl.com/ybqb358r))
>    [image]

can you be bit clear here please. should i solder the pull down resistor on the RX wire or the adjacent connector of RX wire?

valdragas, 2019-01-23:
hello, is it not possible to calibrate the pzem 016 sensor?

I'm trying to enter the commands CurrentSet, PowerSet, etc. but they do not work

arendst, 2019-01-23:
No. Not supported.

dion0511, 2019-01-28:
> With the latest release the PZEM-014/016 works when connected to the esp8266 using the interface below:
> 
> [image]
> 
> [image]
> 
> [image]
> 
> Notice the correct GPIO naming (98 PZEM TX and 99 PZEM Rx).

Hello @arendst 

I wanna asking, how to set in tasmota module if i use ttl to rs485 like below

[image]

Thank you very much

GeirTS, 2019-02-04:
I`m using pzem 017 with a 200A shunt(peacefair).  The current readings shows only half of what it`s supposed to. How do I change the current resolution?

gridi88, 2019-05-27:
> It was working fine until today when I get this reading:
> 
> [image]

Hello friend. I have exactly the same problem, can you tell me how to resolve it? Thank you in advance.

mkosta74, 2019-05-27:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#power-monitoring

EnergyReset1	0..42500 = ((p)re)set Energy Today in Wh

tizius79, 2019-05-28:
Same problem.
I think it's related to the resistances (I put them at 1.8kohm) or the wifi module too close to the analog part.
[image]

xCapeITA, 2019-06-02:
I bet you are using tasmota coded with 2.4.2 core version, right? It does that also to me. Using tasmota with 2.3.0 brings to bootloop (10k bootloop per day) ,- pzem016 modded with resistances

meingraham, 2019-06-02:
I was running 6.4.1.19 with 2.4.2 with many exception restarts a day. Not a complete bootloop because the device settings were not getting reset. I upgraded the device to 6.5.0.12 and 2.5.2 Core a week ago. Zero exception restarts! And, unlike 2.4.2 only 6 link drops in the week. Very similar to my 2.3.0 devices within similar RSSI ranges.

pite81, 2019-09-17:
> Just release experimental support for PZEM-003,014,016,017.
>
> If you manage to connect the PZEM device to a esp8266 device then configure the GPIOs as below
>
> [image]
>
> Pay attention to the correct [...]

Hey guys, thanks for the firmware, you are doing something wonderful work over there! I am on a project to build a link between a Eastron SDM630 3 phase energy meter and my node-red. I plan to use an ESP8266 nodemcu, which I successfully installed the tasmota. Would the SDM630 work with the PZEM settings or should I use serial RX-TX for gpio 1 and 3. Opinions would be much appreciated. Thanks in advance.

fragolinux, 2019-11-09:
followed this guide but can't get it working... weird spot readings, then nothing... desoldered the tx pin of the max485 chip, put 2k resistor between 1 and 2, 2k between 1 and 3, pin2 to wemos tx, pin4 to rx, power from bottom connector, as per images below... what am i missing?

[image]

[image]

[image]

```
12:35:10 CMD: status 0
12:35:10 MQT: stat/PZEM016/STATUS = {"Status":{"Module":0,"FriendlyName":["PZEM016"],"Topic":"PZEM016","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:35:10 MQT: stat/PZEM016/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:39","StartupUTC":"2019-11-09T11:34:31","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":19,"SaveAddress":"F9000"}}
12:35:10 MQT: stat/PZEM016/STATUS2 = {"StatusFWR":{"Version":"7.0.0.4(5ebf04f-tasmota)","BuildDateTime":"2019-11-09T12:00:24","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
12:35:10 MQT: stat/PZEM016/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SharkIOT","Shark-HOME2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
12:35:10 MQT: stat/PZEM016/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":436,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:35:10 MQT: stat/PZEM016/STATUS5 = {"StatusNET":{"Hostname":"PZEM016-3757","IPAddress":"192.168.1.121","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:7E:0E:AD","Webserver":2,"WifiConfig":2}}
12:35:10 MQT: stat/PZEM016/STATUS6 = {"StatusMQT":{"MqttHost":"mrshark.ddns.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7E0EAD","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:35:10 MQT: stat/PZEM016/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 09 11:35:10 2019","Local":"Sat Nov 09 12:35:10 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:49","Sunset":"17:18"}}
12:35:10 MQT: stat/PZEM016/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:35:10 MQT: stat/PZEM016/STATUS10 = {"StatusSNS":{"Time":"2019-11-09T12:35:10","ENERGY":{"TotalStartTime":"2019-11-09T11:59:48","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
12:35:11 MQT: stat/PZEM016/STATUS11 = {"StatusSTS":{"Time":"2019-11-09T12:35:10","Uptime":"0T00:00:39","UptimeSec":39,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"SharkIOT","BSSId":"20:28:18:A0:F0:42","Channel":11,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:09"}}}
12:39:51 MQT: tele/PZEM016/STATE = {"Time":"2019-11-09T12:39:51","Uptime":"0T00:05:20","UptimeSec":320,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"SharkIOT","BSSId":"20:28:18:A0:F0:42","Channel":11,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:09"}}
12:39:51 MQT: tele/PZEM016/SENSOR = {"Time":"2019-11-09T12:39:51","ENERGY":{"TotalStartTime":"2019-11-09T11:59:48","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
```

mkosta74, 2019-11-24:
[image]

This combination works for me...

bpmittal, 2020-01-05:
how to connect PZEM 017 DC with nodemcu ?
do i need any converter to connect it with esp8266 nodemcu ?
or i can directly connect ?

[image]

Gio-dot, 2020-01-10:
I, i would like to ask: PZEM 017 DC can measure bidirectional dc current? It does support negative current?

Gio-dot, 2020-01-10:
> I wanna asking, how to set in tasmota module if i use ttl to rs485 like below
> 
> [image]
> 
> Thank you very much

I think that is not good: you need a converter with automatic flow detection (RE/DE) and that can support 3.3V. Something like this one:
https://it.aliexpress.com/item/32977187783.html?spm=a2g0o.productlist.0.0.6c116706Kuvsxx&algo_pvid=d34ba343-f401-4298-ad98-19b162f61419&algo_expid=d34ba343-f401-4298-ad98-19b162f61419-20&btsid=e29e2fa8-6fa9-416a-bc83-1f4cd3d688ed&ws_ab_test=searchweb0_0,searchweb201602_7,searchweb201603_53

skeletik07, 2020-02-27:
Hi everybody. I'm trying to connect my pzem-016 to nodemcu with rs845-ttl converter. I am using this settings for module.(and i am trying a lot of more). Led RX and TX on the converter are blinking, but tasmota does't show any information about voltage, or current. When i connect pzem to pc using rs845-usb converter and original soft all works fine. Could you please help me with my trouble?
[image]
[image]
[image]

Gio-dot, 2020-02-27:
> Hi everybody. I'm trying to connect my pzem-016 to nodemcu with rs845-ttl converter. I am using this settings for module.(and i am trying a lot of more). Led RX and TX on the converter are blinking, but tasmota does't show any information about voltage, or current. Rx pin on the converter are connected to RX on nodemcu, and TX to TX. When i connect pzem to pc using rs845-usb converter and original soft all works fine. Could you please help me with my trouble?
> [image]
> [image]
> [image]

Did you try to reverse tx/rx wires?

skeletik07, 2020-02-27:
> Did you try to reverse tx/rx wires? Rx must go to tx and Tx to rx.
>
Led indicators stop blinking and tasmota shows random values.
Maybe reverse RX and TX in module settings too?
[image]

Gio-dot, 2020-02-27:
What did you connect to vcc and gnd?

skeletik07, 2020-02-27:
> What did you connect to vcc and gnd?

5V output and ground from pzem-016

Gio-dot, 2020-02-27:
I'm quite sure that there you have to connect gnd and 3.3V coming from node mcu

skeletik07, 2020-02-27:
> I'm quite sure that there you have to connect gnd and 3.3V coming from node mcu

Yes! It's working now!
Thanks a lot!
And one last question, please. How i can save data with current and power for making daily graps? With MQTT brockers? Or it's some easy other way? Making graps i will be doing in excel or matlab, so i can save data in txt or some other text format.

Gio-dot, 2020-02-27:
I use home assistant to manage all domotic stuff.

skeletik07, 2020-02-28:
But can i take exectly the digits of power and current in home assistant? Because i'm gonna do some research work with this data and i need power measurement every 3/5 seconds and saving this stuff somewhere i can operate with this

rt45433, 2020-05-18:
I am using the pzem-017 and have got it working with tasmota. Just wanted to ask if there was a way to read the soc state or charge reading so I can use this through tasmato or have I missed it somewhere. I need this as I turn of my solar charger at 80% soc of the battery so want to see if I can did it using the pzem-017 and tasmota. Any help would be appreciated.

Gio-dot, 2020-05-19:
> I am using the pzem-017 and have got it working with tasmota. Just wanted to ask if there was a way to read the soc state or charge reading so I can use this through tasmato or have I missed it [...]

It is not possible: soc calculation is very hard task  that depend of a huge number of variables. It is not possible to calculate soc without knowing all those variables (battery type, battery capacity, battery state of healt, temperature, charge/discharge current, etc.)

rt45433, 2020-05-20:
Thanks for the reply and information. I will use it with the data it currently has. Thanks again 👍 -)

kittipongpoosri, 2020-10-07:
I need to code

kittipongpoosri, 2020-10-07:
ฉันต้องการรหัสเพื่อใช้กับ esp8266

Oleg-Sob, 2021-03-10:
Installed PZEM -017, everything works, the voltage shows an integer, tell me how you can make the voltage show with tenths or better hundredths.

Oleg-Sob, 2021-03-10:
[image]

austrisv, 2021-03-11:
@Oleg-Sob it's VoltRes #
https://tasmota.github.io/docs/Commands/

Oleg-Sob, 2021-03-11:
Thanks, I will try.

Oleg-Sob, 2021-03-11:
Thank you very much, it worked out.

Oleg-Sob, 2021-03-11:
It is not possible to calibrate the current in PZEM 017, it does not respond to the CurrentSet and CurrentCal commands. Tell me why.

[48 more replies]