arendst/Tasmota #3695: ARILUX AL-LC10 "toggle" button not working for RGBW

gitolicious, 2018-09-04:
**Describe the bug**
I am using Arilux RF on an RGBW LED strip. The toggle button does not trigger any action.

- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : ARILUX AL-LC10
- [x] _Provide the output of command_``status 0`` :
```
19:53:16 MQT: stat/sonoff-paper/STATUS = {"Status":{"Module":36,"FriendlyName":["xxx"],"Topic":"xxx","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:53:16 MQT: stat/sonoff-paper/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://xxx:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T01:01:18","StartupUTC":"2018-09-04T17:51:58","Sleep":20,"BootCount":53,"SaveCount":311,"SaveAddress":"F7000"}}
19:53:16 MQT: stat/sonoff-paper/STATUS2 = {"StatusFWR":{"Version":"6.2.0.2","BuildDateTime":"2018-09-04T19:35:22","Boot":7,"Core":"STAGE","SDK":"2.0.0(656edbf)"}}
19:53:16 MQT: stat/sonoff-paper/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"xxx","LogPort":514,"SSId":["xxx","xxxx"],"TelePeriod":300,"SetOption":["00008009","41818000","00000000"]}}
19:53:16 MQT: stat/sonoff-paper/STATUS4 = {"StatusMEM":{"ProgramSize":412,"Free":524,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0800E590","00000000","00000096","00000000"]}}
19:53:16 MQT: stat/sonoff-paper/STATUS5 = {"StatusNET":{"Hostname":"xxx","IPAddress":"xx.xx.xx.xx","Gateway":"xx.xx.xx.xx","Subnetmask":"255.255.254.0","DNSServer":"xx.xx.xx.xx","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":2}}
19:53:16 MQT: stat/sonoff-paper/STATUS6 = {"StatusMQT":{"MqttHost":"xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ABF785","MqttUser":"xx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:53:16 MQT: stat/sonoff-paper/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 04 18:53:16 2018","Local":"Tue Sep 04 19:53:16 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
19:53:16 MQT: stat/sonoff-paper/STATUS10 = {"StatusSNS":{"Time":"2018-09-04T19:53:16"}}
19:53:16 MQT: stat/sonoff-paper/STATUS11 = {"StatusSTS":{"Time":"2018-09-04T19:53:16","Uptime":"0T01:01:18","Vcc":3.580,"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"ON","Speed":3,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"xx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":7,"RSSI":98}}}
```

**To Reproduce**
Switch on ARILUX AL-LC10, push "Toggle" button on remote.

**Expected behavior**
Colors changing on my LED strip.

**Additional context**
The command is properly received and `arilux_rf_toggle` variable is rotating from 200-203.
https://github.com/arendst/Sonoff-Tasmota/blob/ebb7f929a07db646fedbf17e82f3360796c6eec5/sonoff/xdrv_04_light.ino#L206-L209
```
20:03:23 RFR: Hostcode 0x6101, Received 0x610102
20:03:23 SRC: Light
20:03:23 RSL: Received Topic /Color, Data Size 3, Data 200
20:03:23 RSL: Group 0, Index 1, Command COLOR, Data 200
20:03:23 MQT: stat/sonoff-paper/RESULT = {"Color":"000000FF"}
20:03:24 RFR: Hostcode 0x6101, Received 0x610102
20:03:24 SRC: Light
20:03:24 RSL: Received Topic /Color, Data Size 3, Data 201
20:03:24 RSL: Group 0, Index 1, Command COLOR, Data 201
20:03:24 MQT: stat/sonoff-paper/RESULT = {"Color":"000000FF"}
```

Now I think it should be picked up somewhere along these lines:
https://github.com/arendst/Sonoff-Tasmota/blob/ebb7f929a07db646fedbf17e82f3360796c6eec5/sonoff/xdrv_04_light.ino#L1009-L1016

Probably it is not, because I only have RGBW (AL-LC10), not RGBWC (AL-LC11)? How would I need to change the code to support warm white?

gitolicious, 2018-09-04:
Yes I did. Currently I am using a custom module template though (added above). As you can see in my description, the command is received correctly. Also I can set all RGBW colors manually without any problems.

gitolicious, 2018-09-04:
I think the root cause actually is that the `Color 20x` commands is ignored for RGBW. If I add PWM5 for GPIO3 I am getting
```
20:44:40 CMD: color 201
20:44:40 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"0,0,0","Channel":[0,0,0,100,0],"CT":153}
```
If only 4 PWMs are in use, the command is ignored.
```
20:55:56 CMD: color 201
20:55:56 MQT: stat/sonoff/RESULT = {"Color":"000000FF"}
```
Something like
``` else if (LST_RGBW == light_subtype) { ```
needs to be added in `xdrv_04_light.ino:L1013` I guess but my tries with 
```
    else if (LST_RGBW == light_subtype) {
      memcpy_P(&light_entry_color[3], &kFixedColdWarm[value -200], 2);
      entry_type = 1;                                 // Hexadecimal
    }
```
gave me weird results as I didn't fully understand what the command does.

gitolicious, 2018-09-04:
So I think I have a working solution now. I duplicated the code which is available for ColdWarm but only put values for a single white value.
```
struct LWColor {
  uint8_t W;
};
#define MAX_FIXED_WHITE  3
const LWColor kFixedWhite[MAX_FIXED_WHITE] PROGMEM = { 0, 255, 128 };
```
Then I am cycling through this.
```
    if (LST_RGBW == light_subtype) {
      memcpy_P(&light_entry_color[3], &kFixedWhite[value -200], 1);
      entry_type = 1;                                 // Hexadecimal
    }
```
Now when pressing the toggle button, my LED strip goes through full brightness white, medium brightness white and off. Is that according to what would happen for an RGBCW strip?

ascillato, 2018-09-04:
> Now when pressing the toggle button, my LED strip goes through full brightness white, medium brightness white and off. Is that according to what would happen for an RGBCW strip?

I think so.

Nice solution :+1:

gitolicious, 2018-09-04:
Ok, cool. Not sure if the "off" part makes any sense, as there is an off button right next to the toggle button on the remote. But in this first step, I wanted to reproduce the RGBCW behavior. Maybe will change it in the future...

ascillato2, 2018-09-05:
Hi,

Theo has added your solution.

Thanks for share :+1:

closed 2018-09-05 by ascillato2