arendst/Tasmota #3696: SerialSend not working

au190, 2018-09-04:
**Describe the bug**
Config:
GPIO1   - SerBr Tx
GPOI3   - SerBr Rx
Ardiuno nano Rx, Tx

When Iam using Serial Bridge Hassio-ESP- Arduino the SerialSend is not gouing trough.
Boud rate is 115200

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : ESP8266 - arduinonano
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_

1. Receiveing data from Arduino Nano is ok (SSerialReceived):
{"topic":"tele/s_9/RESULT","payload":"{\"SSerialReceived\":\"{I:13.44,P:3092.28}\"}","qos":0,"retain":false,"_msgid":"1169269.36d65d9"}

2. Sending MQTT msg is not gouing thru to the arduino nano. When I send the message from the Hassio the message can be seen on the ESP log, but not received on the arduino. This log is from ESP console.

19:55:59 MQT: stat/s_9/RESULT = {"SerialSend":"Done"}
19:55:59 MQT: stat/s_9/RESULT = {"SerialSend":"Done"}
19:56:00 MQT: stat/s_9/RESULT = {"SerialSend":"Done"}

3. If I sending messge with the IDE console to the Ardunio, that massage is sending back to the ESP and MQTT.

19:54:23 MQT: tele/s_9/RESULT = {"SSerialReceived":"I received: 49I received: 10"}
19:54:27 MQT: tele/s_9/RESULT = {"SSerialReceived":"I received: 49I received: 10"}
19:54:29 MQT: tele/s_9/RESULT = {"SSerialReceived":"I received: 49I received: 10"}
19:54:32 MQT: tele/s_9/RESULT = {"SSerialReceived":"I received: 50I received: 10"}
19:54:34 MQT: tele/s_9/RESULT = {"SSerialReceived":"I received: 49I received: 10"}

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-04:
Hi,

In your post said tha from MQTT the answer is `19:55:59 MQT: stat/s_9/RESULT = {"SerialSend":"Done"}`

So, you are using SERIALSEND instead of **S**SERIALSEND. You are sending by hardware serial pin and not by software serial pin

au190, 2018-09-04:
I have tried both command the result is the same. When I sending msg with the Arduino IDE console then I getting message back form the Arduino to the console and also to the ESP and Hassio MQTT, tha you can see on the log I have posted.

ascillato2, 2018-09-04:
Ah, ok.

If you want to use Hardware serial, you have to leave those pins set to none.
SerBR means SerialBridge for Software Serial Feature (SSerialsend command) to use serial communication in other GPIOs different than the hardware TX and RX.

I have just tested the following:

* two sonoff connected by serial 
* module configuration left everything to none (default)
* by MQTT I sent: `cmnd/sonoff1/serialsend hisonoff`

The sonoff1 receives the command and execute it.
Then sonoff2 receives it and resends it by mqtt using the `tele-` message

So, may be the problem is just a mix of hardware and software serial features. Please, try to set in module configuration TX and RX as none

ascillato, 2018-09-04:
Please, let us know if this is working for you. In that case, please close this issue. Thanks

au190, 2018-09-05:
After removing the config from Tasmota its working !
Thx four your help !

closed 2018-09-05 by au190