arendst/Tasmota #3699: Tasmota 6.2.0 - Sonoff TH-16 (POW) erratic power values

petez69, 2018-09-05:
**Describe the bug**

Sonoff POW, sends mqtt values to Homeseer (Automation).  Event threshold 50watts.  After upgrading from 5.10 to 6.2.0 (latest) I am now getting multiple triggers from this threshold being crossed when the dryer is running.  I am currently not sure if its being caused during max power consumption 400W and dipping or in the cooling cycle 30w and spiking.  I've altered the trigger threshold from 50 to 150W, still multiple triggers.  I am not sure if the POW is occasionally sending a "0" power and 100 seconds later the real value.  Over a 1 hour period I get my event being triggered 6+ times.

 _Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ : Sonoff POW Tasmota upgraded from 5.10a to 6.2.0 (03/09/2018) using Arduino IDE
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```
13:35:09 MQT: stat/sonoff-dryer/STATUS = {"Status":{"Module":6,"FriendlyName":["Sonoff Dryer"],"Topic":"sonoff-dryer","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:35:09 MQT: stat/sonoff-dryer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"1T17:25:10","StartupUTC":"2018-09-03T10:39:59","Sleep":0,"BootCount":59,"SaveCount":425,"SaveAddress":"F5000"}}
13:35:09 MQT: stat/sonoff-dryer/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-08-31T15:15:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:35:09 MQT: stat/sonoff-dryer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"synology","LogPort":514,"SSId":["___","..."],"TelePeriod":100,"SetOption":["00000009","55818000","00000001"]}}
13:35:09 MQT: stat/sonoff-dryer/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
13:35:09 MQT: stat/sonoff-dryer/STATUS5 = {"StatusNET":{"Hostname":"sonoff-dryer-5591","IPAddress":"192.168.11.53","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"EC:FA:BC:00:95:D7","Webserver":2,"WifiConfig":3}}
13:35:09 MQT: stat/sonoff-dryer/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.home.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0095D7","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:35:09 MQT: stat/sonoff-dryer/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 05 04:05:09 2018","Local":"Wed Sep 05 13:35:09 2018","StartDST":"Sun Jan 28 01:00:00 2018","EndDST":"Mon Jan 08 02:00:00 2018","Timezone":99,"Sunrise":"14:42","Sunset":"03:55"}}
13:35:09 MQT: stat/sonoff-dryer/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:35:09 MQT: stat/sonoff-dryer/STATUS10 = {"StatusSNS":{"Time":"2018-09-05T13:35:09","ENERGY":{"Total":65.190,"Yesterday":0.625,"Today":0.397,"Power":0,"Factor":0.00,"Voltage":238,"Current":0.000}}}
13:35:09 MQT: stat/sonoff-dryer/STATUS11 = {"StatusSTS":{"Time":"2018-09-05T13:35:09","Uptime":"1T17:25:10","Vcc":3.182,"POWER":"ON","Wifi":{"AP":1,"SSId":"___","RSSI":88,"APMac":"80:2A:A8:5A:E9:E1"}}}

**To Reproduce**
Draw power from the device over 300W for a 60 minute period and report the values via MQTT every 100 seconds.  Suspect there will be a spurious value produced

**Expected behavior**
V5.10 of tasmota reported the power usage correctly every 100 seconds.  When the dryer is in the cooling cycle it is under 50W and remained so.  This was 100% reliable to track dryer "on" and "off" state.  

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
#3425
#3585

I have calibrated the unit again, issue remains

**(Please, remember to close the issue when the problem has been addressed)**

petez69, 2018-09-05:
New binary uploaded after commit: 07dafe1,  6.2.0.2 released.  Have OTA flashed to see if this resolves the issue ?

petez69, 2018-09-05:
Upgraded to 6.2.0.2

Still seeing erratic power values on the POW.  Telemetry period set to 300 seconds, console showing telemetry being sent between 1-40 seconds constantly, power ranging from 0-70.  Something isnt right :-)

arendst, 2018-09-05:
Set powerdelta to 0.

ascillato, 2018-09-05:
Please, test and report back if new release v6.2.1 is having this problem due to a significant change to POW measurements subroutines.

arendst, 2018-09-05:
@ascillato the significant change is not in 6.2.1.

ascillato, 2018-09-05:
ah, ok, it is only in development branch

petez69, 2018-09-05:
Theo, hard to say if its fixed. I still had high values ie: 60W pulse when the dryer had finished, this seems strange.   I have a 2nd POW powering a 12V power supply with constant load. Power is 23W and I see occasional high of 42W on web interface, there is absolutely NO change in the load.  Just watching web interface, I saw a high of 102W, is this a timing issue ?

arendst, 2018-09-05:
If it is NOT a POW R2 then it is a timing/interrupt issue which is hard to solve. You might want to compile a small image without unneeded features but I suppose it won't do much.

A POW R2 or S31 does the measuring using a more intelligent approach offloading Tasmota. This is more stable.

closed 2018-09-05 by ascillato2