arendst/Tasmota #3700: fatal exception on setting KNX PA

jpadie, 2018-09-05:
**Describe the bug**
when setting a public address the sonoff crashes and reboots with a fatal exception 28.
core 2.3.0
tasmota v 6.2.0.2

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ :  sonoff basic
- [ ] _Provide the output of command_``status 0`` :
```
{"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"***","RestartReason":"Exception","Uptime":"0T00:01:46","StartupUTC":"2018-09-05T08:43:02","Sleep":0,"BootCount":2,"SaveCount":5,"SaveAddress":"F7000"}}
{"StatusFWR":{"Version":"6.2.0.2","BuildDateTime":"2018-09-05T10:15:59","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Livebox-A217","Gaia-Rabastens-Free"],"TelePeriod":300,"SetOption":["02008009","55818000","00000000"]}}
{"StatusMEM":{"ProgramSize":420,"Free":580,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","1D003080","000003A1","21021796","00000000"]}}
{"StatusNET":{"Hostname":"bedswitch-2077","IPAddress":"192.168.1.25","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:AD:88:1D","Webserver":2,"WifiConfig":0}}
{"StatusMQT":{"MqttHost":"mqtt.no3.co.uk","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AD881D","MqttUser":"***","MqttType":3,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
{"StatusTIM":{"UTC":"Wed Sep 05 08:44:48 2018","Local":"Wed Sep 05 10:44:48 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
{"StatusSNS":{"Time":"2018-09-05T10:44:48"}}
{"StatusSTS":{"Time":"2018-09-05T10:44:48","Uptime":"0T00:01:46","Vcc":3.116,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Livebox-A217","BSSId":"30:7C:B2:4D:A2:17","Channel":6,"RSSI":80}}}

```

**To Reproduce**
send the following command 
mosquitto_pub -u *** -P *** -t hannah/bedswitch/cmnd/KNX_PA -m "1.1.1"

received response:
`
1.1.1
Online
{"Module":"Sonoff Basic","Version":"6.2.0.2","FallbackTopic":"DVES_AD881D","GroupTopic":"sonoffs"}
{"WebServerMode":"Admin","Hostname":"bedswitch-2077","IPAddress":"192.168.1.25"}
{"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x40203ae3 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
{"POWER":"OFF"}
`

**Expected behavior**
the address is accepted by the module

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-05:
You are right. will investigate. Thanks for report

ascillato, 2018-09-05:
As a workaround for now, you can set the KNX Physical Address using the webmenu. That works fine. Will investigate the problem with the KNX_PA command

jpadie, 2018-09-05:
yes - I can't do that unfortunately as I am several hundred kms from the location!  I may be able to use another sonoff to send a web command on the same subnet.

jpadie, 2018-09-05:
just in case it helps I don't see a KNX_addr declared in Knx_Command, so it will be using the global variable, which may already be holding values.  Is this intended?

ascillato, 2018-09-05:
I don't understand what you meant. The KNX commands are only the ones in the wiki.

ascillato, 2018-09-05:
You can make all KNX configuration by the webmenu or by the KNX commands.

jpadie, 2018-09-05:
I was referring to the method (I think actually a function) called KnxCommand() at line 1058 of xdrv_11_knx.ino

ascillato, 2018-09-05:
Ah, ok, you are talking from the code side, not the user side! The configuration values are global because they are being stored in eeprom. Local variables are temporal. There is no problem with that. I will investigate what is the issue with that particular command and report back. Thanks.

jpadie, 2018-09-05:
we are talking at cross purposes.

in the function that saves the data from the web-server (Save_Settings(), line 1005) a local variable of type address_t is declared and that is used for storing the input from the webserver.  the union value of that struct (.value) is then  set using the function knx.physical_address_set( ) and the settings stored by assigning the .value of the struct to the setting Settings.knx_physsical_addr (a uint16_t).  Which looks correct to me.

this is not the same as in the mqtt command (and I assume the same is true for the console and perhaps serial input).  instead the _global_ variable of KNX_addr is directly manipulated without first clearing it and the helper function knx.physical_address_set is not used (which would be necessary to make the config start working).   Writing the setting looks correct.  

so if there is an error in that segment, the likely culprit might be the struct.  i may have time to test this later.

jpadie, 2018-09-05:
The method in the code called Knx_Command.  It returns a Boolean.

jpadie, 2018-09-05:
I just uploaded the pre compiled binary of 6.2.1 (knx) to a device. 

Maybe a connected issue but the webserver craps out when you try to add a group address to receive from 3.3.3 -> output 1 toggle.  I'm testing on a mobile so I can't debug any further.  Nothing on the mqtt output though.

ascillato, 2018-09-05:
I'm investigating this Bug.

The KNX commands in the driver were working in previous releases.

Now the commands `KNX_PA`, `KNX_CB` and `KNX_GA` are not working. They gave an exception 28 error (out of memory).

Tracing this error for `KNX_PA` command, it gives the exception at line 1163:
https://github.com/arendst/Sonoff-Tasmota/blob/c6507482c4969d452a764c209ebe603265ff7394/sonoff/xdrv_11_knx.ino#L1156-L1165

So, processing the input data to extract each parameter it ends up with no memory.

Looking for a solution.

@arendst Any idea?

jpadie, 2018-09-05:
Ah. Interesting. I was looking at that and was concerned about it.  But saw it was used elsewhere so though it was old code. I will review and refactor.

https://github.com/arendst/Sonoff-Tasmota/blob/c6507482c4969d452a764c209ebe603265ff7394/sonoff/xdrv_11_knx.ino#L1156-L1165

ascillato, 2018-09-05:
@andrethomas has just found the error.

fixing soon.

ascillato, 2018-09-05:
The commit https://github.com/arendst/Sonoff-Tasmota/commit/c6a14c5fe72ead31c3bc869e6cf93ea2e8331121 made some changes in support subroutines. This modify the way how the function subStr() works. 

Reverting just the change of that function solves the issue.

ascillato, 2018-09-05:
Fixed with PR #3710

closed 2018-09-05 by ascillato2