arendst/Tasmota #3701: Rule trigger same condition repeteadly even Rule 5 is set

mkh595, 2018-09-05:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Hardware used : Wemos D1 mini, HC SR04
- [X] Provide the output of command ``status 0`` :
```
10:43:33 CMD: Status 0
10:43:33 MQT: stat/home198/STATUS = {"Status":{"Module":18,"FriendlyName":["home198"],"Topic":"home198","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":7,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:43:33 MQT: stat/home198/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffxxx","OtaUrl":"http://192.168.1.15:80/arduino/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:42:30","StartupUTC":"2018-09-05T07:01:03","Sleep":0,"BootCount":14,"SaveCount":56,"SaveAddress":"F4000"}}
10:43:33 MQT: stat/home198/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-09-04T23:14:28","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:43:33 MQT: stat/home198/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"192.168.1.15","LogPort":514,"SSId":["mkh","mkh-samsung"],"TelePeriod":60,"SetOption":["00008009","55818000","00000000"]}}
10:43:33 MQT: stat/home198/STATUS4 = {"StatusMEM":{"ProgramSize":433,"Free":568,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0F482794","24000001","02000496","00000000"]}}
10:43:33 MQT: stat/home198/STATUS5 = {"StatusNET":{"Hostname":"home198","IPAddress":"192.168.1.198","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.25.1","Mac":"5C:CF:7F:C0:14:3F","Webserver":2,"WifiConfig":5}}
10:43:33 MQT: stat/home198/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.14","MqttPort":1883,"MqttClientMask":"ESP_%06X","MqttClient":"ESP_C0143F","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:43:33 MQT: stat/home198/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 05 08:43:33 2018","Local":"Wed Sep 05 10:43:33 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:08","Sunset":"19:22"}}
10:43:33 MQT: stat/home198/STATUS10 = {"StatusSNS":{"Time":"2018-09-05T10:43:33","SR04":{"Distance":82}}}
10:43:33 MQT: stat/home198/STATUS11 = {"StatusSTS":{"Time":"2018-09-05T10:43:33","Uptime":"0T01:42:30","Vcc":2.743,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":58,"APMac":"2C:4D:54:83:7F:68"}}}
```

I am using following rule and expect that there will be same state changed only once until oposite condition is meet.

```
10:44:52 MQT: stat/home198/RESULT = {"Rule2":"ON","Once":"ON","StopOnError":"OFF","Free":435,"Rules":"on SR04#Distance<20 do POWER ON endon on SR04#Distance>40 do POWER OFF endon"}
```
But there are randomly repeated 
```
RUL: SR04#DISTANCE>40 performs "POWER OFF"
MQT: stat/home198/RESULT = {"POWER":"OFF"}
```
without trigger of oposite condition SR04#DISTANCE<20

```
10:07:39 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:07:39","SR04":{"Distance":62}}
10:08:40 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:08:40","Uptime":"0T01:07:37","Vcc":2.741,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":74,"APMac":"2C:4D:54:83:7F:68"}}
10:08:40 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:08:40","SR04":{"Distance":115}}
10:09:40 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:09:40","Uptime":"0T01:08:37","Vcc":2.747,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":76,"APMac":"2C:4D:54:83:7F:68"}}
10:09:40 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:09:40","SR04":{"Distance":131}}
10:10:41 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:10:41","Uptime":"0T01:09:38","Vcc":2.740,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:10:41 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:10:41","SR04":{"Distance":49}}
10:11:41 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:11:41","Uptime":"0T01:10:38","Vcc":2.741,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:11:41 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:11:41","SR04":{"Distance":49}}
10:12:41 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:12:41","Uptime":"0T01:11:38","Vcc":2.742,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":68,"APMac":"2C:4D:54:83:7F:68"}}
10:12:41 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:12:41","SR04":{"Distance":47}}
10:13:41 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:13:41","Uptime":"0T01:12:38","Vcc":2.741,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:13:41 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:13:41","SR04":{"Distance":47}}
10:14:07 RUL: SR04#DISTANCE>40 performs "POWER OFF"
10:14:07 MQT: stat/home198/RESULT = {"POWER":"OFF"}
10:14:07 MQT: stat/home198/POWER = OFF
10:14:42 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:14:42","Uptime":"0T01:13:39","Vcc":2.741,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:14:42 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:14:42","SR04":{"Distance":48}}
10:14:55 RUL: SR04#DISTANCE>40 performs "POWER OFF"
10:14:55 MQT: stat/home198/RESULT = {"POWER":"OFF"}
10:14:55 MQT: stat/home198/POWER = OFF
10:15:19 RUL: SR04#DISTANCE>40 performs "POWER OFF"
10:15:19 MQT: stat/home198/RESULT = {"POWER":"OFF"}
10:15:19 MQT: stat/home198/POWER = OFF
10:15:36 RUL: SR04#DISTANCE>40 performs "POWER OFF"
10:15:36 MQT: stat/home198/RESULT = {"POWER":"OFF"}
10:15:36 MQT: stat/home198/POWER = OFF
10:15:42 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:15:42","Uptime":"0T01:14:39","Vcc":2.742,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":68,"APMac":"2C:4D:54:83:7F:68"}}
10:15:42 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:15:42","SR04":{"Distance":47}}
10:16:33 RUL: SR04#DISTANCE>40 performs "POWER OFF"
10:16:33 MQT: stat/home198/RESULT = {"POWER":"OFF"}
10:16:33 MQT: stat/home198/POWER = OFF
10:16:43 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:16:43","Uptime":"0T01:15:40","Vcc":2.743,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:16:43 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:16:43","SR04":{"Distance":47}}
10:16:56 RUL: SR04#DISTANCE>40 performs "POWER OFF"
10:16:56 MQT: stat/home198/RESULT = {"POWER":"OFF"}
10:16:56 MQT: stat/home198/POWER = OFF
[... 45 lines cut ...]
10:31:16 MQT: stat/home198/POWER = OFF
10:31:35 RUL: SR04#DISTANCE>40 performs "POWER OFF"
10:31:35 MQT: stat/home198/RESULT = {"POWER":"OFF"}
10:31:35 MQT: stat/home198/POWER = OFF
10:31:45 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:31:45","Uptime":"0T01:30:42","Vcc":2.739,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:31:45 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:31:45","SR04":{"Distance":51}}
10:32:46 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:32:46","Uptime":"0T01:31:43","Vcc":2.741,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":62,"APMac":"2C:4D:54:83:7F:68"}}
10:32:46 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:32:46","SR04":{"Distance":72}}
10:33:46 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:33:46","Uptime":"0T01:32:43","Vcc":2.739,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":68,"APMac":"2C:4D:54:83:7F:68"}}
10:33:46 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:33:46","SR04":{"Distance":81}}
10:34:46 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:34:46","Uptime":"0T01:33:43","Vcc":2.744,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":68,"APMac":"2C:4D:54:83:7F:68"}}
10:34:46 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:34:46","SR04":{"Distance":79}}
10:35:46 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:35:46","Uptime":"0T01:34:43","Vcc":2.739,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:35:46 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:35:46","SR04":{"Distance":76}}
10:36:47 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:36:47","Uptime":"0T01:35:44","Vcc":2.738,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:36:47 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:36:47","SR04":{"Distance":72}}
10:37:47 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:37:47","Uptime":"0T01:36:44","Vcc":2.744,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":70,"APMac":"2C:4D:54:83:7F:68"}}
[... cut ...]
10:39:47 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:39:47","Uptime":"0T01:38:44","Vcc":2.742,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:39:47 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:39:47","SR04":{"Distance":104}}
10:40:47 MQT: tele/home198/STATE = {"Time":"2018-09-05T10:40:47","Uptime":"0T01:39:44","Vcc":2.738,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":66,"APMac":"2C:4D:54:83:7F:68"}}
10:40:47 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T10:40:47","SR04":{"Distance":87}}
```

arendst, 2018-09-05:
I guess you see the repeating values above 40 because it detects values below 40 but above 20 and then above 40 again. As long as the values below 40 or not below 20 you won't see them there.

mkh595, 2018-09-05:
Yes, I included STATUS 8 and result corresponds to what you wrote
```
12:16:22 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T12:16:22","SR04":{"Distance":92}}
12:16:32 MQT: tele/home198/STATE = {"Time":"2018-09-05T12:16:32","Uptime":"0T03:15:29","Vcc":2.741,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":68,"APMac":"2C:4D:54:83:7F:68"}}
12:16:32 MQT: tele/home198/SENSOR = {"Time":"2018-09-05T12:16:32","SR04":{"Distance":37}}
12:16:37 RUL: SR04#DISTANCE>40 performs "backlog POWER OFF; status 8"
12:16:37 MQT: stat/home198/RESULT = {"POWER":"OFF"}
12:16:37 MQT: stat/home198/POWER = OFF
12:16:37 MQT: stat/home198/STATUS8 = {"StatusSNS":{"Time":"2018-09-05T12:16:37","SR04":{"Distance":90}}}
12:16:43 MQT: tele/home198/STATE = {"Time":"2018-09-05T12:16:43","Uptime":"0T03:15:40","Vcc":2.741,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mkh","RSSI":72,"APMac":"2C:4D:54:83:7F:68"}}
```
Instead of command STATUS 8 is there command to trigger same MQTT mesage as for teleperiod?
```
tele/home198/SENSOR = {"Time":"2018-09-05T12:16:22","SR04":{"Distance":92}}
```
because HA subscribes topic 'tele/home198/SENSOR' and not 'stat/home198/STATUS8'

Issue is that teleperiod cannot be less that 10 sec so if distance is changed for less then teleperiod it is not sent to HA. But rules can do this. In this case I not need so small teleperiod.

ascillato, 2018-09-05:
> Instead of command STATUS 8 is there command to trigger same MQTT mesage as for teleperiod?

yes,

`cmnd/sonofftopic/teleperiod 10`

put 10 or the value you want to be teleperiod.

> But rules can do this.

Exactly.

If your measurements values are going up and down for whatever reason, the ONCE in rule will trigger your rules again. If you want you can add more logic to your rules to fit your needs.

closed 2018-09-05 by ascillato2

mkh595, 2018-09-07:
> Instead of command STATUS 8 is there command to trigger same MQTT mesage as for teleperiod?
> yes,
> ```cmnd/sonofftopic/teleperiod 10```

Unfortunately this is not solution to be used. 

```"backlog POWER ON; cmnd/sonofftopic/teleperiod 10"```

It generates teleperiod message but interval is now hardcoded in rule and will always owerwrite value added in WEB UI.

Instead of this is perfect solution is in PR #2619