arendst/Tasmota #3706: Sonoff Dual R2 change how to use buttons

jk2020, 2018-09-05:
Is it possible to:

Configure both, Button0 and Button1 to get a message to MQQT whenever button1 or button 2 has been pressed but those button must not toggle the switches of the device.

Get another "button3" (use of another GPIO address) to toggle switch1 and switch 2 together whenever that new "Button3" is pressed?

ascillato, 2018-09-05:
Yes, using rules. See examples in the wiki please

jk2020, 2018-09-05:
Many thanks for your answer
> Yes, using rules. See examples in the wiki please

Trust I need something like below to send a command via MQTT
`rule on button1#state do publish cmnd/ring2/power %value% endon on button2#state do publish cmnd/strip1/power %value% endon`

However any possibility to create another "button 3" which I could use to toggle switch1/switch2?

Many thanks for your answers

ascillato, 2018-09-05:
If you have added a third button in module config you can use it in rules.

You are saying a button to toggle switches? What that means? Both are inputs. 

* A button is a pushbutton. i.e. You push gives you 1, you release gives you 0
* A switch is a key like the ones in you home. You push goes to 1. You push again go to 0.

If you meant a third physical pushbutton to toggle 2 different relays output, just use in a rule the command backlog power1 2; power2 2

See wiki at commands and rules for more infomation.

jk2020, 2018-09-05:
> You are saying a button to toggle switches? What that means? Both are inputs.
You are right, sorry for confusion. By saying switches I ment relays.
So actually what I want is another physical push button to toggle both relays on the sonoff dual device.

Will follow your hint using wiki and try to find out how that could work, having in mind that I need to use following command.
`backlog power1 2; power2 2`

Please excuse my beginner questions but quite new to that thing but realize that combination of sonoff with tasmota is an exciting combination

ascillato, 2018-09-05:
There is no problem. You do not need to excuse yourself. I think that should be better and dynamic to continue with your question in the Tasmota Chat.

closed 2018-09-05 by ascillato2