arendst/Tasmota #3707: iFan02, status changes to fanspeed 1 before changing to 3 intermitently

chietanjain, 2018-09-05:
While using the iFan02 remote, if the following pattern is followed, it RESULTS in the fanspeed status turns to `{"FanSpeed":1}` and then` {"FanSpeed":3}`. The intermediate` {"FanSpeed":1} `may be coming up because of the way iFan02 switches internally, perhaps.

to reproduce this: On the iFan02 RF remote:
1 - hit speed 1, then
2 - hit speed 2, then
3 - hit speed 3, then
4 - hit speed 2, then
5 - hit speed 3, then
*repeat 4 and 5, two or three times, keeping the weblog open.

If the status log is required for this, pls do let me know.

Its not a big issue, and maybe I am the only one facing it too - so, no worries, no hurries.

thank you for Tasmota!!

ascillato, 2018-09-05:
Please, share all the information you have in order to help you properly. Status 0, weblog 4 and console output.

arendst, 2018-09-05:
It's the inbuilt microcontroller trying to keep current surge as low as possible. It does the fanspeed steps. I have no control over anything you do with the RF remote other than following what the microcontroller does.

As designed.

closed 2018-09-05 by ascillato2