arendst/Tasmota #3708: Enhancement: Support for TSL2591 to measures illuminance

dk4pa, 2018-09-05:
Hello, is it possible to add the light sensor TSL2591?
It interfaces with a BME280 sensor on the same board.

https://www.exp-tech.de/en/sensors/temperature/8819/bluedot-bme280-tsl2591-advanced-weather-station

Thanks in advance
Stefan

ascillato, 2018-09-05:
have you tried if that sensor works with the already supported TL2561?

dk4pa, 2018-09-05:
Yes, since he is also connected as I2C.

Two sensors are also detected with the CMD I2CScan.

0x77 is the BME280
0x29 is according to the datasheet of the TSL2591

MQT: stat / sonoff / RESULT = {"I2CScan": "Device (s) found at 0x29 0x77"}

However, only the values ​​from the BME280 sensor are displayed.

The TSL2561 was also activated in the config. #define USE_TSL2561

ascillato2, 2018-09-05:
The TL2561 is different from TL2591.
The TL2561 uses the library made by @joba-1

mike2nl, 2018-09-05:
@dk4pa 
You mean that one: Joba_Tsl2561
Link: https://github.com/joba-1/Joba_Tsl2561

dk4pa, 2018-09-06:
@mike2nl
No i mean TSL2591

Here the Link to Datasheet:
https://www.bluedot.space/app/download/11202619521/TSL25911_Datasheet_EN_v1.pdf?t=1518723516

Here the Link to Library:
https://www.bluedot.space/app/download/11217098221/BlueDot_BME280_TSL2591-master.zip?t=1525183872

ascillato2, 2018-11-16:
Hi, there is someone working on a driver for Tasmota for this?

ascillato2, 2018-11-21:
Hi,

Closing as there isn't anyone working on this.

Added label HOLD OVER for future reference.

If there is someone interested to work on this, we can reopen this issue.

Thanks everyone for sharing their ideas.

closed 2018-11-21 by ascillato2

CoolEDS, 2019-01-22:
@ascillato2 
Does it help any that Adafruit now also have the sensor?
https://learn.adafruit.com/adafruit-tsl2591
Thinking some of the code could help implementing

//EDS

ascillato, 2019-01-22:
Yes, that helps. :+1: thanks.

If there is someone interested to work on this, we can reopen this issue.

ascillato, 2021-04-13:
It has been added, see https://github.com/arendst/Tasmota/pull/6873

CoolEDS, 2021-04-28:
Hi Adrian

 

Yes, I have seen it and are using it already (but only on one position for now) and more will come. I am working on a system that will set/tune the sensitivity automatic depending on light level.

 

Thanks and regards

 

Erik