arendst/Tasmota #3709: settings.h setoption flags // why "user_esp8285_enable : 1;      // bit 31 (v6.1.1.14)" is added at the end

lobocobra, 2018-09-05:
**Describe the bug**
I develop  additional mods and want to keep them up to date with the master branch here. Those mods are used by others (like my shutter mod) and are implemented as drivers and activated with setoption.
Logically I assumed that the master branch will use the spare SETOPTIONS from TOP to DOWN, while   modders use them BOTTOM to UP.

I saw now in the latest version of the master branch, that SPARE00 AND SPARE31 are used for extensions.
=> This  makes it somehow impossible for modders to keep their mods up to date, if we do not know, which setoptions we can use. 
=> With this approach, we need to tell all end-users of the mods that the setoption82 can not be used and is now replaced by blablabla

Example.... following setoptions are now used
`    uint32_t timers_enable : 1;                  // bit 0 (v6.1.1b)`
`    uint32_t user_esp8285_enable : 1;      // bit 31 (v6.1.1.14)`

**Expected behavior**
I would be happy if the master branch uses spare variables like setoptions from TOP to DOWN so that modders can use them BOTTOM to TOP.

**Additional context**
I use SETOPTION80 to activate the SHUTTER driver and SETOPTION81 to enable INTERLOCK SPLIT.
=> As SETOPTION81 is now used by esp8285 and also SPARE00 is used, I am a bit lost, which SETOPTIONS I can use. I want to avoid to change the code each time.

Many thanks to give me at least a hint, how SETOPTIONS and other SPARE variables are used within the code.

arendst, 2018-09-06:
In fact nothing or all has been reserved for a user as he can change anything in his own repository.

Any variable can change as long as no feedback and reasonable request has been made. Others just ask for a location to be used and if that fits my future plans and the impact is not to great I reserve space for it.

You could have done that too and even provide a PR which might get implemented too.

For now I removed my claim for the last setoption location in the just released v6.2.1.2 and will continue at the top as other options did. This allows you to use the options from the bottom unless I need them in the future...

I suggest you keep using your bottom locations.

BTW: do not assume anything and just ask as I might not be that bad guy after all.

closed 2018-09-06 by arendst

lobocobra, 2018-09-06:
@arendst 
THANKS!

My apologies, I just realized that my request sounds quite harsh. This was not my intention (I was too concentrated to follow the template)  :-).

=> Before I make a PR, I want to make sure that the code is stable and that my 4 use cases (Split Interlock CH4 // Shutter // Watering // Oil heating) are covered with 1 driver and the code size is minimized.

meingraham, 2018-09-06:
@lobocobra 

Not sure you saw my previous suggestion, but this might be a nice enhancement. For a 4 channel devices, you either have no groups or two groups. Perhaps a parameter to specify the number of groups if there are more than 4 channels on the device. One then could specify which channels make up a group and the number of channels (or specify the channel numbers) in the group.

Regards.

Mike

lobocobra, 2018-09-06:
@meingraham 

I probably missed it... 
=> but the first part of your request is btw already implemened here:
https://github.com/lobocobra/sonoff-shutter

You can split 4 channels into 2 groups of 2 (activated by setoptions). If you see an use case for other splits (3+1 or or are aware about devices that have more than 4 channels, then let me know and I will implement a solution with MASKs and possibilities to split more than 4 channels.

meingraham, 2018-09-06:
@lobocobra 

The two and two grouping is one scenario. Logically one would group 1&2 and 3&4. But, for unforeseen reasons, one might select the channels in each group differently. I wouldn't, but someone might (say in a retrofit where they now what to take advantage of this new firmware feature but they had already wired something differently. Just surmising). One might want only one group and then the remaining channels ungrouped.

The real reason this occured to me was a group of three scenario. I have this scenario for my ceiling fans. Three channels are used to control the speed of the fan. The interlocking varies from manufacturer to manufacturer.

Full interlock:
A - Low
B - Medium
C - High

Hybrid interlock:
A - Low
A&B - Medium
C - High

I use the fourth channel to control the ceiling fan light. This channel operates independently from the other three.

A part of the group configuration would be to specify which channels make up the group. Thus, if one has the ability to specify which three channels to group, then this same capability could be used in the 2/2 grouping to specify which channels are in each group. One could extend this thinking to a 5Ch switch (still two groups at most).

Cheers!

Mike

ascillato, 2018-09-06:
Please, add this very useful information to the correspondent issue https://github.com/arendst/Sonoff-Tasmota/issues/824

stefanbode, 2018-09-06:
Hi Mike, you request a lot of flexibility that has a major impact on the code. Any additions have to be ranked for the benefits and the additional amount of code they generate because the ESP8266 is really limited here and at the end, every byte counts. I found that a paired interlock is more or less easy to integrate to pair 1&2, 3&4 and so on. Additionally putting a mask on the relays like 11110000 to have the interlock only on the first four relays and the rest works independently is also doable. 
Defining Interlock[1] and then putting relay 1,3,6 in it, for example, is a nightmare in code because we use quite a lot of binary operations with XOR, OR and so on. Sometimes you have to rewire :-)

meingraham, 2018-09-06:
@stefanbode 

I can't disagree... I would re-wire to keep things "intuitive". But that's me. Like you said, it makes sense to keep two groups of paired sequential channels are the coded approach.

But then you come to my real reason for posing the suggestion - the three channel "group" which varies according to use case (e.g., fan manufacturer). Thus, if this enhanced capability would actually be coded in the firmware, "masking" would be required... and if it's there for a three-channel group... it might as well be there for a two-channel group as well.

Mike

stefanbode, 2018-09-06:
Hi Mike,
let me think about the 2,3,4 group. This is one additional configuration parameter and might be not that difficult to integrate. Maybe easier than I think