arendst/Tasmota #3713: Sonoff Tasmota 6.2 on Sonoff Pro 4CH R2

dastrix80, 2018-09-06:
Hi Team

Running this version:

Program Version | 6.2.0
-- | --
2018-08-31T15:15:32
2_3_0/1.5.3(aec24ac9)

And upon wifi loss or power loss, it will trigger the dry contact relay and as a result, open my gate! Not ideal

Can anyone suggest why this is occuring?

Thanks
Kris

andrethomas, 2018-09-06:
Where is your status 0 output?

Frogmore42, 2018-09-06:
Without the information of the troubleshooting template it really is not possible to help you, without an extreme amount of guessing.

My guess is that you do have a retained message.  The startup web log will show if I am right about that or not.

I am curious though.  You are not the first and won't be the last person to ignore the troubleshooting template.
Why did you choose to ignore it?
Did you not see it?
Was it too complicated to fill out?
Did you not want to provide the information?
Did you think it was not necessary?
Something else?

Some projects will immediately close any issue that misses even a small part of their template, or is even slightly off topic.  That is not the case here, but it really does make it harder for everyone when the template information is missing.

dastrix80, 2018-09-06:
Sorry I dont know what the status 0 output is. 

How would I see retained messages? A restart shows this:

00:00:00 Project sonoff Gate_Garage (Topic gate, Fallback gate, GroupTopic sonoffs) Version 6.2.0-2_3_0
00:00:00 RSL: stat/gate/RESULT = {"POWER4":"ON"}
00:00:00 RSL: stat/gate/POWER4 = ON
00:00:00 RSL: stat/gate/RESULT = {"POWER4":"ON"}
00:00:00 RSL: stat/gate/POWER4 = ON
00:00:00 WIF: Connecting to AP1 Tard in mode 11N as gate-4157...
00:00:01 RSL: stat/gate/RESULT = {"POWER4":"OFF"}
00:00:01 RSL: stat/gate/POWER4 = OFF
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on gate-4157.local with IP address 192.168.0.239
18:12:19 RSL: stat/gate/RESULT = {"POWER1":"ON"}
18:12:19 RSL: stat/gate/POWER1 = ON
18:12:19 MQT: Attempting connection...
18:12:19 MQT: Connected
18:12:19 MQT: tele/gate/LWT = Online (retained)
18:12:19 MQT: cmnd/gate/POWER = 
18:12:19 MQT: tele/gate/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"6.2.0","FallbackTopic":"gate","GroupTopic":"sonoffs"}
18:12:19 MQT: tele/gate/INFO2 = {"WebServerMode":"Admin","Hostname":"gate-4157","IPAddress":"192.168.0.239"}
18:12:19 MQT: tele/gate/INFO3 = {"RestartReason":"Software/System restart"}

andrethomas, 2018-09-06:
Type status 0 into the console and press enter

dastrix80, 2018-09-06:
I just did a restart and the darn gate just opened and my dog went straight out! F!!!!

andrethomas, 2018-09-06:
Looks like you have setoption0 enabled though, but status 0 will tell us

dastrix80, 2018-09-06:
Status 0

18:15:46 CMD:  status 0
18:15:46 MQT: stat/gate/STATUS = {"Status":{"Module":23,"FriendlyName":["Gate_Garage","gate","gate","gate"],"Topic":"gate","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:15:46 MQT: stat/gate/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:31","StartupUTC":"2018-09-06T08:12:15","Sleep":0,"BootCount":269,"SaveCount":2664,"SaveAddress":"F5000"}}
18:15:46 MQT: stat/gate/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-08-31T15:15:32","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:15:46 MQT: stat/gate/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Tard","indebuurt2"],"TelePeriod":300,"SetOption":["00002009","55818000","00000001"]}}
18:15:46 MQT: stat/gate/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
18:15:46 MQT: stat/gate/STATUS5 = {"StatusNET":{"Hostname":"gate-4157","IPAddress":"192.168.0.239","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"60:01:94:A1:90:3D","Webserver":2,"WifiConfig":3}}
18:15:46 MQT: stat/gate/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.3","MqttPort":1883,"MqttClientMask":"gate","MqttClient":"gate","MqttUser":"mosquitto","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:15:46 MQT: stat/gate/STATUS7 = {"StatusTIM":{"UTC":"Thu Sep 06 08:15:46 2018","Local":"Thu Sep 06 18:15:46 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10,"Sunrise":"15:14","Sunset":"04:23"}}
18:15:46 MQT: stat/gate/STATUS10 = {"StatusSNS":{"Time":"2018-09-06T18:15:46","Switch4":"ON"}}
18:15:46 MQT: stat/gate/STATUS11 = {"StatusSTS":{"Time":"2018-09-06T18:15:46","Uptime":"0T00:03:31","Vcc":3.066,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Tard","RSSI":100,"APMac":"28:CF:DA:B7:E0:99"}}}

andrethomas, 2018-09-06:
"SaveState":1 and "PowerOnState":3 is causing this

Do:

setoption0 0
PowerOnState 0

arendst, 2018-09-06:
Pls change wificonfig too to 5 to stop reboots on wifi loss.

dastrix80, 2018-09-06:
thank you! ill change that too

andrethomas, 2018-09-06:
No sweat, glad you got sorted!

ascillato2, 2018-10-21:
Please, update to last Tasmota from http://thehackbox.org/tasmota/

ascillato2, 2018-10-21:
Also please, provide the output of your console when this occurs. Please, use command weblog 4 in order to have more debug information. Thanks

dastrix80, 2018-10-21:
OK, ive enabled that and upgraded to the latest from your link above.

I just restarted it and it didnt do it...

```
00:00:00 CFG: Loaded from flash at F9, Count 66
00:00:00 APP: Boot Count 40
00:00:00 Project sonoff GateControl (Topic gate, Fallback DVES_462E34, GroupTopic sonoffs) Version 6.2.1.18-2_3_0
00:00:00 CFG: Saved to flash at F8, Count 67, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 Tard in mode 11N as gate-3636...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on gate-3636.local with IP address 192.168.0.239
00:00:03 APP: (UTC) Sun Oct 21 02:36:56 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
03:36:57 MQT: Attempting connection...
03:36:57 MQT: Connected
03:36:57 MQT: tele/gate/LWT = Online (retained)
03:36:57 MQT: cmnd/gate/POWER = 
03:36:57 MQT: Subscribe to cmnd/gate/#
03:36:57 MQT: Subscribe to cmnd/sonoffs/#
03:36:57 MQT: Subscribe to cmnd/DVES_462E34/#
03:36:57 MQT: tele/gate/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"6.2.1.18","FallbackTopic":"DVES_462E34","GroupTopic":"sonoffs"}
03:36:57 MQT: tele/gate/INFO2 = {"WebServerMode":"Admin","Hostname":"gate-3636","IPAddress":"192.168.0.239"}
03:36:57 MQT: tele/gate/INFO3 = {"RestartReason":"Software/System restart"}
03:36:57 MQT: stat/gate/RESULT = {"POWER1":"OFF"}
03:36:57 MQT: stat/gate/POWER1 = OFF
03:36:57 MQT: stat/gate/RESULT = {"POWER2":"OFF"}
03:36:57 MQT: stat/gate/POWER2 = OFF
03:36:57 MQT: stat/gate/RESULT = {"POWER3":"OFF"}
03:36:57 MQT: stat/gate/POWER3 = OFF
03:36:57 MQT: stat/gate/RESULT = {"POWER4":"OFF"}
03:36:57 MQT: stat/gate/POWER4 = OFF
03:36:57 HTP: Main Menu
```

dastrix80, 2018-10-21:
Thanks Adrian, ill do that and report back.

andrethomas, 2018-10-21:
Just to make sure - you did not by accident learn a remote control for the relays?

ascillato2, 2018-10-21:
Hi,

Please, when you have time, please tell us if your tests were ok. Thanks

closed 2018-10-21 by ascillato2

dastrix80, 2018-10-21:
No i didnt accidentally learn a remote.

andrethomas, 2018-10-22:
Did you try restart / shut down mqtt server to see if it will reproduce the behaviour as @ascillato requested?

dastrix80, 2018-10-22:
Yes I did, it didnt restart. However when programming the GPIO2 yesterday and restarting the unit, it did restart and trigger the door. Its difficult to reproduce.

dastrix80, 2019-02-17:
Im restarting the unit and its still doing it. I have to turn the Sonoff OFF to prevent this. Huge PITA

11:15:27 MQT: stat/gate/STATUS = {"Status":{"Module":23,"FriendlyName":["GateControl","GateControl2","GateControl3","GateControl4"],"Topic":"gate","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"SENSOR","SwitchMode":[0,0,0,1,0,0,0,2],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:15:27 MQT: stat/gate/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/GarageController.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:19","StartupUTC":"2019-02-17T10:14:08","Sleep":0,"BootCount":59,"SaveCount":102,"SaveAddress":"F5000"}}
11:15:27 MQT: stat/gate/STATUS2 = {"StatusFWR":{"Version":"6.2.1.18","BuildDateTime":"2018-10-20T18:00:42","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:15:27 MQT: stat/gate/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Tard","indebuurt2"],"TelePeriod":300,"SetOption":["00000008","55818000","00000001"]}}
11:15:27 MQT: stat/gate/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00001BC0"]}}
11:15:28 MQT: stat/gate/STATUS5 = {"StatusNET":{"Hostname":"gate-3636","IPAddress":"192.168.0.239","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"84:0D:8E:46:2E:34","Webserver":2,"WifiConfig":5}}
11:15:28 MQT: stat/gate/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.3","MqttPort":1883,"MqttClientMask":"Gate","MqttClient":"Gate","MqttUser":"mosquitto","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:15:28 MQT: stat/gate/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 17 10:15:28 2019","Local":"Sun Feb 17 11:15:28 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":1,"Sunrise":"07:55","Sunset":"18:13"}}
11:15:28 MQT: stat/gate/STATUS10 = {"StatusSNS":{"Time":"2019-02-17T11:15:28","Switch8":"OFF"}}
11:15:28 MQT: stat/gate/STATUS11 = {"StatusSTS":{"Time":"2019-02-17T11:15:28","Uptime":"0T00:01:20","Vcc":2.982,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Tard","BSSId":"FC:EC:DA:D1:6D:D7","Channel":11,"RSSI":100}}}
11:15:31 WIF: Checking connection...

d0v1k, 2019-02-22:
try to power off and on restart via webgui trigger the relay.

andrethomas, 2019-02-22:
Use of GPIO2 is somewhat limited - See https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#the-esp8266---hardware-more-info for more information.