arendst/Tasmota #3714: Fade in PWM mode

marekzehra, 2018-09-06:
Hi,
would it be possible to allow fade command also in PWM mode (SetOption15 0)?

I'm using H801 to control two white led strips (W1 and W2). Fade command is not found until SetOption15 is ON.

arendst, 2018-09-06:
Not at the moment.

closed 2018-09-06 by arendst

albzn540, 2019-02-26:
Will it be added in the feature?

vstonkus, 2019-10-10:
Currently I'm testing a prototype of using H801 for multiple led strips in my house. Fade looks awesome, but it only works if Option15 is turned ON. However multiple strips doesn't work well with Option15 enabled - if I use any RGB channel, it turns off white channels.

Povilas1, 2019-11-21:
Can we expect this feature implemented in near future? It would be handy to use one H801 module for all my led strips.

s-hadinger, 2019-11-21:
Yes, work in progress, see #6942