arendst/Tasmota #3715: Restart for blocked loop on update

jpadie, 2018-09-06:
**Describe the bug**
module flased with pre-compiled bin yesterday.
buggy knx so try to recompile and reflash today.
web update does not work.  restart reason given as blocked loop.

I'm just reporting this bug.  Not looking for a resolution for me, nor a workaround (OTA download works)

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ :  ifan 02, mac to compile.  web to upload
- [x ] _Provide the output of command_``status 0`` :
```
{"Status":{"Module":44,"FriendlyName":["Sonoff"],"Topic":"fan","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
{"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Blocked Loop","Uptime":"0T00:03:40","StartupUTC":"2018-09-06T07:48:19","Sleep":0,"BootCount":16,"SaveCount":28,"SaveAddress":"FA000"}}
{"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-05T10:41:59","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gaia-Rabastens-Free",""],"TelePeriod":300,"SetOption":["0A008009","55818000","00000000"]}}
{"StatusMEM":{"ProgramSize":491,"Free":512,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","1FDAE594","24000010","23B617CE","00000000"]}}
{"StatusNET":{"Hostname":"fan-3708","IPAddress":"192.168.0.19","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"84:0D:8E:42:CE:7C","Webserver":2,"WifiConfig":2}}
{"StatusMQT":{"MqttHost":"***","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_42CE7C","MqttUser":"***","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
{"StatusTIM":{"UTC":"Thu Sep 06 07:52:00 2018","Local":"Thu Sep 06 08:52:00 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:14","Sunset":"19:23"}}
{"StatusSNS":{"Time":"2018-09-06T08:52:00"}}
{"StatusSTS":{"Time":"2018-09-06T08:52:00","Uptime":"0T00:03:41","Vcc":3.182,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"Gaia-Rabastens-Free","RSSI":68,"APMac":"E4:9E:12:A8:20:38"}}}
```

**To Reproduce**
as above

**Expected behavior**
clean upload

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-06:
Hi,

Thanks for the report.

I have just tested OTA with last v6.2.1.2 and my output is `tele/living/INFO3 = {"RestartReason":"Software/System restart"}`

Please, can you try again with last version and report your findings? Thanks

jpadie, 2018-09-06:
This was not an ota download (which worked) but an ota upload.  

It failed on the standard pre compiled binary of yesterday.  

I have no easy way to reproduce the bug (although I repeated it a half dozen times) as I reflashed the module using an ota download.

ascillato, 2018-09-06:
What is the difference you state with ota download and ota upload ?

closed 2018-09-06 by ascillato2

jpadie, 2018-09-06:
OTA Download worksOTA upload gives the fatal exception.

ascillato2, 2018-09-06:
Ok, but what is the difference between OTA download and OTA upload?

OTA is Over The Air, and it is just one type. It is just sending a BIN file to your device by WIFI and saving it in the flash memory.

There is 2 easy ways of putting a software in ESP8266:
* OTA (over the air), using wifi
* Serial. using a cable and adapter.

jpadie, 2018-09-06:
I see your confusion. And this is why I was distinguishing between upload and download. The former is when the web server is used.  A push mechanism. The latter is when the sonoff fetches the file from otaurl.  A pull mechanism. The former scenario failed with yesterday's bin installed, trying to upload a bin I compiled myself; as I posted earlier.  It failed every time.  Without exception. Giving a fatal exception and spawning a restart.  Using the same binary that I compiled this morning, copied to a webserver, the latter method was successful.Incidentally I am  also getting intermittent failures setting knx addresses via the web interface.  Annoyingly sometimes it works and sometimes it does not.  So nothing objectively reproduceable.

ascillato2, 2018-09-06:
Ah, Ok. you were refering to OTA using OTAurl and OTA using file. Ok

curzon01, 2018-09-06:
Be aware if you handle bin file size greater >512kb you have to deal manual correctly using x-minimal.bin when you upload a bin using the web interface.

OtaMagic handles this automatically when using upgrade via an url AND a correct matching minimal bin version is located on same url.

All this can fail if you use url upgrade and OtaMagic is starting (because your bin on your Sonoff or the new file is >512 kb) and the files are mixed in different versions.

Conclusion: 
If your bin size on device or the new is greater than 512 kb and
 - you use Upgrade via url, be aware always having a matching x-minimal.bin version on same url location
 - you using upload via Web interface, first use a matching x-minimal.bin, then use your final.bin

jpadie, 2018-09-06:
Pre compiled binary was 477kb. My compiled bin was 434kb