arendst/Tasmota #3724: Enhancement: Nova Fitness SDS011 Dust sensor implement working period command

edelstahlratte, 2018-09-07:
Because the sensor has a limited lifespan of abt. 8000 hours ist would be nice to have the 'working period' command implemented.

Working period means that in a given timeframe the sensor is only active for 30 seconds and goes to sleep after that.

I'm not a C++ programmer, so it is very hard for me to do it on my own :(

The [SDS Control Protocol Specification](https://cdn.sparkfun.com/assets/parts/1/2/2/7/5/Laser_Dust_Sensor_Control_Protocol_V1.3.pdf) states the following:

[image]

So it looks that you only have to send these 19 bytes to set the feature on the SDS011.

Maybe someone can evaluate if it's possible to use that feature in tasmota.

Thank you.

mike2nl, 2018-09-07:
@edelstahlratte 
Have you looked here?
-> https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-Nova-Fitness-SDS011-Laser-Dust-Sensor

edelstahlratte, 2018-09-07:
@mike2nl

Yes, I read it and the sensor is working fine, but the 'working period' command would give the sensor a longer lifespan...

edelstahlratte, 2018-09-07:
@mike2nl

Maybe you can help:

I added some code to tasmota but I'm flying blind here...

e.g. in the xns_20_novasds.ino:

```
uint8_t novasds_workperiod[19] = {
	0xAA,	// head
	0xB4,	// command id
	0x08,	// data byte 1
	0x01,	// data byte 2 set
	0x05,	// data byte 3 work period 5 minutes: measure for 30 seconds + sleep the rest
	0x00,	// data byte 4
	0x00,	// data byte 5
	0x00,	// data byte 6
	0x00,	// data byte 7
	0x00,	// data byte 8
	0x00,	// data byte 9
	0x00,	// data byte 10
	0x00,	// data byte 11
	0x00,	// data byte 12
	0x00,	// data byte 13
	0xFF,	// data byte 14 (device id byte 1)
	0xFF,	// data byte 15 (device id byte 2)
	0x07,	// checksum (low 8 bit of sum of data bytes)
	0xAB	// tail
};

void NovaSdsSetWorkPeriod()
{
  //maybe there is some data
  while (NovaSdsSerial->available() > 0) {
      NovaSdsSerial->read();
  }

 NovaSdsSerial->flush();
 
 NovaSdsSerial->write(novasds_workperiod, sizeof(novasds_workperiod));
 
 // read the answer and ignore
 while (NovaSdsSerial->available() > 0) {
     NovaSdsSerial->read();
 }
}
```

I call the function in NovaSdsInit();

But it does not have any effect... Maybe the CRC wrong? Is the send command right? Don't know... How to get some output to the webconsole?

Thank you.

edelstahlratte, 2018-09-07:
I'm making progress:

The checksum was wrong:

`0x0C,	// checksum (low 8 bit of sum of data bytes)`
This is right...

The sensor will be active for 30 seconds every 5 Minutes...

But I have to understand Tasmota better to really integrate it... Digging further...

Frogmore42, 2018-09-07:
You might want to check out this PR
https://github.com/arendst/Sonoff-Tasmota/pull/2841
I believe the enhancement you seek is already available.

edelstahlratte, 2018-09-07:
@Frogmore42 

You are totally right! But the changes are not merged into the development tree... That's why I didn't find them in the code...

My approach was a litttle different but it achieves the same thing...
At least I have learnt more about tasmota and C++ in general :)

edelstahlratte, 2018-09-07:
@ascillato2

Will the changes from #2841 be merged into development?

ascillato, 2018-09-07:
Please, ask Theo. @arendst

arendst, 2018-09-07:
Have to look at it. The way it is done currently will upset people as their GPIOs will change....

edelstahlratte, 2018-09-07:
@arendst

Could you please specify further? You mean bécause of the new `GPIO_SDS0X1_TX` and the renamed `GPIO_SDS0X1`?

arendst, 2018-09-07:
The additional GPIO define will shift all GPIOs below one position down resulting in changed GPIO definitions.

edelstahlratte, 2018-09-07:
@arendst

Thank you, that clarifies a lot...

edelstahlratte, 2018-09-07:
@arendst

Would it be a solution for the GPIO Problem to hardcode the ` GPIO_TXD ` as serial transmit?

arendst, 2018-09-08:
Yeah, also a good idea.

I'll focus on a solution like I use for the modules. I'll investigate a sorted list. Once done will pick this up.

edelstahlratte, 2018-09-10:
@arendst

The last measurement is saved in the SDS011. So when queried it presents the saved data without waking up... 

It only wakes up to the specified 'working period' interval and saves the new measuremet internal... I have it running  since 2 days without problems...

arendst, 2018-09-10:
That's very useful!

Well if you can make a working PR and try to support this specific solution I will replace the current driver with yours.

edelstahlratte, 2018-09-10:
It's only a quite small change in the novasds.ino, because all the `GPIO_SDS0X1_TX` stuff is now already implemented... First I have to figure out how to make a proper pull request ;)

edelstahlratte, 2018-09-10:
@arendst

I just created pull request #3749... Hope that it fits the bill... My first one :)

closed 2018-09-11 by edelstahlratte