arendst/Tasmota #3726: Sonoff T1 UK problems after flashing Tasmota

L00000813, 2018-09-07:
Hi,

I had installed a T1 Uk 2 gang switch in my home and it functioned as expected. I then flashed the latest Tasmota onto the switch and fitted the switch back in the wall. I changed the configuration to the Sonoff T1 2CH.

The problems are:
The blue Wi-Fi LED (signalling a connection) should be constant when the switch has a connection - but the LED goes out when I switch either of the gangs on. The Wi-Fi LED comes back on when I switch the gang off again.

The switch is very not responsive by touch - Sometimes when pressing either of the gangs (switches) manually nothing happens. Even after several presses nothing happens.

Controlling (toggling) each gang through the web interface works better but sometimes it is not responsive either.

I suspect I have settings wrong somewhere causing this system to act strange.

Thanks in advance for the help.. 

```
STATUS 0 OUTPUT HERE
```
00:00:00 Project A2 Sonoff (Topic A2, Fallback DVES_81F75C, GroupTopic sonoffs) Version 6.2.1.2-2_4_0_RC2
00:00:00 WIF: Connecting to AP1 Wi-Fi-Raspberry in mode 11N as A2-5980...
00:00:10 WIF: Connected
00:00:10 DNS: Initialized
00:00:10 HTP: Web server active on A2-5980.local with IP address 192.168.1.2
00:00:10 MQT: Attempting connection...
00:00:10 MQT: Connected
00:00:10 MQT: tele/A2/LWT = Online (retained)
00:00:10 MQT: cmnd/A2/POWER = 
00:00:10 MQT: tele/A2/INFO1 = {"Module":"Sonoff T1 2CH","Version":"6.2.1.2","FallbackTopic":"DVES_81F75C","GroupTopic":"sonoffs"}
00:00:10 MQT: tele/A2/INFO2 = {"WebServerMode":"Admin","Hostname":"A2-5980","IPAddress":"192.168.1.2"}
00:00:10 MQT: tele/A2/INFO3 = {"RestartReason":"Software/System restart"}
00:00:10 MQT: stat/A2/RESULT = {"POWER1":"ON"}
00:00:10 MQT: stat/A2/POWER1 = ON
00:00:10 MQT: stat/A2/RESULT = {"POWER2":"OFF"}
00:00:10 MQT: stat/A2/POWER2 = OFF
00:00:19 MQT: tele/A2/STATE = {"Time":"1970-01-01T00:00:19","Uptime":"0T00:00:19","Vcc":3.461,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Wi-Fi-Raspberry","BSSId":"DC:09:4C:A8:6B:16","Channel":4,"RSSI":80}}
00:02:00 MQT: tele/A2/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:00"}
00:03:21 CMD: status 0
00:03:21 MQT: stat/A2/STATUS = {"Status":{"Module":29,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"A2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:03:21 MQT: stat/A2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:21","StartupUTC":"","Sleep":0,"BootCount":9,"SaveCount":49,"SaveAddress":"FB000"}}
00:03:21 MQT: stat/A2/STATUS2 = {"StatusFWR":{"Version":"6.2.1.2","BuildDateTime":"2018-09-06T19:20:31","Boot":6,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
00:03:21 MQT: stat/A2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wi-Fi-Raspberry","eir46325537-2.4G"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:03:21 MQT: stat/A2/STATUS4 = {"StatusMEM":{"ProgramSize":511,"Free":492,"Heap":9,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
00:03:21 MQT: stat/A2/STATUS5 = {"StatusNET":{"Hostname":"A2-5980","IPAddress":"192.168.1.2","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:81:F7:5C","Webserver":2,"WifiConfig":5}}
00:03:21 MQT: stat/A2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_81F75C","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:03:21 MQT: stat/A2/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:03:21 1970","Local":"Thu Jan 01 00:03:21 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:03:21 MQT: stat/A2/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:03:21"}}
00:03:21 MQT: stat/A2/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:03:21","Uptime":"0T00:03:21","Vcc":3.460,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Wi-Fi-Raspberry","BSSId":"DC:09:4C:A8:6B:16","Channel":4,"RSSI":78}}}

Frogmore42, 2018-09-07:
You are using core 2.4.0_RC2.  All of the 2.4.0 releases are known to be bad.  Look at this section of the wiki and GB referenced issues for solutions.
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-fw--512x-expressif-versions

ascillato2, 2018-09-07:
Please, use esplib 2.4.2 and change the MQTT library (in user_config.h) to TASMOTA_MQTT to avoid those issues.

Thanks

closed 2018-09-07 by ascillato2

L00000813, 2018-09-08:
I have made the changes described above by Frogmore42 and ascillato2.

Things are acting stranger than ever.

I cannot change the device configuration to Sonoff T1 2CH. Below is the output from the web interface:

Configure Module Output:

turn document.getElementById(s);}function u(){if(cn>=0){eb('t').innerHTML='Restart in '+cn+' seconds';cn--;setTimeout(u,1000);}}function c(l){eb('s1').value=l.innerText||l.textContent;eb('p1').focus();}function lx(){if(to==1){if(tp<30){tp++;lt=setTimeout(lx,33);}else{lt=setTimeout(la,1355);}return;}if(x!=null){x.abort();}x=new XMLHttpRequest();x.onreadystatechange=function(){if(x.readyState==4&&x.status==200){var s=x.responseText.replace(/{t}/g,"").replace(/{s}/g,"").replace(/{c}/g,"%'>
<="" option="" style="color: rgb(0, 0, 0); font-family: "Times New Roman"; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-style: initial; text-decoration-color: initial; text-align: center;">");sk(0,99);o0="}1'0'>00 None}2}1'1'>01 DHT11}2}1'2'>02 AM2301}2}1'3'>03 SI7021}2}1'4'>04 DS18x20}2}1'5'>05 I2C SCL}2}1'6'>06 I2C SDA}2}1'7'>07 WS2812}2}1'8'>08 IRsend}2}1'9'>09 Switch1}2}1'10'>10 Switch2}2}1'11'>11 Switch3}2}1'12'>12 Switch4}2}1'13'>13 Switch5}2}1'14'>14 Switch6}2}1'15'>15 Switch7}2}1'16'>16 Switch8}2}1'18'>18 Button2}2}1'19'>19 Button3}2}1'20'>20 Button4}2}1'22'>22 Relay2}2}1'23'>23 Relay3}2}1'24'>24 Relay4}2}1'25'>25 Relay5}2}1'26'>26 Relay6}2}1'27'>27 Relay7}2}1'28'>28 Relay8}2}1'30'>30 Relay2i}2}1'31'>31 Relay3i}2}1'32'>32 Relay4i}2}1'33'>33 Relay5i}2}1'34'>34 Relay6i}2}1'35'>35 Relay7i}2}1'36'>36 Relay8i}2}1'37'>37 PWM1}2}1'38'>38 PWM2}2}1'39'>39 PWM3}2}1'40'>40 PWM4}2}1'41'>41 PWM5}2}1'42'>42}1'59'>59 Led4i}2}1'60'>60 MHZ Tx}2}1'61'>61 MHZ Rx}2}1'62'>62 PZEM Tx}2}1'63'>63 PZEM Rx}2}1'64'>64 SAir Tx}2}1'65'>65 SAir Rx}2}1'69'>69 PMS5003}2}1'70'>70 SDS0X1}2}1'71'>71 SerBr Tx}2}1'72'>72 SerBr Rx}2}1'73'>73 SR04 Tri}2}1'74'>74 SR04 Ech}2}1'82'>82 Switch1n}2}1'83'>83 Switch2n}2}1'84'>84 Switch3n}2}1'85'>85 Switch4n}2}1'86'>86 Switch5n}2}1'87'>87 Switch6n}2}1'88'>88 Switch7n}2}1'89'>89 Switch8n}2}1'90'>90 Button1n}2}1'91'>91 Button2n}2}1'92'>92 Button3n}2}1'93'>93 Button4n}2}1'94'>94 Counter1n}2}1'95'>95 Counter2n}2}1'96'>96 Counter3n}2}1'97'>97 Counter4n}2";os=o0.replace(/}1/g,"<="" option="">
").replace(/{m}/g,"

	").replace(/{e}/g,"

Status O output:
00:00:00 Project A2 Sonoff (Topic A2, Fallback DVES_81F75C, GroupTopic sonoffs) Version 6.2.1.2-2_4_0_RC2
00:00:00 WIF: Connecting to AP1 Wi-Fi-Raspberry in mode 11N as A2-5980...
00:00:10 WIF: Connected
00:00:10 DNS: Initialized
00:00:10 HTP: Web server active on A2-5980.local with IP address 192.168.1.2
00:00:10 MQT: Attempting connection...
00:00:10 MQT: Connected
00:00:10 MQT: tele/A2/LWT = Online (retained)
00:00:10 MQT: cmnd/A2/POWER = 
00:00:10 MQT: tele/A2/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.1.2","FallbackTopic":"DVES_81F75C","GroupTopic":"sonoffs"}
00:00:10 MQT: tele/A2/INFO2 = {"WebServerMode":"Admin","Hostname":"A2-5980","IPAddress":"192.168.1.2"}
00:00:10 MQT: tele/A2/INFO3 = {"RestartReason":"Software/System restart"}
00:00:10 MQT: stat/A2/RESULT = {"POWER":"OFF"}
00:00:10 MQT: stat/A2/POWER = OFF
00:00:19 MQT: tele/A2/STATE = {"Time":"1970-01-01T00:00:19","Uptime":"0T00:00:19","Vcc":3.460,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wi-Fi-Raspberry","BSSId":"DC:09:4C:A8:6B:16","Channel":5,"RSSI":84}}
00:00:33 CMD: status 0
00:00:33 MQT: stat/A2/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"A2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:33 MQT: stat/A2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:33","StartupUTC":"","Sleep":0,"BootCount":1,"SaveCount":2,"SaveAddress":"FA000"}}
00:00:33 MQT: stat/A2/STATUS2 = {"StatusFWR":{"Version":"6.2.1.2","BuildDateTime":"2018-09-08T10:28:26","Boot":6,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
00:00:33 MQT: stat/A2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wi-Fi-Raspberry","eir46325537-2.4G"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:00:33 MQT: stat/A2/STATUS4 = {"StatusMEM":{"ProgramSize":515,"Free":488,"Heap":9,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAEB94","000003A0","23B617CE","00000000"]}}
00:00:33 MQT: stat/A2/STATUS5 = {"StatusNET":{"Hostname":"A2-5980","IPAddress":"192.168.1.2","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:81:F7:5C","Webserver":2,"WifiConfig":5}}
00:00:33 MQT: stat/A2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_81F75C","MqttUser":"","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:33 MQT: stat/A2/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:33 1970","Local":"Thu Jan 01 00:00:33 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:33 MQT: stat/A2/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:33"}}
00:00:33 MQT: stat/A2/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:33","Uptime":"0T00:00:33","Vcc":3.460,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wi-Fi-Raspberry","BSSId":"DC:09:4C:A8:6B:16","Channel":5,"RSSI":78}}}

Frogmore42, 2018-09-08:
Your output from status 0 indicates you are still using core 2.4.0 RC2

{"Version":"6.2.1.2","BuildDateTime":"2018-09-08T10:28:26","Boot":6,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}

Please go read the wiki again on how to change core versions.  You are using a doubly known bad version.  It was replaced by the final version of 2.4.0, which was replaced by 2.4.1 (and it's RCs) and again by 2.4.2 (and it's RCs).  The issue you are seeing is caused by the defective core WiFi code.

L00000813, 2018-09-12:
Can you provide me correct details on how to change to core 2.4.2

This is the instructions I am following ...
HowTo Change: Atom&PlatformIO: Locate the platformio.ini file in the project directory of TASMOTA Change the following:

[common] ; ************************************************************
; *** Esp8266 core for Arduino version
platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266@1.7.0  ; v2.4.1
;platform = espressif8266

...

I make the changes to the platformio.ini file, compile and upload to switch but its still using 2.4.0 ??

Thanks..

Frogmore42, 2018-09-12:
You need to do a clean and rebuild probably.  That will remove everything that has already been compiled and recompile everything.

L00000813, 2018-09-18:
Still facing this issue with the T1 UK 2 gang switch...

The blue Wi-Fi LED (signalling a connection) should be constant when the switch has a Wi-Fi and MQTT connection - but the LED goes out when I switch either of the gangs on. The Wi-Fi LED comes back on when I switch the gang off again.

10:37:32 CMD: status 0
10:37:32 MQT: stat/A2/STATUS = {"Status":{"Module":29,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"A2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:37:32 MQT: stat/A2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:02:55","StartupUTC":"2018-09-18T09:34:37","Sleep":0,"BootCount":10,"SaveCount":33,"SaveAddress":"FB000"}}
10:37:32 MQT: stat/A2/STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018-09-18T10:27:10","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
10:37:32 MQT: stat/A2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["eir46325537-2.4G","Wi-Fi-Raspberry"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
10:37:32 MQT: stat/A2/STATUS4 = {"StatusMEM":{"ProgramSize":509,"Free":492,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAEB94","000003A0","23B617CE","000002C0"]}}
10:37:32 MQT: stat/A2/STATUS5 = {"StatusNET":{"Hostname":"A2-5980","IPAddress":"192.168.1.4","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"DC:4F:22:81:F7:5C","Webserver":2,"WifiConfig":5}}
10:37:32 MQT: stat/A2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.9","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_81F75C","MqttUser":"","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:37:32 MQT: stat/A2/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 18 09:37:32 2018","Local":"Tue Sep 18 10:37:32 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:31","Sunset":"18:57"}}
10:37:32 MQT: stat/A2/STATUS10 = {"StatusSNS":{"Time":"2018-09-18T10:37:32"}}
10:37:32 MQT: stat/A2/STATUS11 = {"StatusSTS":{"Time":"2018-09-18T10:37:32","Uptime":"0T00:02:55","Vcc":3.461,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"eir46325537-2.4G","BSSId":"50:01:D9:11:E0:10","Channel":6,"RSSI":60}}}