arendst/Tasmota #3728: Flashing D1 Mini V3 with Tasmota and Termite access

ripvega, 2018-09-07:
I'm having an issue with Tasmota and D1 mini too.

I'm getting a successful flash using ESPEasy but then when I try to access the D1 mini V3 using Termite, it communicates but all I see is gibberish, so can't set SSID and password to be able to access it's config.

I've tried every baud rate setting in Termite with the same result.

Flashing any Sonoff device this way works fine.

Should I be grounding GPIO3 to enabe flash mode? I am having to and wonder if this is part of the issue.

Being fairly new to all of this it's probably something simple.

Many thanks for any advice.

andrethomas, 2018-09-07:
GPIO 0 -> GND to enter flash mode

After that, you should have console access @11500bps 1N8 with software flow control (i.e. no RTS/CTS)

ascillato2, 2018-09-07:
Hi,

If you need more information, please follow the links. Thanks

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.<br />
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.<br />
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-09-07 by ascillato2

ripvega, 2018-09-07:
Thank you for the help.