arendst/Tasmota #3729: Can we store the console info/log or can we push the data to thingspeak?

lesonquan, 2018-09-07:
Hi all,
I have a project to monitor Electric Current by Sonoff Pow with Tasmota fw. I not only want to real time monitor, I also want to monitor the historical data. I dont know whether we have any way to store all info of Console or can we push the Current value directly to Thingspeak through MQTT?
 I also find the Configure Logging in Tasmota, but I dont know how to use it. 

EX: 
```
15:52:50 MQT: tele/temp/SENSOR = {"Time":"2018-09-07T15:52:50","AM2301":{"Temperature":29.6,"Humidity":71.3},"TempUnit":"C"}
15:53:00 MQT: tele/temp/STATE = {"Time":"2018-09-07T15:53:00","Uptime":"0T00:03:55","Vcc":2.997,"Wifi":{"AP":1,"SSId":"Anonymous","RSSI":76,"APMac":"10:6F:3F:4A:43:B0"}}
15:53:00 MQT: tele/temp/SENSOR = {"Time":"2018-09-07T15:53:00","AM2301":{"Temperature":29.6,"Humidity":71.4},"TempUnit":"C"}
15:53:10 MQT: tele/temp/STATE = {"Time":"2018-09-07T15:53:10","Uptime":"0T00:04:05","Vcc":2.997,"Wifi":{"AP":1,"SSId":"Anonymous","RSSI":88,"APMac":"10:6F:3F:4A:43:B0"}}
15:53:10 MQT: tele/temp/SENSOR = {"Time":"2018-09-07T15:53:10","AM2301":{"Temperature":29.6,"Humidity":71.4},"TempUnit":"C"}
15:53:20 MQT: tele/temp/STATE = {"Time":"2018-09-07T15:53:20","Uptime":"0T00:04:15","Vcc":2.997,"Wifi":{"AP":1,"SSId":"Anonymous","RSSI":98,"APMac":"10:6F:3F:4A:43:B0"}}
15:53:20 MQT: tele/temp/SENSOR = {"Time":"2018-09-07T15:53:20","AM2301":{"Temperature":29.6,"Humidity":71.4},"TempUnit":"C"}
15:53:30 MQT: tele/temp/STATE = {"Time":"2018-09-07T15:53:30","Uptime":"0T00:04:25","Vcc":2.997,"Wifi":{"AP":1,"SSId":"Anonymous","RSSI":84,"APMac":"10:6F:3F:4A:43:B0"}}
15:53:30 MQT: tele/temp/SENSOR = {"Time":"2018-09-07T15:53:30","AM2301":{"Temperature":29.6,"Humidity":71.4},"TempUnit":"C"}
```

lesonquan, 2018-09-07:
Can you share how to config to get the the info from Sonoff with Tasmota fw? Everything is new to me :D

ascillato2, 2018-09-07:
Hi,

If you need more information, please follow the links. Thanks

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-09-07 by ascillato2

lesonquan, 2018-09-08:
@mike2nl thank you so much for your kink support. it works like a charm 👍