arendst/Tasmota #3730: Wrong layout for sonoff basic if PWM is used

mkh595, 2018-09-07:
It is related to #3674 which was closed but I don't know how to reopen it.

Also in last version Version 6.2.1.2-2_3_0 if PWM1-5 is selected there is not only wrong layout 

[image]

[image]

but also PWM does not work

```
17:25:40 MQT: stat/home199/RESULT = {"Command":"Unknown"}
17:25:43 MQT: stat/home199/RESULT = {"Command":"Unknown"}
17:25:45 MQT: stat/home199/RESULT = {"Command":"Unknown"}
```

ascillato, 2018-09-07:
Hi,

I just configure a blank new device with the exact same GPIOs and PWM works perfect.

There is something wrong in your firmware. Please, erase all flash and flash again the precompiled firmware from release or development.

closed 2018-09-07 by ascillato2

mkh595, 2018-09-07:
It was used file from http://thehackbox.org/tasmota/sonoff-sensors.bin.
I didn't compile version 6.2.1.2 myself.

ascillato, 2018-09-07:
Ok, let me check that one

mkh595, 2018-09-07:
I used Esptool and steps like descibed in Wiki
```
esptool --port COM5 erase_region 0x0F4000 0x008000
esptool --port COM5 erase_flash
esptool --port COM5 write_flash --flash_mode dout --flash_size 1MB 0x0 firmware.bin
```
file was from http://thehackbox.org/tasmota/sonoff-sensors.bin version 6.2.1.2, no my compilation.

Then I tried ESPEasy Flasher, same result.

It has to be related to PWM, not to module, because same issue I have with Wemos D1 mini

ascillato, 2018-09-07:
I'm sorry. I can not reproduce your error. It works fine in a sonoff basic and in a wemos. I have just tested.

After you flash your devices and you set your SSID. What else you do?

arendst, 2018-09-07:
What's the problem?

If setoption15 is set pwm commands are ignored and user is expected to use full light commands like color, dimmer etc.

If you want pwm control set setoption15 0 and all nice features are gone.

Again, I see nothing wrong in above pictures.

mkh595, 2018-09-08:
Thanks, setoption15 is the problem.
In version 5.10 it was default OFF, therefore I was not aware about this flag. And upgrade via OTA keeps it OFF also in version 6.x.  Unlike direct flash of version 6.x set this flag ON by default.

Again thank you.