arendst/Tasmota #3732: Changing DST from console supported?

carryonrewardless, 2018-09-08:
II am on "Version":"6.2.0" with a Sonoff POW.
We are on timezone 10 in the Southern hemisphere with daylight saving starting Sun Oct 10 02:00 and std time resuming Sun Mar 28 02:00 2019.

So in the console:
03:07:37 CMD: timezone 99
03:07:37 MQT: stat/sonoff2/RESULT = {"Timezone":99}
03:07:41 MQT: tele/sonoff2/SENSOR = {"Time":"2018-09-08T03:07:41","ENERGY":{"Total":0.784,"Yesterday":0.174,"Today":0.037,"Power":11,"Factor":0.52,"Voltage":260,"Current":0.081}}
03:07:42 CMD: TIMEDST "1, 1, 10, 1, 2, 660"
03:07:42 MQT: stat/sonoff2/RESULT = {"TimeDst":{"Hemisphere":0,"Week":0,"Month":3,"Day":1,"Hour":2,"Offset":120}}
03:07:45 MQT: tele/sonoff2/SENSOR = {"Time":"2018-09-08T03:07:45","ENERGY":{"Total":0.784,"Yesterday":0.174,"Today":0.037,"Power":3,"Factor":0.13,"Voltage":259,"Current":0.098}}
03:07:52 CMD: TIMESTD "1, 0, 3, 1, 2, 600"
03:07:52 MQT: stat/sonoff2/RESULT = {"TimeStd":{"Hemisphere":0,"Week":0,"Month":10,"Day":1,"Hour":3,"Offset":60}}
03:08:01 MQT: tele/sonoff2/SENSOR = {"Time":"2018-09-08T03:08:01","ENERGY":{"Total":0.784,"Yesterday":0.174,"Today":0.037,"Power":4,"Factor":0.21,"Voltage":259,"Current":0.075}}
03:08:01 CMD: timestd
03:08:01 MQT: stat/sonoff2/RESULT = {"TimeStd":{"Hemisphere":0,"Week":0,"Month":10,"Day":1,"Hour":3,"Offset":60}}
03:08:05 CMD: timedst
03:08:05 MQT: stat/sonoff2/RESULT = {"TimeDst":{"Hemisphere":0,"Week":0,"Month":3,"Day":1,"Hour":2,"Offset":120}}

I thought I had syntax right from the Wiki and reading https://github.com/arendst/Sonoff-Tasmota/issues/2575 but get the strange results above.  Am I doing something wrong or is changing from the console not yet supported?  It seems to be reverting to defaults on setting changes.

Thanks for Tasmota and great support.

mike2nl, 2018-09-08:
@carryonrewardless 
You had read the possibnle commands?
Can be found in the WIKI.
Search for DST
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

carryonrewardless, 2018-09-08:
@mike2nl 
"I thought I had syntax right from the Wiki and reading #2575"

mike2nl, 2018-09-08:
OK, lets see...
You set:  TIMEDST **"**1, 1, 10, 1, 2, 660**"**
and got: RESULT = {"TimeDst":{"Hemisphere":0,"Week":0,"Month":3,"Day":1,"Hour":2,"Offset":120}}

Mistake in syntax, read the wiki ;-).

Dont use quotation marks. Use it in this way: TIMEDST 1, 1, 10, 1, 2, 660
Then it works. Same when i do this..

carryonrewardless, 2018-09-08:
Excellent. Thank you I will test when I get back home and report back here.

closed 2018-09-08 by ascillato2

AlBrough, 2019-03-06:
The wiki's command syntax is not that easy to understand for newer users, although it is amazingly comprehensive. Seeing the above example it makes it clearer on how to use it