arendst/Tasmota #3734: Switch S20 with Harmony App doens't work

ol3k, 2018-09-08:
As mentioned in #875 (the topic is already stale) I flashed a new S20
The status of the switch ist correctly seen in the harmony app but I'm not able to turn it on/off with the app

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff S20
- [X] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
11:07:56 CMD: status 0
11:07:56 SRC: WebConsole from 192.168.178.100
11:07:56 RSL: Received Topic /status, Data Size 1, Data 0
11:07:56 RSL: Group 0, Index 1, Command STATUS, Data 0
11:07:56 RSL: stat/sonoff/STATUS = {"Status":{"Module":8,"FriendlyName":["lights"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:07:56 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:57","StartupUTC":"2018-09-08T10:03:59","Sleep":0,"BootCount":10,"SaveCount":26,"SaveAddress":"FA000"}}
11:07:56 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-05T10:40:32","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:07:56 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["klaudic",""],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]}}
11:07:56 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
11:07:56 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1698","IPAddress":"192.168.178.50","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"5C:CF:7F:79:A6:A2","Webserver":2,"WifiConfig":5}}
11:07:56 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_79A6A2","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:07:56 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 08 10:07:56 2018","Local":"Sat Sep 08 11:07:56 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:17","Sunset":"19:18"}}
11:07:56 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-08T11:07:56"}}
11:07:56 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-08T11:07:56","Uptime":"0T00:03:57","Vcc":3.111,"POWER":"OFF","Wifi":{"AP":1,"SSId":"klaudic","RSSI":100,"APMac":"38:10:D5:82:28:BD"}}}

**To Reproduce**

- enable hue bridge emulation
- sync to harmony
- turn on/off with harmony app

**Expected behavior**
Turn on/off with harmony app

Jason2866, 2018-09-11:
Why do you open a new issue for the same as #875 ?

ol3k, 2018-09-11:
@Jason2866 

yes, it's correct that other apps are working. I tested on android:
- com.prodpeak.huehello
- com.philips.hueswitcher.quickstart 

The S20 should be part of an action in a harmony scene. And this is not working as mentioned.

So, there will be no workaround to get the "native" app to work?
Or is this a "logitech" failure/feature to stop using third-party-devices?

ascillato2, 2018-09-11:
Not as far as I know