arendst/Tasmota #3738: Wemos D1 to control(!) touch sensor?

prankousky, 2018-09-08:
Hi everybody,

This is more a wiring or "how do I do it" question... I have this LED light I would like to controll via tasmota. I have a Wemos D1 Mini **compatible** device that runs tasmota without any problems.

[image]

As you can see, the touch sensor is connected via a single wire to the board. It will accept a single press only, going from off->color1, color1->color2, color2-color3, colorx-colorfade, colorfade->off.

Can I connect this single wire to a Demos D1 Mini pinout and controll this light with it? It would be a rather dumb kind of controll, as I would have to emulate single button presses (so if I wanted color 4, I would have to send 4 button presses, etc), but this is fine to begin with.

(I have already seen a better project that will use an mini LED board, allowing one to actually set specific colors with one command etc., but I do not have the parts to build this here at the moment)

If this is possible, which GPIO would I connect the single wire to, and what would I set this GPIO as in tasmota?

Thanks in advance for your help :)

Frogmore42, 2018-09-08:
This is actually several questions.
1. Can you connect a wire to the D1?
Sure, but the D1 doesn't work with touch unless it has additional circuitry.  It is unclear if the touch sensitive circuit is on the touch sensor or on the main board.  If the black wire puts out 3V3 or ground based on touch, you should be fine on hooking it up.

2. Which GPIO should I use?  The wiki has a list of what to avoid.  I usually use GPIO 4 and 14 first.  The ones to not avoid are all about same.  Which number do you like?

3. How do I control the color?  That will depend on the circuitry of the board.  If you are not familiar with electronics, this will be hard to figure out.  Normally there is a control for each color channel.  The D1 is not able to control anything that requires power directly.  It can provide a logic signal that can control other circuitry that can control power.  The board obviously has such circuitry, but how to control it is not obvious from the pictures.
4. How do you make the button presses control the lights?  If you have figured out the items above, rules might be able to do this.  If that doesn't work, you probably need rules on your Home Automation system.

Not to discourage you, but this is actually quite a complex topic.  It is definitely not a good first project for Tasmota, unless you are already an expert in electronics.

ascillato2, 2018-09-09:
Hi,

If you have further questions, please ask. Also, remember that there is the discussion group and the Tasmota Chat. Thanks

closed 2018-09-09 by ascillato2

prankousky, 2018-09-09:
Sorry, this is not resolved yet as I didn't have time to test it.

How do I make a specific GPIO put out either 3V3 or touch? Like, let's say I connect this black wire (that is originally connected to the touch sensor) to GPIO2 / D4. What do I configure this pinout as in tasmta webgui (for example, button, relay, ...)?

I am pretty sure this touch button will only put out GND as there is nothing else attached to it. I can touch it, and it will switch the lights. There is no power source attached to it or anything of the like.

Also, maybe I did not correctly explain what I need it to do: I do not want tasmota to take the touch **as input** and then do something with the signal. I would like to **replace** this touch sensor **with** one of the pinouts on my nodemcu board. So instead of touching the sensor, I will send an mqtt message to tasmota and have it emulate my pressing the sensor (by, assumingly, either putting out 3V3 or GND directly to this board from my photo).

All the other questions (2-4) that @Frogmore42 has posted are irrelevant to me. Once I can have tasmota "emulate" me pressing this touch sensor, I know how to do everything else that might be possible.

tl;dr: how do I replace the touch sensor **with** one pinout from my nodemcu and control it...

Thanks

ascillato, 2018-09-09:
To use a Gpio as output set in Tasmota as relay

To emulate a button from another device, just send a mqtt cmnd from that device to Tasmota

Frogmore42, 2018-09-09:
Be aware that this might or might not work and might or might not damage the touch input circuit on your light.  It is going to depend on the particular circuit.  Here are some examples
https://www.allaboutcircuits.com/technical-articles/circuits-and-techniques-for-implementing-capacitive-touch-sensing/
To minimize the chance of damaging it, you should put a resistor on the output of the esp8266 (between it and the black wire).  If the circuit in the light uses the first example, it might not work with this technique, but at least with the resistor in place it won't destroy either part.

As I said before, not a best first project, since it is likely to lead to frustration and a lack of success.

prankousky, 2018-09-09:
@Frogmore42 

Thank you! I will check it out from home later :)

My plan for the actual project was to use this https://www.creatroninc.com/product/8-rgb-ws2812-addressable-led-stick/ led stick. However, I am just a hobbyist with no professional experience, so I have to find tutorials for all this stuff... as for now, I am not sure how to wire this stick to my nodemcu / Wemos D1 mini and then controll it via mqtt.

(that being said, I know how to control anything I can via mqtt, I can look that up in the wiki; but the wiring required in order to even being able to controll it via mqtt is all not that clear to me)

I was just going to do this as a test project... like I mentioned in the first post, once I get this to work (work= my nodemcu triggering the touch sensor), everything else -while probably overcomplicated and annoying- will just be a learning experience. 
I was planning on just going by presses: device is off by default, if I want red color, emulate 2 button presses, save state "color=red"; if I then want blue, look up "how many presses from red to blue", then send this amount of presses, and so on. Definitely not great, but just for fun until I can figure out how to use this led stick from the link above...

Frogmore42, 2018-09-09:
The ws2812 will probably be easier.  There are a lot of tutorials on using them and they look pretty easy.  I bought some recently, but have not tried them yet.  It really is only three wires and Tasmota can control them easily.  So you can send a MQTT command and the LEDs will turn red, without having to show any other colors.

prankousky, 2018-09-09:
This little stick is ws2812... I even own a ws2812 led strip, but it would be way too long for this (only need about 5cm of length)... but now that you've mentioned it I will look into controlling that led strip once more :)