arendst/Tasmota #3739: Interchange Option in Tasmota

DocGame69, 2018-09-08:
I have been looking for a long time for a reasonable solution for a Tasmota interchange. There are different approaches (relays, scripts in IOBroker etc.).
My question: Would not it be possible that a function is installed, in which one can enter the IP address of a "partner Tasmota". In this "Partner Tasmota" then the IP address of the other would be specified. When a device is switched on, the other one is also on and off in exactly the same way. To this white would have to be synonymous to realize a change circuit or I have a mistake?

Sorry vor the Bad Englisch (Google Translate :-) )

andrethomas, 2018-09-08:
The simplest way would be to get a mqtt server on your lan (like mosquitto) and use a HA software such as home assistant, hassio or node-red

ascillato2, 2018-09-08:
There are example or that in the wiki. You can control one device with another using mqtt broker, mqtt switchtopic, rules to publish mqtt commands, rules to send web commands and knx.

Hope this helps

closed 2018-09-08 by ascillato2