arendst/Tasmota #3741: ws2812 LED's

Chipz0r, 2018-09-08:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- x[ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Hardware used : wemos d1 mini
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

21:33:20 MQT: stat/tasled1/STATUS = {"Status":{"Module":18,"FriendlyName":["tasled1"],"Topic":"tasled1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:33:20 MQT: stat/tasled1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:33:58","StartupUTC":"2018-09-08T19:59:22","Sleep":0,"BootCount":77,"SaveCount":253,"SaveAddress":"F7000"}}
21:33:20 MQT: stat/tasled1/STATUS2 = {"StatusFWR":{"Version":"6.2.0.1","BuildDateTime":"2018-09-02T16:00:38","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:33:20 MQT: stat/tasled1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hax0r5",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
21:33:20 MQT: stat/tasled1/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
21:33:20 MQT: stat/tasled1/STATUS5 = {"StatusNET":{"Hostname":"tasled1-2949","IPAddress":"192.168.2.200","Gateway":"192.168.2.1","Subnetmask":"255.255.254.0","DNSServer":"8.8.8.8","Mac":"68:C6:3A:E1:4B:85","Webserver":2,"WifiConfig":5}}
21:33:20 MQT: stat/tasled1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.117","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E14B85","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:33:20 MQT: stat/tasled1/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 08 20:33:20 2018","Local":"Sat Sep 08 21:33:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:17","Sunset":"19:18"}}
21:33:20 MQT: stat/tasled1/STATUS10 = {"StatusSNS":{"Time":"2018-09-08T21:33:20"}}
21:33:20 MQT: stat/tasled1/STATUS11 = {"StatusSTS":{"Time":"2018-09-08T21:33:20","Uptime":"0T00:33:58","Vcc":2.706,"POWER":"ON","Dimmer":100,"Color":"FF0000","HSBColor":"0,100,100","Channel":[100,0,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"hax0r5","BSSId":"00:50:7F:E2:AD:5C","Channel":3,"RSSI":58}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Hi

I have 2 different type of ws2812 led strips

1 strip has RGB and the other strip has GRB, 
when i issue color 1 one goes red and the other goes green
color 2  green  and the other red
color 3 both blue

Is there a way to permanently tell tasmota about this reversal of colour ?

Many thanks

Frogmore42, 2018-09-09:
Put them on different devices and you can with a compile time option.  I don't believe you will find any library that supports different order on the same strip/pin.

arendst, 2018-09-09:
Yes you can.

In user_config.h is a define where you select the different color values needed. You'll have to recompile as the internal rgb/grb buffers are different.

Chipz0r, 2018-09-09:
Thanks Theo , that's worked great, and it has taught me how to compile and upload via atom/platformio which is much more easy than the arduino ide .

Chipz0r, 2018-09-09:
On a side note , i cannot use OTA upgrade anymore i presume as your prebuilt binary will be compiled differently, so could i build it as normal , and take the firmware.bin that it makes , put it on my internal web server and point my device to this bin and upgrade that way ?

ascillato, 2018-09-09:
Yes, exactly

closed 2018-09-09 by ascillato2

NorthernMan54, 2020-09-20:
Not to re-open a old issue but this was resolved with PR #5326 and with the implementation of setoption37