arendst/Tasmota #3742: MQTT keeps searching for server besides that i have deactivated it

dotbanana, 2018-09-08:
I have set following in user_config.h:
#define MQTT_USE               0                 // [SetOption3] Select default MQTT use (0 = Off, 1 = On)

But still Tasmota is constantly, every couple seconds, trying to connect to some ominous non-existing MQTT-Server. The connection is also blocking. I had to uncomment

#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT    // Use TasmotaMqtt library (+4k4 code, +4k mem) - non-TLS only
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support

for the networking to work and to access the webinterface.  
I would like to stop the system constantly searching for this MQTT-Server.   
How do i disable this _completely_?

ascillato, 2018-09-08:
Hi,

`#define MQTT_USE 0` is only the DEFAULT value, so, if you have a previous configuration in your sonoff, that configuration will remains until you reset (push the button for over 40 secs) or you a reset by console command.

So, if you want to disable MQTT in your actual configuration (as explained in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#mqtt-specific)), please just type in the console  `SetOption3 0`

Hope this helps.

dotbanana, 2018-09-08:
Thanks for your fast answer.  
I did the 40 second reset. Tried. Flashed again. Tried. Resetted again. Tried.  
Now i have this output in console:  
[image]  
MQTT has stopped trying to connect constantly and it looks like WIFI is connected accordingly but now i cannot access it from the browser unfortunately it says that the connection broke down while the page was loaded (translation could be wrong i have no english browser). 
I have never imagined that this is going to be that tricky in the end just to fire up a webserver and control a relay and the tutorials i found in the net about it are not really complete. Some mention this CFG_holder-value that it should be made higher but don't tell how much you should higher the value other tuts ignore this point completely. Is it from interest to do this? Is there a IRC maybe where such things can be asked. I know that i am here in the complete wrong category and i am sorry for that. 
I'll try further thanks. a lot.

ascillato, 2018-09-08:
Hi,

If you are having wifi issues, please, erase all flash and then flash again. In that case you will have the default config you set in user_config.h file.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.<br />
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.<br />
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-09-08 by ascillato2