arendst/Tasmota #3743: D1 Mini 3.3v relay output ?

c3reeman, 2018-09-09:
Hi,

I’m currently setting out to produce a large circuit board that can switch 40 different circuits I’d  originally planned to use 40 individual sonoff’s ( my knolage is still quite basic ) however I’ve now come across the D1 Mini and found I’d only need 10 of them as each D1 can have 4 switches and 4 relays on them, I found I can easily flash this with tasmota and can make it a very simple plugin module on a circuit board.. I like the fact that if I ever get a fault I can simply unplug it of the pin header and fit a new one without having to touch the 240v side of the circuit.

My question is this... I’ve tested setup of the switch relay module and got it all working great in tasmota however I noted that the relay output voltage from the D1 was only 3.3v I was expecting 5v.

I’ve orderd 1 of these off aliexpress a few weeks ago and hopefully it should arrive soon.

http://s.aliexpress.com/UvIjUnau

I figured because it was a 5v trigger ( my exp elected D1 output ) I could put the output from the D1 straight onto the trigger points on The board.... however it seems I can’t do that now given it’s only 3.3v ??

Any advice would be much appreciated, will the relay switch with just 3.3 v or is some extra components going to be required ? And if so what

Thanks

Jason2866, 2018-09-09:
Please read in wiki the part https://github.com/arendst/Sonoff-Tasmota/wiki/MCP23008-MCP23017
which this extension you can expand Input/Output GPIOS.
For further questions ask for help in DISCORD Chat https://discord.gg/Ks2Kzd4

c3reeman, 2018-09-09:
Many thanks @Jason2866 that’s great regarding extra input/output I’ve orderd some from Aliexpress,

Any recommendations regarding my relay switch supply voltage ??

ascillato, 2018-09-09:
You will have to test that. There are some relay boards that have photodiode isolation and need 5v for the relay coils and 3.3v for the logic. So, in that case you will need an extra 5v source for the relays but the logic is connected directly from your wemos.

ascillato2, 2018-09-09:
If you have further questions, please ask. Also you can use the discussion group and the Tasmota Chat

closed 2018-09-09 by ascillato2