arendst/Tasmota #3745: Sonoff Pow R2: Measurement less than 5W not possible

PPP01, 2018-09-09:
Hi, i recently got my Pow2 and flashed v6.2.1.
When I try to measure a consumer with less than 5w, the POW R2 measures only 0A and 0W. Only 225V is displayed correctly. Everything above 6W is measured correctly and fairly accurately.

A second Pow R2 with original firmware can display smaller loads correctly. Also my Pow R1. 
I already tried to calibrate with PowerSet, CurrentSet and VoltageSet, but that did not helped.
Is there another setting I can try? 

Best regards
Patric

ascillato2, 2018-09-09:
No. The R2 is better than R1. But both are not accurate with very low consumptions.
For calibration you need a 60W load at least. From the original firmware it cames already calibrated. When you flash Tasmota, a calibration is needed.

PPP01, 2018-09-09:
Well, I tried to calibrate with a 60W bulb, but as you said: it is very bad with low consumptions. My pow R1 has no problems to measure 2,5W, the R2 shows only 0W. 
Cause I can't use the original firmware at all, i will flash the second Pow R2 and will see if it will perform better.
Thanks anyway!

closed 2018-09-09 by ascillato2