arendst/Tasmota #3746: Sonoff basic unable to connect to wifi after flash

Mira80, 2018-09-09:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Hardware used : Sonnof basic,  flashed by Arduino IDE, version in log dump above
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
After flashing, sonoff basic us unable to connect to wifi. In serial monitor, I see this

"
0:00:00 CFG: Use defaults
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_691A6F, GroupTopic sonoffs) Version 6.2.1-2_4_2
00:00:00 WIF: Connecting to AP1 ob43 in mode 11N as sonoff-6767...
00:00:09 WIF: Connect failed as AP cannot be reached
00:00:09 WIF: Connecting to AP1 ob43 in mode 11N as sonoff-6767...
00:00:18 WIF: Connect failed as AP cannot be reached
00:00:18 WIF: Serial active for 3 minutes
"

I tried wifi manager also, but I see sonoff AP, but it is unable to connec to it (probably AP did not assign IP address)

I looked up by google, if i found some solve, but unsuccessful.

Thank you for support

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-09:
Hi

Sonoff.bin v6.2.1, from release had a compilation issue. Please try to download it again or use the precompiled bins from http://thehackbox.org/tasmota/

Thanks

For more information issue #3678

closed 2018-09-09 by ascillato2

Mira80, 2018-09-09:
Im sorry, i tried download again (sonoff.ino was changed 6 hours ago) but with the same result, precompiled .bin file I dont know, how to upload (i tried install python and flash by esptool, but i got error "no module named serial"
Excuse me, im newbie in this.

ascillato, 2018-09-09:
Hi,

Try to download the _sonoff.bin_ from http://thehackbox.org/tasmota/
Those are the ones of the actual code in development but are working fine.

For flashing with **esptool.py**, please follow the instructions of the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#esptool-usage

### For Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.