arendst/Tasmota #3747: Customizing GPIOs on compile time

aspitznyc, 2018-09-10:
Hi there!

I have a bunch (like 70) Sonoff Basic and (around 25) Sonoff Dual R2, and everytime I update their firmware due to a new release I have to manually restore each one's configuration from backup (Restore Configuration from Configuration Menu on WEB UI).

What I'm trying to achieve is to set the GPIO14 on Sonoff Basic and GPIO00 and GPIO09 on Sonoff Dual R2 as Switch1 and Switch2 by default so it's faster the whole update process (I know, I might just update and not reset the config, but I like a "fresh" start).

I realize that I can set this up on sonoff_template.h by setting GPIO_SWT1 and GPIO_SWT1 on each GPIO, but that makes the option dissapear from the WEB UI.

Is there a way to define those Switch on compile but make the appear on WEB UI? (Does that makes sense?)

Thanks!!!

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-09-10:
You really should fill out the troubleshooting template.  Updating the FW should NOT erase the settings (unless, of course, you do a complete erase first).

There is some information about this in the wiki and other issues.  There are ways that you can do it incorrectly and it is possible that the process might fail.  But, generally the settings are preserved.

There was an issue from someone who wanted to preselect a configuration, since he had hundreds of devices that he needed to flash.  I don't remember the exact outcome, but I remember suggesting using automation to program the settings he needed, since he was flashing by serial.  Your situation might be different.  But your reason for wanting to do it indicates a problem with the way you are updating the firmware.  On reading your issue again, it seems you are doing an erase.  So, you are getting what you wanted.  I don't believe there is a way to assign user GPIO pin selections like you want.

ascillato, 2018-09-10:
duplicated issue: https://github.com/arendst/Sonoff-Tasmota/issues/3258

closed 2018-09-10 by ascillato2