arendst/Tasmota #3750: Remote upgrades when OTA failing

smadds, 2018-09-10:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Hardware used :Various 
- [ ] Provide the output of command ``status 0`` :
```
12:24:57 MQT: stat/th16-wc01/STATUS = {"Status":{"Module":4,"FriendlyName":"th16-wc01","Topic":"th16-wc01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:24:57 MQT: stat/th16-wc01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://172.16.106.1:80/data/firmwares/sonoff-minimal.bin","Uptime":37,"Sleep":0,"BootCount":2537,"SaveCount":3926,"SaveAddress":"F7000"}}
12:24:57 MQT: stat/th16-wc01/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDateTime":"2017-12-01T13:55:52","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:24:57 MQT: stat/th16-wc01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"EquipmentTCL","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"55800009"}}
12:24:57 MQT: stat/th16-wc01/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
12:24:57 MQT: stat/th16-wc01/STATUS5 = {"StatusNET":{"Hostname":"th16-wc01","IPAddress":"172.16.106.10","Gateway":"172.16.0.1","Subnetmask":"255.255.0.0","DNSServer":"8.8.8.8","Mac":"5C:CF:7F:20:CC:66","Webserver":2,"WifiConfig":2}}
12:24:57 MQT: stat/th16-wc01/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.106.1","MqttPort":1883,"MqttClientMask":"DVES_X","MqttClient":"DVES_20CC66","MqttUser":"tcl-admin","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
12:24:57 MQT: stat/th16-wc01/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 10 12:24:57 2018","Local":"Mon Sep 10 12:24:57 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":0}}
12:24:57 MQT: stat/th16-wc01/STATUS10 = {"StatusSNS":{"Time":"2018-09-10T12:24:57","AM2301":{"Temperature":25.4,"Humidity":43.9},"TempUnit":"C"}}
12:24:57 MQT: stat/th16-wc01/STATUS11 = {"StatusSTS":{"Time":"2018-09-10T12:24:57","Uptime":37,"Vcc":3.156,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EquipmentTCL","RSSI":56,"APMac":"02:4F:57:D3:D3:80"}}}
1
```
Does anyone know of a way of doing upgrades when the normal OTA route fails and the devices are difficult to get to? I think a previous update may have been done without the DOUT set and now even the Minimal version uploads fail.

ascillato, 2018-09-10:
Hi,

Have you tried upgrade first to 5.14 and then to 6.2.1?

smadds, 2018-09-10:
Yes, if I file upload 5.14 Minimal I can see it stall at between 30-50% upload

smadds, 2018-09-10:
Yes (TasmoAdmin), but it fails on manually uploaded files too

ascillato, 2018-09-10:
If your OTA hungs in the middle of the process, sorry but to upgrade you will need to flash it by cable. If the previous version had a different memory map, OTA will not work.

Frogmore42, 2018-09-10:
Are you doing OTA from a web server or upload from a file on your PC?

If all of the upload is failing to get to the device, that seems to indicate an issue with networking.  Your rssi at 56 is not bad, but is definitely not good either.

If you are unsuccessful in getting all of the binary on the device, there are only two options.  One, be happy with the version you have.  As long as it is doing what you need, why do you need/want to update it?  Two, dig it out of where ever you have it and connect it via serial to flash it correctly.

I have a device that is running 5.08.  It has been running great for a log time.  I have several others that are running different versions.  Most of them aren't hard to access, but there has been no response good reason to update them.

smadds, 2018-09-10:
Makes no difference whether its file upload or server, @Frogmore42 (except I get a bit more status with file upload). I have also tried other devices with much higher wifi signal strength, but the same results. 

I guess I'm going to have to do some ladder work...